aruttkamp/ioBroker.rct #4: Keine Instanz hinzugefügt

derdaene, 2021-07-15:
Ich habe den Adapter installiert. Leider habe ich lediglich einen leeren Punkt in den "objects". Bei den Instanzen ist kein Adapter gelistet. So kann ich auch keine Einstellungen vornehmen.
Was kann ich tun? 

**Versions:**  
 - Adapter version: latest
 - JS-Controller version: 14.17.2
 - Node version: 14.17.2
 - Operating system: debian

lauff, 2021-07-15:
Nachdem der Adapter installiert wurde, muss danach auch noch eine Instanz angelegt (wie bei allen Adaptern / mit dem "Plus"-Zeichen).

Wenn das nicht geklappt hat, was steht dazu im Log / Protokoll?

derdaene, 2021-07-15:
😅ich habe nach „rct“ gesucht, da hat er mir das bei „Instanzen“ nicht angezeigt.
Nun habe ich nach „Power“ gesucht und dann war er da. 
sorry. 🙈
Funzt wunderbar! Danke!

closed 2021-07-16 by lauff

sffranke, 2022-08-27:
Hat jemand einen Tipp wie ich den Adapter hinzufüge?  Ich stehe völlig auf dem Schlauch :-/
iobroker 5.3.8
NodeJS 14.20.0
JS-Controller 3.3.22
npm install iobroker.rct bewirkt nichts, iobroker list adapters -> Adapter erscheint nicht.

Danke!