aruttkamp/ioBroker.rct #5: fsevents expecting darwin but is linux on raspberry buster

Ralf31337, 2021-09-17:
I have trouble installing on an RPI 4 with debian buster:
make: Leaving directory '/opt/iobroker/node_modules/canvas/build'
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: fsevents@2.3.2 (node_modules/fsevents):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for fsevents@2.3.2: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})
npm WARN optional SKIPPING OPTIONAL DEPENDENCY: osx-temperature-sensor@1.0.7 (node_modules/osx-temperature-sensor):
npm WARN notsup SKIPPING OPTIONAL DEPENDENCY: Unsupported platform for osx-temperature-sensor@1.0.7: wanted {"os":"darwin","arch":"any"} (current: {"os":"linux","arch":"arm"})

I already executed some preconditions, which brought me further but still have trouble with fsevents:
- libpixman
- libpangocairo

lauff, 2021-09-18:
Hi Ralf,

As the modules mentioned in the error log, such as fsevents, osx-temperature-sensor, libpixman, libpixman, libpangocairo are not use by this adapter, I expect that this not related to this RCT adapter. 

@OTHERS reading this. In case you know how this might be related, please let me know.

Warm regards
Markus

closed 2021-09-18 by lauff

Ralf31337, 2021-09-18:
Hi Markus,

I suspect the issue was that my npm tried to compile things it didn't find for arm (raspberry).

After apt update, apt upgrade and npm install -g npm the issue was gone.

I managed to install and use the adapter now. Thanks for your efforts so far.

Best,
Ralf