aruttkamp/ioBroker.rct #6: Adapters stops working if the inverter goes to sleep

Haeusele, 2021-11-14:
When the inverter goes to sleep, the adapter stops working and ioBroker shows a message that the adapter crashes to often...

Could you implement something to avoid the crash and wait until the inverter comes back to life? 
The inverter start working when dc power is provided. I had this problem two times until now. This means, that I have to restart the adapater manually. Until this point in time I loose all data (not stored in the database) ... :(

Thanks! :)

lauff, 2021-11-16:
Hallo,

`When the inverter goes to sleep, the adapter stops working`

den Fall hatte ich noch nicht. Wann genau geht die Anlage in den „Sleep-Modus“?

Verwenden Sie die aktuelle Version (v0.0.6) des RCT-Adapters?

`Until this point in time I loose all data (not stored in the database)`

Für das Loggen und Speichern von Daten ist der History-Adapter verantwortlich. Damit sollten auch beim Beenden des RCT-Adapters keine Daten verloren gehen.

Mit freundlichem Gruß
Markus Lauff

Haeusele, 2021-11-28:
Hallo,

in den letzten Tagen ist das Problem nicht mehr aufgetreten. Warum es überhaupt zweimal bisher vorkam, kann ich leider nicht sagen.

Der Wechselrichter schaltete sich "ab" bzw. ging in den Schlafmodus und schaltet sich wieder ein, wenn Strom von den Modulen kommt. Könnte es daran liegen, dass wir eine Ersatzstromversorgung haben? Es gab zwar keinen Stromausfall, aber unser WR bietet die Möglichkeit.

Ja, ich verwende die Version 0.0.6 des RCT-Adapters.

Das der Adapter nicht mehr arbeitet fällt mir meist dann auf, wenn ich die aktuellen Daten WR anschauen möchte. Kann also gut sein, dass der WR wieder seit 2 Stunden arbeitet, aber der ioBroker den Adapter nicht wieder startet (die Anzeige sagt, dass es zu viele Abstürze gab).

Welche Infos kann ich noch liefern?

Vielen Dank
Marco

lauff, 2021-12-03:
Hallo,

 

ich habe auch eine Ersatzstromversorgung von RCT, daran kann der Unterschied wohl eher nicht liegen. Meine Anlage geht aber auch nicht in StandBy.

 

Im Code habe ich eigentlich schon die Funktion eingebaut, dass der Adapter versucht eine verlorene Verbindung wieder aufzubauen, vermutlich macht er das aber nicht über einen längeren Zeitraum. Ich schaue mir das bei nächster Gelegenheit nochmal an, da ich momentan aber stark eingebunden bin, kann es etwas dauern.

 

Mit freundlichem Gruß

Markus

lauff, 2022-08-18:
Ich sehe hier keine Möglichkeiten etwas zu ändern und schließe daher das Ticket.
Bitte melden, wenn das Problem weiter besteht und mehr Informationen vorliegen.

closed 2022-08-18 by lauff