aruttkamp/ioBroker.rct #7: Weiterentwicklung?

Dirk-I, 2022-01-21:
Hallo Markus, 
da ich auch eine Anlage von RCT in betrieb habe. 
Würde mich interessieren, ob du diesen Adapter weiterentwickelst?

Wenn ja - würde ich auch gleich meine bescheidene Unterstützung und Hilfe anbieten.

VG
Dirk

lauff, 2022-01-22:
Hallo Dirk, 

in dem aktuellen Stand läuft der Adapter für mich stabil und erfüllt meine notwendigen Funktionen. 

Bei Bedarf würde ich auch mit begrenzter Kapazität weiterentwickeln. Unterstützung ist aber auch mehr als willkommen. Genre mach ich das Repo dafür auf.

Gruß Markus

Dirk-I, 2022-01-22:
Hallo Markus, 
danke für die schnelle Antwort. Leider kann ich nicht wirklich programmieren daher kann ich helfen beim testen und Ideen einbringen. Quellcode lesen geht so gerade eben noch. 

Grundsätzlich reichen mir die Daten auch. 
Mir würde nur der Wert aus -- rct.cmd['power_mng.soc_min'] = { serial: 546, id: 'CE266F0F' }
Diesen habe ich per Hand im Adapter hinzugefügt. 
Diesen Wert würde ich auch gerne ändern können. In der App geht das auch ohne den Flashspeicher zu nutzen.

Was ich auch noch nicht herausgefunden habe, wie ich sicher herausbekomme ob sich das Haus im Ersatzstrom befindet. 
Abfragen ob kein Strom aus dem Netz bezogen wird geht zwar ist aber nicht sehr sicher. 
Vielleicht hast Du da noch eine Idee.

VG
Dirk

ldittmar81, 2022-01-24:
@lauff wenn dein Adapter schon funktioniert, wäre es cool ihn ins ioBroker Repository als latest/beta zum Testen freizugeben. Viele würden sich freuen :-)

Infos dazu findest du hier: https://github.com/ioBroker/ioBroker.repositories#iobrokerrepositories - bei Fragen sind wir jederzeit da :-)

andybiv, 2022-10-28:
Hi Markus, iobroker.rct wir mir in meiner iobroker installation nicht zur Auswahl angezeigt - was muss ich machen? Vielen Dank für deine Arbeit

closed 2023-03-26 by aruttkamp