aruttkamp/ioBroker.rct #13: Adapter startet neu

SpitfireXP, 2022-11-01:
Der Adapter Startet immer wieder neu und wird dann vom iobroker gestoppt.

Ich habe den neuesten iobroker mit dem Adapter 1.0.1 am laufen.

Mit der alten Version lief es reibungslos.
[image]

topcake01, 2023-02-01:
Hi.
Das passiert wohl wenn der Wechselrichter mangels PV und Batteriestrom offline geht und die IP nicht mehr erreichbar ist.
Dann startet der Adapter ständig neu (laut Host Adapter).
Ich habe mir jetzt erstmal so geholfen, dass der RCT Adapter bei Sonnenaufgang gestartet wird und (da ich noch keine Batterie habe) bei Sonnenuntergang per Script gestoppt wird.
Evtl. fehlt hier ein Failover in Form eines __client.on("error"...

Gruß
Michael

SpitfireXP, 2023-02-01:
Der PowerStorage ist immer an.
durch die Batterie hat der auch immer Spannung.
Beim Inverter wäre das klar, da der sich abschaltet.

Mal davon abgesehen habe ich dieses Problem auch nur auf meinem zweiten iobroker system.
Hier mit der Version 1.0.2.
Auf meinem ersten System mit der 1.0.0 läuft das alles top.

Ich werde das jetzt mal frisch neu aufsetzen und dann schauen ob das Problem noch immer da ist.

dahadd, 2023-02-05:
Hallo zusammen,

bin ganz neu bei der ioBroker Sache und noch total grün hinter den ohren.

Bei mir startet der Adapter nur kurz. Verbindet sich mit dem WR und dann wieder diconnect.
Benutze Version 1.0.1.
Musst ich noch etwas bei "RCT elements" eintragen?
Oder wo liegt mein Fehler. Dankeschön

topcake01, 2023-02-23:
Hallo.
Der 1.0.1 stürzt ab wenn die IP des WR nichtmehr erreichbar ist.
Bei mir läuft der Adapter 1.0.1 "flüssig" solange der WR Strom hat.

Evtl. hast du ein generelles Verbindungsproblem was auch die verzögerte Aktualisierung erklären würde.
Falls du WLAN benutzt wäre ein Test per LAN-Kabel sinnvoll.

Gruß
Michael

dahadd, 2023-02-24:
> Hallo. Der 1.0.1 stürzt ab wenn die IP des WR nichtmehr erreichbar ist. Bei mir läuft der Adapter 1.0.1 "flüssig" solange der WR Strom hat.
>
> Evtl. hast du ein generelles Verbindungsproblem was auch [...]

Danke Michael für die Info.
Bin per LAN verbunden.
Adapter 1.0.0. läuft bei mir sehr gut. 
1.0.1macht immer wieder Probleme.
Und ja, wenn der WR aus geht, dann wird der Adapter geschlossen.
Das sollte ja dann funktionieren, wenn der Speicher endlich angeschlossen ist.

mfg und schönes WE
Daniel

aruttkamp, 2023-03-02:
Das Problem habe ich aktuell auch noch. Kein Switch - kein Speicher. ... 
Ich habe ein Skript , dass prüft ob der WR noch erreichbar ist. Wenn nicht ,dann wird der Adapter runtergefahren. 
Morgens dann entsprechend wieder gestartet wenn wieder online

lauff, 2023-03-24:
Bzgl. dem Fehler mit der Netzwerkverbindung. Mann kann den sicherlich über try/catch abfangen und dann immer wieder (z.B. alle 5 Minuten) probieren, ob die Verbindung neu aufgebaut werden kann. Ich hatte das Problem nicht, da ich eine Batterie habe und der Wechselrichter damit immer zur Verfügung stand. Leider kann ich es auch nicht mehr einbauen / testen, da ich kein ioBroker mehr am laufen habe.

aruttkamp, 2023-08-07:
Hab das Problem mal hier dokumentiert
https://github.com/aruttkamp/ioBroker.rct/discussions

Schliesse daher dieses Issue - Eine Erweiterung des Adapters ist nicht sinnvoll , da zu aufwändig gegenüber der aufgezeigten Lösung.

closed 2023-08-07 by aruttkamp