aruttkamp/ioBroker.rct #14: NOTE: PROJECT NO LONGER MAINTAINED

lauff, 2023-02-02:
After I switched to Home Assistant, I do no longer maintain this adapter.
If you are interested to take it over, please let me know.

Warm regards,
Markus

duejo46, 2023-02-11:
Hallo und guten Tag,
dies ist sehr schade. Ich versuche gerade, meine neue rct-Anlage in den iobroker zu bringen.  Eine Übernahme Deiner Arbeit traue ich mir nicht zu. Installation und Einrichtung funktionierten, Datenpunkte werden angelegt und aktualisiert. Habe leider das Problem, dass ich die Instanz nicht stabil bekomme. Syteme sind rpi4 mit admin 6.3.5, nodejs 16.19.0, npm 8.19.3, jsc 4.0.24. Früher oder später passiert folgendes, sowohl in V1.0.0 als auch in V1.0.1
`<html><body>

host.iobrokerkurs | 2023-02-11 13:26:34.309 | warn | Do not restart adapter system.adapter.rct.0 because restart loop detected
-- | -- | -- | --
host.iobrokerkurs | 2023-02-11 13:26:34.308 | info | Restart adapter system.adapter.rct.0 because enabled
host.iobrokerkurs | 2023-02-11 13:26:34.308 | error | instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
rct.0 | 2023-02-11 13:26:33.624 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
rct.0 | 2023-02-11 13:26:33.623 | info | terminating
rct.0 | 2023-02-11 13:26:33.618 | error | Exception-Code: ECONNRESET: write ECONNRESET
rct.0 | 2023-02-11 13:26:33.617 | error | Error:  write ECONNRESET    at afterWriteDispatched  (node:internal/stream_base_commons:160:15)    at writeGeneric  (node:internal/stream_base_commons:151:3)    at Socket._writeGeneric  (node:net:905:11)    at Socket._write (node:net:917:8)    at  writeOrBuffer (node:internal/streams/writable:391:12)    at _write  (node:internal/streams/writable:332:10)    at Socket.Writable.write  (node:internal/streams/writable:336:10)    at  /opt/iobroker/node_modules/iobroker.rct/rct/rct.js:70:15    at  Array.forEach (<anonymous>)    at Timeout.requestElements [as  _onTimeout] (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:68:16)
rct.0 | 2023-02-11 13:26:33.616 | error | uncaught exception: write ECONNRESET

</body>
</html>`

Kann jemand weiterhelfen?
Beste Grüße
duejo

topcake01, 2023-02-23:
Hallo.

Das passiert meiner Erfahrung nach mit der v1.0.1 dann wenn die IP des WR aus welchen Gründen auch immer nicht(-mehr) erreichbar ist. Ich habe beides RCT WR und ioBroker per LAN-Kabel verbunden.

Evtl. mit einem Skript, dass prüft ob die IP erreichbar ist kompensieren.
Nach 3 maligem neustart mag der host nicht mehr und der Adapter ist erstmal deaktiviert.

Hier mal mein Skript, mit dem ich, da ich noch keine Batterie habe den Adapter zuverlässig starte und stoppe.
Bei dir wäre eine Regelung der ständigen Prüfung der Erreichbarkeit sinvoll.

// RCT start
schedule("*/15 4-12 * * *", async function () {
  exec('ping -c 1 192.168.1.134', async function (error, result, stderr) {
      if (result.indexOf('ttl') + 1 > 0 && getState("system.adapter.rct.0.alive").val == false) {
      // 10 Minuten warten bis zum einschalten
      timeout = setTimeout(async function () {
        exec('iobroker start rct.0');
        console.warn("exec: " + 'iobroker start rct.0');
      }, 600000);
    }
  });
});
// RCT stop
on({id: "rct.0.prim_sm.state"/*state*/, change: "ne"}, async function (obj) {
  var value = obj.state.val;
  var oldValue = obj.oldState.val;
  if (getState("rct.0.prim_sm.state").val == 6 && getState("system.adapter.rct.0.alive").val == true && compareTime("15:00", "null", ">")) {
    // Adapter aus wenn keine Sonne mehr
    exec('iobroker stop rct.0');
    console.warn("exec: " + 'iobroker stop rct.0');
  }
});

Grüße
Michael

topcake01, 2023-02-23:
> After I switched to Home Assistant, I do no longer maintain this adapter. If you are interested to take it over, please let me know.
> 
> Warm regards, Markus

Dear Marcus,
Because I want try to fix this "uncaught exception" issue while I use both RCT Inverter and ioBroker I'm intrested to take over this project.

Regards 
Michael

duejo46, 2023-02-25:
Hallo Topcake01/Michael,
danke für Deine Unterstützung. Bin leider kein Script-Fachkundiger. 
Mithilfe eines workarounds "Neustart des rct-Adapters alle 2 Minuten" habe ich vor ein paar Tagen erstmal Stabilität in die Instanz bekommen und dann schrittweise die Zeiten erhöht. Positiv scheint auch ein update des sytems, ua. nodejs von 16.19.0 auf 16.19.1 zu wirken. Auf alle Fälle starte ich aktuell nur noch jede Stunde den Adapter neu und habe keine Probleme. Ich werde die Zeit weiter verlängern und ggfs. den workaround dann abschalten. Eine Begründung vermag ich nicht zu nennen...
Auch danke ich Dir für Deine Absicht, den Adapter "betreuen" zu wollen. Nach können und vermögen helfe ich gerne.
Beste grüße
Joachim

lauff, 2023-02-25:
Hallo Michael,

super, dass Du das übernehmen möchtest. Ich gebe Dir alle Berechtigungen für das Projekt und kann bei organisatorischen Sachen gerne noch unterstützen.

Gruß Markus

aruttkamp, 2023-03-02:
Moin ! 
Wenn noch Hilfe benötigt wird - Ich würde den Adapter weiter pflegen wollen. 
Ich nutze ihn Schon seit letzem Jahr und denke ich nutze ihn auch noch mind. 10 Jahre ;-) 
Dann kann ich Ihn auch pflegen *lach*

aruttkamp, 2023-03-17:
Schade dass nun gar nichts mehr passiert. Oder übersehe ich irgendwas ?

NCIceWolf, 2023-03-23:
> Schade dass nun gar nichts mehr passiert. Oder übersehe ich irgendwas ?

Soweit ich das verstanden hatte wollte Michael (@topcake01) das weiterführen, ggf. wollte Joachim (@duejo46) noch unterstützen. Als Tester/Supporter wäre ich auch dabei, hab mich nur damals für die Systemintegration entschieden und bin daher nicht so fit in den Programmiersprachen

aruttkamp, 2023-03-23:
Ich wundere mich auch – zumal ich mich auch angeboten hatte.

Soweit ich das verstanden hatte wollte Michael ***@***.***<https://github.com/topcake01>) das weiterführen, ggf. wollte Joachim ***@***.***<https://github.com/duejo46>) noch unterstützen. Als Tester/Supporter wäre ich auch dabei, hab mich nur damals für die Systemintegration entschieden und bin daher nicht so fit in den Programmiersprachen

duejo46, 2023-03-24:
Hallo Mitstreiter, Ich habe leider auch keinerlei Info. Haben wir also Geduld, bis topcake01 sich meldet.
Beste Grüße

lauff, 2023-03-24:
Hallo, ich habe jetzt @NCIceWolf und @aruttkamp dem Projekt hinzugefügt. Wenn jemand fehlt, bitte melden.

lauff, 2023-03-24:
Bzgl. dem Fehler mit der Netzwerkverbindung. Mann kann den sicherlich über try/catch abfangen und dann immer wieder (z.B. alle 5 Minuten) probieren, ob die Verbindung neu aufgebaut werden kann. Ich hatte das Problem nicht, da ich eine Batterie habe und der Wechselrichter damit immer zur Verfügung stand. Leider kann ich es auch nicht mehr einbauen / testen, da ich kein ioBroker mehr am laufen habe.

aruttkamp, 2023-03-24:
> Bzgl. dem Fehler mit der Netzwerkverbindung. Mann kann den sicherlich über try/catch abfangen und dann immer wieder (z.B. alle 5 Minuten) probieren, ob die Verbindung neu aufgebaut werden kann. Ich [...]

Wäre eine Idee - dann wäre der Adapter autark man bräuchte kein Skript drumrum.

aruttkamp, 2023-03-27:
@lauff  Meldest Du dich bitte mal auf meine Mail ? 
DANKE !

closed 2023-04-10 by aruttkamp