aruttkamp/ioBroker.rct #15: Frage: Gibt es eine API? Oder woher kommen die Werte :)

Madaraxa, 2023-02-20:
Hey.
Erstmal vielen Dank für den Iobroker Adapter. Ich habe den und er läuft. Um mir selber die Arbeit eines Skripts zum Überschussladen zu sparen, würde ich gerne eine fertige Lösung nutzen. Hierfür brauche ich (am einfachsten) die URL, die mir die Daten des Objekts "rct.0.g_sync.p_ac_sum_lp" zeigt. Gibt es hier eine API auf die du zugreifst? Oder wie werden die Daten abgerufen?
Danke im Voraus

topcake01, 2023-02-23:
Hallo.
Es gibt zwar eine RCT API, auf dieser basiert ja hier der Adapter, aber diese ist nicht über eine einfache URL Anfrage adressierbar.
https://rctclient.readthedocs.io/en/latest/

Einfache ist es wenn der iobroker.rct bei dir läuft den Datenpunkt per URL mit dem SimpleAPI Adapter abzurufen:
https://github.com/ioBroker/ioBroker.simple-api

Gruß
Michael

Madaraxa, 2023-02-23:
Hi,
ok. Das habe ich befürchtet und zwischenzeitlich (eigentlich nur als Übergangslösung) so gemacht. Ich bin eigentlich der Freund davon nicht alles von einem System (iobroker) abhängig zu machen, wenn es anders geht, aber das scheint hier ja nicht möglich zu sein :)
Danke trotzdem für den Tipp
Grüße
Max

closed 2023-02-23 by Madaraxa