aruttkamp/ioBroker.rct #16: Falscher Werte-Typ zwischen Wechselrichter und Adapter?

NCIceWolf, 2023-03-08:
Ich erhalte regelmäßig unten stehenden Log-Eintrag, für alle Werte.
Kann es sein dass es hier eine Änderung der Typen gab? Oder ist das seit Anfang an so?

Der WR ist per LAN verbunden und ich habe ein 10 Sekunden Intervall für die Abfragen konfiguriert.

Auszug aus dem Log:  
State value to set for "rct.0.g_sync.p_ac_sum_lp" has to be type "number" but received type "string"

Da alles funktioniert, gehe ich davon aus dass die Werte aktuell konvertiert werden und als "number" abgelegt werden.
Wieso vom WR allerdings die Werte als "string" kommen sollten, verstehe ich nicht...

Versionen:
 - Adapter version: <v1.0.1>
 - JS-Controller version: <v16.18.1>
 - Operating system: <Docker Container>
 - RCT Power Storage DC 10.0 Control Software Version: <2.3.5658>

aruttkamp, 2023-03-23:
Moin NCIceWolf,
bist du sicher das es eine "Fehler" Meldung ist und nicht nur eine "Warnung" ? 
Welchen Loglevel hast du eingestellt für den Adapter ?

NCIceWolf, 2023-03-24:
Hätte ich natürlich auch dazu schreiben können...
Handelt sich hierbei um eine **Info**-Meldung, momentan habe ich "noch" alle Adapter auf diesem Log-Level laufen.

Wie beschrieben stellt das aktuell auch kein Problem dar, alle Werte laufen ja sauber in die DB. Nur nicht dass es damit künftig Probleme gibt.

lauff, 2023-03-24:
Das war von Anfang an so. Ich habe eine Funktion zur Formatierung der Zahl (Anzahl Nachkommastellen) verwendet, welche einen String zurückgibt. Bei Bedarf müsste man das wieder eine Zahl umwandeln. Ich gehe aber davon aus, dass es keine Probleme macht.

NCIceWolf, 2023-03-24:
Ok, dann ist das ja kein Problem der aktuellen WR-Software.
Ggf. kann das ja in einem künftigen Release angepasst werden, scheint aber aktuell nicht notwendig zu sein.
Vielen Dank für das Feedback!

aruttkamp, 2023-03-26:
@NCIceWolf - ist der o.g. Wert (rct.0.g_sync.p_ac_sum_lp) der einzige wo das bei Dir auftritt ? 
Nachfrage : Was würdest Du davon halten , wenn man die einzelnen Parameter in der Config "anhaken" könnte ?

NCIceWolf, 2023-03-27:
> @NCIceWolf - ist der o.g. Wert (rct.0.g_sync.p_ac_sum_lp) der einzige wo das bei Dir auftritt ? Nachfrage : Was würdest Du davon halten , wenn man die einzelnen Parameter in der Config "anhaken" könnte ?

Wie beschrieben bekomme ich diesen Log-Eintrag für alle Werte die aus dem WR ausgelesen werden (alle als "info"). Macht ja aber Sinn wenn @lauff die Zahlen durch eine Funktion in einen String konvertiert.
Wenn man das also auflösen will, müsste aus meiner Sicht diese Funktion zur Formatierung der Nachkommastellen einfach auch wieder eine Zahl zurückschreiben.
Dann wäre sicherlich auch das Log wieder happy :)

Zu den Parametern: Du meinst welche Werte aus dem WR ausgelesen werden? Ich habe das bislang einfach in der Instanz entsprechend eingetragen. Ich meine aber da gab es einen Feature-Request zu anderen/neuen Werten. Das wäre natürlich schön die einfach anklicken zu können, anstatt die Wertenamen raussuchen zu müssen.
Wobei einige Werte ja nur sporadisch aktualisiert werden (z.B. energy.e_grid_feed_year_sum), ist also die Frage ob es überhaupt lohnt den abzufragen.
Ist aber vielleicht besser das Thema in einem separaten Feature-Request zu verfolgen.

timoglaess, 2023-04-20:
Ich hab alle Datenpunkte händisch von "number" auf "mixed", also gemischt umgestellt. Seitdem läuft alles und es kommt kein Log Eintrag mehr. Alles gut.

Madaraxa, 2023-05-09:
Zur Weiterverarbeitung und der Sauberkeit halber wäre es natürlich schön, wenn der Datentyp 'stimmen' würde. Ich weiß nicht, wie viel Arbeit das macht, daher hat es natürlich keine Priorität :)

aruttkamp, 2023-05-09:
> Zur Weiterverarbeitung und der Sauberkeit halber wäre es natürlich schön, wenn der Datentyp 'stimmen' würde. Ich weiß nicht, wie viel Arbeit das macht, daher hat es natürlich keine Priorität :)
> 
> `rct.0	2023-05-09 07:47:12.721	info	State value to set for "rct.0.power_mng.u_acc_mix_lp" has to be type "number" but received type "string"
> rct.0	2023-05-09 07:47:12.720	info	State value to set for "rct.0.power_mng.is_heiphoss" has to be type "number" but received type "string"
> rct.0	2023-05-09 07:42:52.617	info	State value to set for "rct.0.prim_sm.state" has to be type "number" but received type "string"
> rct.0	2023-05-09 07:41:42.644	info	State value to set for "rct.0.g_sync.p_ac_sum_lp" has to be type "number" but received type "string"
> rct.0	2023-05-09 07:31:42.520	info	State value to set for "rct.0.g_sync.p_ac_load_sum_lp" has to be type "number" but received type "string"
> rct.0	2023-05-09 07:31:12.508	info	State value to set for "rct.0.fault.2_flt" has to be type "number" but received type "string"
> rct.0	2023-05-09 07:30:42.459	info	State value to set for "rct.0.battery.soc_target_high" has to be type "number" but received type "string"`

aruttkamp, 2023-05-09:
@Madaraxa  - Das sehe ich genau so. Es wird eine entsprechende Version kommen. ich kann nur aktuell nicht sagen wann.
Aug/Sept, scheint mir aktuell realistisch.

aruttkamp, 2023-08-14:
Das hier beschriebene Problem sollte mit der v1.1.0 auch behoben sein ! 
Freue mich auf Rückmeldungen.

closed 2023-08-18 by aruttkamp