aruttkamp/ioBroker.rct #17: Nach Update auf 1.0.3 wird Version weiterhin 1.0.1 gezeigt?

Madaraxa, 2023-04-01:
Wenn ich über den iobroker den Adapter (derzeit 1.0.1) update, scheint er auch ein Update durchzuführen, zeigt dann aber (wieder) an, dass ich auf 1.0.1 bin. 
Entweder scheint also beim Update was falsch zu laufen ohne, dass dies im log angezeigt wird, oder die angezeigte Version von 1.0.3 ist trotzdem 1.0.1
`0
0
Log-Größe: 3.5 KB

1
Zeit
info
Nachricht

admin.0
2023-04-01 10:47:04.085	info	<== Disconnect system.user.admin from ::ffff:192.168.9.84

rct.0
2023-04-01 10:47:04.048	info	State value to set for "rct.0.power_mng.is_grid" has to be type "number" but received type "string"

rct.0
2023-04-01 10:47:34.044	info	State value to set for "rct.0.dc_conv.dc_conv_struct_1_p_dc_lp" has to be type "number" but received type "string"
host.iobroker
2023-04-01 10:47:20.402	info	iobroker exit 0
host.iobroker
2023-04-01 10:47:19.326	info	iobroker upload [0] rct.admin /opt/iobroker/node_modules/iobroker.rct/admin/words.js words.js application/javascript
host.iobroker
2023-04-01 10:47:19.312	info	iobroker upload [1] rct.admin /opt/iobroker/node_modules/iobroker.rct/admin/style.css style.css text/css
host.iobroker
2023-04-01 10:47:19.298	info	iobroker upload [2] rct.admin /opt/iobroker/node_modules/iobroker.rct/admin/rct.png rct.png image/png
host.iobroker
2023-04-01 10:47:19.279	info	iobroker upload [3] rct.admin /opt/iobroker/node_modules/iobroker.rct/admin/rct-logo.square.png rct-logo.square.png image/png
host.iobroker
2023-04-01 10:47:19.261	info	iobroker upload [4] rct.admin /opt/iobroker/node_modules/iobroker.rct/admin/index_m.html index_m.html text/html
host.iobroker
2023-04-01 10:47:19.168	info	iobroker upload [5] rct.admin /opt/iobroker/node_modules/iobroker.rct/admin/admin.d.ts admin.d.ts video/mp2t
host.iobroker
2023-04-01 10:47:19.019	info	iobroker up to date in 15s159 packages are looking for funding run `npm fund` for details
host.iobroker
2023-04-01 10:47:19.014	info	iobroker

admin.0
2023-04-01 10:47:04.085	info	<== Disconnect system.user.admin from ::ffff:192.168.9.84

rct.0
2023-04-01 10:47:04.048	info	State value to set for "rct.0.power_mng.is_grid" has to be type "number" but received type "string"

admin.0
2023-04-01 10:47:04.085	info	<== Disconnect system.user.admin from ::ffff:192.168.9.84

rct.0
2023-04-01 10:47:04.048	info	State value to set for "rct.0.power_mng.is_grid" has to be type "number" but received type "string"

admin.0
2023-04-01 10:47:04.085	info	<== Disconnect system.user.admin from ::ffff:192.168.9.84

rct.0
2023-04-01 10:47:04.048	info	State value to set for "rct.0.power_mng.is_grid" has to be type "number" but received type "string"

admin.0
2023-04-01 10:47:04.085	info	<== Disconnect system.user.admin from ::ffff:192.168.9.84

rct.0
2023-04-01 10:47:04.048	info	State value to set for "rct.0.power_mng.is_grid" has to be type "number" but received type "string"
host.iobroker
2023-04-01 10:47:02.741	info	iobroker Installing iobroker.rct@1.0.3... (System call)
host.iobroker
2023-04-01 10:47:02.739	info	iobroker NPM version: 9.5.0
host.iobroker
2023-04-01 10:47:01.105	info	iobroker Update rct from @1.0.1 to @1.0.3
host.iobroker
2023-04-01 10:46:57.512	info	iobroker upgrade rct`

aruttkamp, 2023-04-01:
Vielen Dank - ist mir auch eben aufgefallen. Ich bin mit den Versionsnummern ein wenig "durcheinander" geraten. Es sollte nur die falsche Version angezeigt werden ! Funktionalität ist tatsächlich die der 1.0.1 - Ich habe nur Frameworkupdates eingearbeitet.

aruttkamp, 2023-07-30:
Auch das hier ist (endlich) erledigt. Nach viel hin und her ist die Version 1.0.5 jetzt veröffentlicht !

closed 2023-07-30 by aruttkamp