aruttkamp/ioBroker.rct #18: rct.0.power_mng.bat_next_calib_date als Datum anzeigen??

timoglaess, 2023-04-20:
Hallo zusammen, 

der Datenpunkt "**rct.0.power_mng.bat_next_calib_date**" bringt eine Zahl zb. "_1,6007828702706314e+22_" zurück. Alle meine Versuche diese in einem lesbaren Datum anzeigen zu waren erfolglos. Ich nehme an, dass es Sekunden ab einem bestimmten Datum (zb. 1.1.1900 oder 1904), aber das ergibt mit den Werten irgendwie keinen Sinn.

Ich würde gerne das aktuelle Datum davon abziehen, um z.b. "verbleibende Tage" bis zur nächsten Batteriekalibrierung anzeigen und in Skripten verwenden zu können. Oder einfach das Datum, das in dem Zeitstempel steckt anzeigen. Zeitstempel und Datumsformat Funktionen in Blockly kommt damit jedenfalls nicht klar.

Jemand ne Idee, um was für ein Format es sich handeln könnte?

NCIceWolf, 2023-04-22:
Hallo @timoglaess 

ich denke Mal dass es sich dabei um die Unix Zeit handelt.
Siehe hier:
https://de.wikipedia.org/wiki/Unixzeit?wprov=sfla1

Das sollte sich sehr einfach umrechnen lassen, da der POSIX-Standard doch Recht verbreitet ist.
Die Idee ist nicht so verkehrt, zumindest die Anzeige wann die nächste Kalibrierung kommt wäre ja doch ganz praktisch.

VG
Matthias

aruttkamp, 2023-04-22:
Davon gehe ich auch aus. Schaue nach wenn aus dem Urlaub zurück. Also nächste Woche.

aruttkamp, 2023-04-24:
> der Datenpunkt "**rct.0.power_mng.bat_next_calib_date**" bringt eine Zahl zb. "_1,6007828702706314e+22_" zurück.

@timoglaess - kannst du mir bitte den exakten Wert zur Verfügung stellen ? Mir fehlt leider noch die Hardware ( Akku ) , daher kann ich die Akkuwerte nicht live abfragen.

NCIceWolf, 2023-04-24:
Habe den Wert aus meinem WR ausgelesen, da stimmt aber was mit dem Format, bzw. dem decode nicht.

Aktueller Wert aus dem RCT WR:
rct.0.power_mng.bat_next_calib_date
1.7175161726120747e+22

Soll-Wert (ca.): 
1684622648

Folgendes sagt die Doku dazu:

OID
0xB6623608

Request Type
UINT32

Response Type
UINT32

Name
power_mng.bat_next_calib_date

Description
Next battery calibration

aruttkamp, 2023-08-02:
Grundsätzlich kann ich bestätigen, dass hier etwas nicht stimmt. der im Datenpunkt hinterlegte Wert ist nicht der richtige. 
Hier wird die Rückgabe vom WR nicht korrekt interpretiert.

aruttkamp, 2023-08-05:
Das Problem wurde gefunden und wird mit der kommenden Version gefixed. Es ist noch einiges an Tests notwendig. Deshalb dauert es noch ein bischen.

aruttkamp, 2023-08-14:
Ab v1.1.0 wird der Wert als Linux Wert ausgegeben. 
Der Bug ist damit behoben

timoglaess, 2023-08-14:
Cool. Werd ich gleich testen wenn wir vom Urlaub zurück sind.
Vielen Dank Andreas fürs fixen!

Gruss Timo

—
Timo Gläss
Waldhornstr. 17
82110 Germering
0163/6984909

aruttkamp, 2023-08-14:
Na, erstmal schönen Urlaub !
Bin auf dein Feedback gespannt !

OnkelFD, 2023-08-15:
Hallo,
ich habe mir die Zahl welche unter power_mng.bat_next_calib_date ausgegeben wird genommen und versucht sie umzuwandeln, es kommt leider nicht das Richtige Datum dabei heraus es müsste eigentlich der "08.09.2023 13:00:00" sein. 

date -u -d @3491429220
Wed Aug 21 01:27:00 AM UTC 2080

Gruß

NCIceWolf, 2023-08-22:
Moin,

hab leider ebenfalls in der aktuellen Version das Problem dass der Wert hier noch keinen Sinn ergibt.
Der Adapter meldet mir diesen hier: _3242261604_
Der Timestamp 3242261604 entspricht dem 28.09.**2072** um 06:13:24 Uhr.
(gilt für UTC+2)

Die nächste Kalibrierung ist laut RCT App der 09.09.2023 um 09:54:09 Uhr und das wäre:
Der 09.09.2023 um 09:54:09 Uhr entspricht dem Timestamp _1694246049_.
(gilt für UTC+2)

Irgendwo ist also zumindest bei diesem Wert hier noch der Wurm drin.

timoglaess, 2023-08-30:
Bei mir wird der Wert leider auch nicht korrekt angezeigt: **3503624805** (gibt den 09.01.2081)
Soll wäre 26.09.2023 13Uhr.
In der RCT App (nur Android/ in der iOS App nicht) findet sich das Datum unter Gerät/ Batterie/ Kalibrierung:
[image]

aruttkamp, 2023-08-30:
Hey ! cool. Da wird mir der 06.02.2022 13:00:00 angezeigt. 
und das ohne das eine Batterie da wäre *ggg*
Hab das Problem schon eingekreist. Aber gleich wird sich der WR abschalten und dann gehts nicht weiter.
Ab Freitag dann hoffentlich mit Speicher ;-)

aruttkamp, 2023-09-13:
So wie es aussieht ist auch hier der Datentype verkehrt gewesen.
Zumindest bekomme ich damit die korrekten Werte angezeigt. 
In Version 1.2.2 habe ich das jetzt angepasst. 
Bitte um Feedback !

timoglaess, 2023-09-13:
Jaaa! Jetzt passt der Wert.
Supi, danke für die ganze Mühe!

Bug finally SOLVED!

closed 2023-09-13 by timoglaess

aruttkamp, 2023-09-13:
Boah... bist du schnell ! Die Version ist erst 8 min. online .... *ggg*

OnkelFD, 2023-09-13:
Ja passt auch bei mir 😘