aruttkamp/ioBroker.rct #20: Feature request: Write soc_target value

philoxio, 2023-05-03:
I would find it very useful if you could incorporate the possibility of writing soc_target values.

There is a post in a pv board that describes a possibilty of how this can be done:

https://www.photovoltaikforum.com/thread/159603-rct-power-storage-soc-zielauswahl-extern-nutzen/?postID=2867630#post2867630

I´m here to assist testing if needed!

Thanks!

KR,

philoxio

kbb444, 2023-05-03:
Es gibt als mögliche Referenz auch eine Implementierung in einem fhem-Modul:
https://forum.fhem.de/index.php/topic,126126.45.html
@aruttkamp: Ich fände es auch super, wenn Du Dich des Themas annehmen würdest. 
Use case wäre insbesondere das Schreiben von "Min SOC Zielwert"="aktueller Soc", damit das Auto in einem Zeitraum geringer Strompreise geladen werden kann, ohne dabei den RCT-Speicher zu entleeren. Im Anschluss dann wieder auf die üblichen 0,07 zurücksetzen. Das ist der einzig mir bekannte Weg einen Netzbezug zu erzwingen.

aruttkamp, 2023-05-04:
Moin ! 
na, ihr habt Ideen. Grundsätzlich kann man das - so es denn funktioniert - implementieren. 
Ich muss aber zugeben , dass ich mir über die möglichen Konsequenzen erst klar werden muss. 
Ich mag ungern etwas implementieren , was dann jedem Laien es ermöglicht "Scheiss" zu bauen. Schreibender Zugriff auf ein System dessen inneres man nicht wirklich kennt halte ich persönlich für kritisch. Ich muss mich damit beschäftigen. Das wird allerdings nicht kurzfristig erfolgen !

philoxio, 2023-05-04:
Hi @aruttkamp! 
Danke, dass du dich der Sache annehmen willst. Für so kritisch halte ich dies nicht, da ja heute schon der Wert über die App geschrieben werden kann (sofern man den installer Zugriff aktiviert hat).

Dirk-I, 2023-06-01:
Hallo zusammen, ich finde die Möglichkeit auch sehr gut. Ein weiterer UseCase (den ich oft per Hand setze über die App) in den Schlechtwetter-Monaten setze ich den min. SOC auf 20% oder 25% damit ich eine Notfall (Blackout) Reserve habe. Hierbei denke ich, es ist sonst eher Sinnfrei wenn in den Wintermonaten dann doch mal ein Stromausfall ist und das Haus keine Reserve hat. Über die PV kommt dann auch nicht genügend um wenigstens die Rollos hoch zu machen.

aruttkamp, 2023-08-19:
Ich gehe dies an, sobald ich die passende Hardware habe und auch entsprechende Werte von der Anlage auslesen kann. 
"Angeblich" kommt die Hardware noch diesen Monat....

aruttkamp, 2023-09-02:
Die passende Hardware ist tatsächlich seit gestern installiert. 
Leider ist Notstrom noch nicht möglich. Hab noch nicht ganz verstanden warum, aber es liegt wohl an der Verkabelung des Hauses .. zumindest an 4 Stromkreisen. 

Die o.g. Anwendungsfälle leuchten mir ein. Ich denke auch das es sinnvoll ist d Werte automatisiert ändern zu können. 
Das "Risiko" ist nach meiner Einschätzung eher minimal. Mal sehen wie wir das sinnvoll beschrieben bekommen.

philoxio, 2023-09-24:
Das klingt gut! Ich freue mich! Sag Bescheid wenn du jemanden zum Testen brauchst! :)

kschaumann, 2023-10-11:
@aruttkamp mein Speicher und mein Power Switch wurden gestern installiert. Der Techniker sagte mir nochmal, dass ich die "Kalibrierung" der Batterien die ca. alle 30 Tage automatisch erfolgt nicht übersteuern sollte. Das macht das System auch über die SoC-Targets mit dann 0<->100 um zu prüfen wie viel wirklich noch möglich ist. 

GGf. verwirrt man dieses System dann wenn man irgendwie anders diese Werte ändert. Leider hab ich auch noch keinen Wert gefunden, der aussagt, dass die Kalibrierung gerade "in Gange" ist. Ansätze anderswo sind eben auszuwerten, welche SoC-Targets gesetzt sind. Was aber keinen Sinn macht, wenn man die wie hier geplant durch ein Script oder so anpasst. 

In der App steht das deutlich erkennbar: "(calib.)" im Batterie-Symbol!

Unabhängig davon bin ich bei @Dirk-I mit der "Notfall-Reserve" und dem Wunsch das Auto nicht über den Speicher zu laden, das wäre ja Quatsch!

philoxio, 2024-01-07:
Hi @aruttkamp!
Läuft dein Anlage mittlerweile? Mit Notstrom? :)
Hast du schon mal Zeit gehabt dich mit der Umsetzung zu beschäftigen? In Kombination mit einem variablen Stromtarif/Wallbox könnte das echt ein spannender Use-Case sein. Eventuell kann man ja die Batteriekalibrierung nutzen um ein Laden aus dem Netz zu "erzwingen"?

aruttkamp, 2024-01-07:
Hallo @philoxio 
>Läuft dein Anlage mittlerweile? Mit Notstrom? :)
Danke der Nachfrage - nein. Tut sie nicht. Das wird jetzt wohl zum Anwalt gehen. 
Aber das ist ja für den Use Case hier eigentlich nicht wichtig. 

Der Dezember war von ein paar anderen "Projekten" geprägt. Aber jetzt habe ich gerade gestern beschlossen das Ding hier anzugehen. Direkt nach dem Problem mit der Variablen. 
Was mir allerdings aufgefallen ist ist die Tatsache , dass die Kalibrierung deutlich öfter als einmal im Monat stattfindet. Das beschäftigt mich im Moment ein wenig. Der zieht sich völlig "unmotiviert" den Strom aus dem Netz und läd den Akku bis 100% auf. Ich verstehe nur nicht warum er das tut.

philoxio, 2024-01-07:
Das klingt top! :)
Das "unmotivierte" Kalibrieren der Batterie habe ich auch beobachtet. 
Eine Antwort von RCT dazu habe ich hier https://community.symcon.de/t/modul-rct-power-inverter/53056/112 gefunden.
Sag mit gerne Bescheid wenn ich was testen soll!

NCIceWolf, 2024-01-08:
Ist bei mir auch so.
Im Sommer merkt man das nicht so, da er ja immer wieder vom PV-Strom voll wird, momentan ist da leider nicht viel zu holen...

Ist also normal und ja inzwischen auch erkennbar (abseits davon dass der Ziel-SOC auf 100% steht):

The object "battery.bat_status" indicates the status of a connected battery:

    0 -> charge/discharge (normal operation)
    8 -> charge (calibration)
    1024 -> discharge (calibration)
    --> 2048 -> balancing <---

VG
-- 
Gesendet mit pEp für Android.

phinotv, 2024-01-08:
Ich wäre auch sehr interessiert.
Ich habe gleich 2 Fälle, bei denen es Interessant ist.
Ich habe seit kurzem Tibber als Stromanbieter. Dieser rechnet stündlich ab. Somit könnte ich im Winter bei günstigen Preise die Batterie füllen und dann wieder verbrauchen. Bei Schwankungen von 10 cent (23 zu 33 cent) kann man die Ladeverluste verkraften.
Hier hat jemand Tibber schon in FEHM eingebunden:  https://www.juergenstechnikwelt.de/smarthome-2/tibber-in-fhem-einbinden-beste-zeiten-fuer-niedrige-strompreise-ermitteln/
Zum Anderen das oben angesprochene Laden des Autos, ohne die Batterie zu leeren. Dafür habe ich zur Zeit eine App zur Steuerung der Wallbox, die auch mit dem Heimspeicher von RCT umgehen kann. Vielleicht könnte man dort sich Anleihen vom Programmierer holen. Christian ist sehr aufgeschlossen. https://android.chk.digital/de/ecar-charger-control/

AllElectrified, 2024-05-08:
Auch ich wäre sehr daran interessiert wenigstens den SOC Wert schreiben zu können (ohne Flash, nur dynamisch ändern). Anwendungsfälle in Zusammenhang mit Tibber: in der Nacht mit güstigem Windstrom Auto Laden oder Wäremepumpe betreiben und den Akkuspeicher für die teure Zeit am Vormittag aufsparen. usw. Wäre toll das direkt im Iobroker machen zu können und nicht noch zusätzlich FHEM installieren zu müssen .

petermeter2000, 2024-08-27:
Sehr cool der Adapter. Ist ohne Probleme im Einsatz :-)
Aber auch ich würde mich freuen, wenn die gewünschten Features implementiert würden - ich denke davon würden viele profitieren. Und wenn man flexible Stromtarife nutzen will (Tibber etc.) kommt man sonst nicht weiter.
Gibt es hier aktuell noch Entwicklung?

Folgende Szenarien könnte man damit abdecken - siehe auch https://github.com/do-gooder/rctpower_writesupport

**1. Batterie laden erzwingen zu günstigen Zeiten. Dafür bräuchte man folgende Einstellungen:**
- power_mng.is_grid : true (Externe Leistung zulassen)
- power_mng.battery_power_extern : Ladewert in Watt (negativ zum Laden)
- power_mng.soc_strategy: 2 (Extern)

**2. Batterieentladung hemmen, für sehr starkes Laden von z.B. eAuto, dafür reicht meiner Meinung nach**
- power_mng.soc_strategy: 0 (SOC target = SOC)

**3. Notfallreserve der Batterie**
- CE266F0F power_mng.soc_min (Min SOC target)
- BD3A23C3 power_mng.soc_charge (SOC min maintenance charge)

Per Pythonscript funktioniert das testweise alles prima.
Aber eine Integration in Euren Adapter würde natürlich alles sehr viel einfacher machen...
Auf jeden Fall an dieser Stelle herzlichen Dank für Eure Arbeit!!!

petermeter2000, 2024-08-29:
Hab bei 1. Batterie laden erzwingen... einen Fehler drin - muss heissen:
power_mng.use_grid_power_enable : true (Externe Leistung zulassen)

NCIceWolf, 2024-09-11:
Da ich seit dem 01.09. auch bei Tibber bin, wird das Thema auch für mich interessanter :)
Die Szenarien hat do-gooder ja schon sehr gut beschrieben, das wären auch erst einmal die 3 die angegangen werden sollten.
Später kann man das sicherlich noch weitere Ideen einbringen.

@aruttkamp: Zu beachten wäre dass wir ja keine beständige Verbindung zum WR (mehr) haben, die Kommandos müssten also entweder beim nächsten Aufbau verschickt werden, oder wir brauchen einen separaten Verbindungsaufbau. Das würde ich aber eher vermeiden wollen, sonst kollidiert das im dümmsten Fall.

Dazu sollte das optional sein, also im Adapter-Menü ein neuer Reiter eingebaut werden, über den die Funktion eingeschaltet und dann konfiguriert werden kann (habe das ja schon halb vorbereitet).

Bei der Implementierung kann ich allerdings nur wenig helfen, da ich gerade beruflich sehr stark eingebunden bin...
Für Tests bin ich natürlich zu haben!

aruttkamp, 2024-09-11:
Moin ! 
Auch bei mir war privat und beruflich ziemlich viel los die letzten Monate. Ganz langsam wird es besser und ich kann darüber nachdenken dieses Feature anzugehen. Da dies ein "Spassprojekt" ist muss es allerdings noch auf den Abschluss einer anderen Sache warten die zu meinem Lebensunterhalt beiträgt. 
Also ... wir machen das , aber es wird dauern. Ich denke der November ist ein realistischer Monat. 
Ich schreib mal extra nicht dazu welches Jahr :-P

petermeter2000, 2024-09-12:
Das klingt vielversprechend! Dann hoffe ich auf Verdopplung der Urlaubstage bei gleichzeitiger Lohnerhöhung.
Ich hatte übrigens vor ein paar Wochen bei RCT eine Anfrage bzgl. der Umsetzung von den 3 Punkten gestellt.
Bislang keine Antwort...
Wäre ja auch schön, wenn die Features "offiziell" von RCT angeboten würden.

NCIceWolf, 2024-11-03:
Ein Thema was mir aufgefallen ist als ich jetzt mal manuell den Min-Soc über die RCT-App verändert habe:
Das dauert ganz schön lange...

@aruttkamp: hier muss also bei Änderungen entweder ein timeout rein, oder man kann das über den Inverter-State auslesen. Will nicht wissen was passiert wenn der WR neu beschrieben wird wenn die letzten Änderungen noch nicht durch sind oder da generell zu oft Änderungen rüber gehen.

aruttkamp, 2024-12-07:
Moin ! 

nachdem ich mich nun mit der Materie beschäftigt habe muss ich euch leider mitteilen dass ich den Schreibzugriff nicht in den Adapter implementieren werde. 
Die Rechtslage ist hier recht eindeutig. Wenn RCT mit deren APP was versaut sind diese auch haftbar. 
Wenn ich etwas versaue bin ich es auch ! Und das kann und will ich mir nicht leisten. 
Wir haben es hier mit keiner offiziellen API zu tun die wir nutzen um zuzugreifen. Damit wäre ich voll haftbar. 

Das Skript welches hier von einem User zur Verfügung gestellt wurde funktioniert und liefert das gewünschte Ergebnis so wie ich das interpretiere. Nutzt das wenn ihr dies möchtet. 

Sollten sich die Vorraussetzungen ändern können wir gerne nochmals drüber reden. 

Wünsche Euch allen trotzdem schöne Weihnachten. 

Andreas

NCIceWolf, 2024-12-09:
Ist absolut verständlich.
Hatte schon an Formulierungen zum Haftungsausschluss überlegt, wäre aber mindestens schwierig.

Wäre natürlich sehr praktisch wenn RCT so etwas in Zukunft nativ in deren App bringen würde. Dann wäre auch eine Integration hier wieder eine Option. Vielleicht hilft es wenn da jeder betroffene einen Request an RCT schickt.
Gibt ja inzwischen auch den ein oder anderen Hersteller der es vormacht. Eine bessere Steuerung wird künftig absolut unabdingbar sein.

Bis dahin habe ich mir auch mit dem Python-Skript von Do-Gooder beholfen.
Zur Sicherheit deaktiviere ich jedoch den Adapter vor Schreibaktionen und aktiviere ihn danach wieder (einfach per blockly). Weiß nicht ob ich herausfinden will was passiert wenn die beiden kollidieren.
Zumindest der Lesevorgang schlägt in jedem Fall fehl, da eben auch die Daten vom Adapter mit ankommen und das Skript das Ergebnis dann nicht interpretieren kann. Das hatte ich schon getestet.

Ebenfalls frohe Weihnachten an alle 🎄 🎅

petermeter2000, 2024-12-09:
RCT arbeitet anscheinend daran:
https://play.google.com/store/apps/details?id=com.solarmanager.solarmanagerenduser.rct

Ist halt wieder cloudbasiert und anscheinend als Bezahldienst gedacht :-(

NCIceWolf, 2024-12-09:
Das scheint mir eher was lokales zu sein.
Also eine zusätzliche Box, die quasi das tun soll was wir bereits mit dem ioBroker tun:

Der RCT Power Manager besteht aus dem Gateway,
der Web Plattform und der Power Manager App.

Der Power Manager ist ein für den Privatgebrauch bestimmtes HEMS, das für die Innenrauminstallation geeignet ist. Mit der Einbindung in das Heimnetzwerk steuert und überwacht dieses digitale System die Energieerzeugung, -speicherung und -nutzung im Haushalt.
Kompatible Geräte, die in dasselbe Ethernet-Netzwerk eingebunden sind, können gesteuert werden.

Jetzt macht das natürlich noch viel mehr Sinn dass die "Extern"-Steuerungsmöglichkeit vor einiger Zeit entfernt wurde.
Da hatte RCT bereits eine eigene Box in Planung 😄

closed 2024-12-15 by aruttkamp