aruttkamp/ioBroker.rct #23: Hausverbrauch nicht auszuwerten ..

jonaslinhose, 2023-07-05:
Hallo,

vielleicht stehe ich auf dem Schlauch, aber ich versuche den Hausverbrauch auszulesen. Leider finde ich keinen passenden Wert, der die entsprechende Werte zurück gibt.
Ich habe schon die main.js bzw. rct_core.js durchgesehen und einige andere nützliche Werte gefunden, aber leider nichts passendes für den Hausverbrauch. Hat hier jmd. einen Tip für mich?

Viele Dank

Jonas

NCIceWolf, 2023-07-06:
Moin!

Hab ich per Zufall gerade gestern auf mein Dashboard gepackt und nehme dafür diesen hier:
rct.0.g_sync.p_ac_load_sum_lp

Bitte um Rückmeldung wenn das passt.

VG
Matthias

aruttkamp, 2023-07-06:
Moin ! 

Ich kann aktuell nicht helfen, da meine Anlage immer noch nicht vollständig ist und ich den Hausverbrauch nicht von RCT bekomme. Ich hoffe das hat im August ein Ende ;-) 
Rein von der Logik her müsste der Wert von NCIceWolf korrekt sein , also rct.0.g_sync.p_ac_load_sum_lp
Schau mal was der dir ausgibt. 
Ich ermittel mir das aktuell aus dem Wert der Erzeugung - Einspeisung ;-) 
Nicht so genau , aber besser wie nix.

jonaslinhose, 2023-07-08:
Hallo,

also bei mir zeigt p_ac_load_sum_lp aktuell 4584W an, die RCT - Power App aktuell zwischen 150W und 250W (wir sind nicht zuhause), dass passt also zumindest bei mir nicht. Es könnte ungefähr der Tagesverbrauch sein (18 * 250W).

Ich habe den aktuellen Hausverbrauch auch nur rechnerisch ermitteln können:
    AC Leistung meiner beiden Strings (rct.0.g_sync.p_ac_sum_lp) -- Die Ladung in den RCT-Akku ist da meiner Meinung nach nicht mit dabei --
+ AltePV-Anlage als externer Generator (rct.0.io_board.s0_external_power) 
+ Netzeinspeisung (rct.0.g_sync.p_ac_grid_sum_lp) 
+ Wallbox-Ladeleistung 

Falls jmd noch eine bessere Idee hat gerne. Ich suche auch noch eine Möglichkeit den Hausverbrauch (täglich, wöchentlich, mtl. jährlich) zu ermitteln, dass Finanzamt wäre begeistert.

NCIceWolf, 2023-07-09:
[image]

Ich würde da noch einmal drauf schauen, bei mir passt das. Zuerst der "Standby-Verbrauch", dann würden zwei Herdplatten angeschaltet.

Dabei zu beachten: der Wert wird ja je nach Einstellung aus dem Wechselrichter ausgelesen (bei mir alle 10 Sekunden).
Das RCT Power Portal ist wesentlich langsamer (da in die Cloud synchronisiert), die RCT Power App wiederum ist schneller, da lokal.

Könnte aber noch ein Unterschied bei der Konfiguration  sein, habe den RCT WR 10 und eine Batterie. Vielleicht unterscheiden sich die Werte wenn keine Batterie vorhanden ist?

jonaslinhose, 2023-07-09:
Hallo,

es ist seltsam, jetzt scheinen sich auch bei mir die Werte entsprechend zu aktualisieren und passen auch mit leichter Verzögerung zu der RCP PowerApp.  Es ist also wirklich der p_ac_load_sum_lp , was mir auch logisch erscheint. Danke für die Hinweise / Hilfe.

NCIceWolf, 2023-07-09:
Na klar, sehr gerne!
Das Thema der täglichen/wöchentlichen Werte würde ich Mal hiermit testen, aber ohne Gewähr:

energy.e_load_day_sum

Sollte es auch für week und month geben.

closed 2023-07-09 by NCIceWolf

Corneys, 2023-07-23:
Hallo, 

ich habe auch eine Frage zu dem Hausverbrauch.
Für mich sieht es so aus, als sei rct.0.g_sync.p_ac_load_sum_lp ein komplett anderer Wert als der tatsächliche aktuelle Hausverbrauch. 

Wenn ich mir in der RCT Power App den Verbrauch anschaue, welcher in der Übersicht als das Haus angezeigt wird, stehen hier zur Zeit ca. 200W (für mich ein realistischer Wert, welcher sich auch z.B. beim Kochen nach oben bewegt)
Diese 200 W bekomme ich aber leider nirgends ausgelesen

Tippe ich in der App auf das Haus wird ganz oben der Wert "Haushaltslast - externe Erzeuger [W]" angezeigt, dies ist der Wert, welchen ich über rct.0.g_sync.p_ac_load_sum_lp ausgegeben bekomme.
Dieser Wert liegt aber bei -900, was ich mir nicht so richtig erklären kann.

Vielleicht hat da jemand von euch ja eine Idee.

Vielen Dank
Felix

camast, 2023-07-23:
In meiner Anlage (PS6) ohne "externe Erzeuger" stimmen alle 3 Werte (rct.0.g_sync.p_ac_load_sum_lp, App, "Hauslast - Externe Erzeuger") in Betrag und Vorzeichen überein. Ich habe zu keinem Zeitpunkt unplausible Werte festgestellt.
Hast Du externe Erzeuger konfiguriert bzw. angeschlossen?

Corneys, 2023-07-23:
In der Anlage sind zwei Wechselrichter als Master und Slave integriert. 
Der Slave wird in der Übersicht als "G" angezeigt. 

Auf dem Hinweis mit der externen Erzeugung habe ich noch mal etwas herumgerechnet.
Wenn ich den Wert welcher unter G angezeigt wird (rct.0.io_board.s0_external_power) von dem Wert welcher mir als "Haushaltslast - externe Erzeuger [W]" (rct.0.g_sync.p_ac_load_sum_lp) abziehe ergibt das den Wert welcher unter dem Haus als tatsächlicher Verbrauch dargestellt wird. 

So könnte ich mir auf jeden Fall mit einem Scrip behelfen, falls es keinen direkten Datenpunkt für mich gibt.

[image]