aruttkamp/ioBroker.rct #28: Status Batterie

OnkelFD, 2023-08-12:
Ist es irgendwie möglich den den Status der Batterie welche auch in der App (auf der Übersichtsseite) angezeigt wird auszulesen?
Sprich charge / discharge / idel / charge (calib.) / discharge (calib.)

aruttkamp, 2023-08-14:
Moin ! 
Schau dir mal die Version 1.1.0 des Adapters an. Es sind dort viele weitere Parameter implementiert. 
rct_Core2.js enthält alle unterstützten Parameter. 
Ich habe aktuell keine Batterie, weshalb ich dies nicht ausprobieren kann.

OnkelFD, 2023-08-14:
Okay danke ich werde es mir mal anschauen und wenn ich einen passenden Parameter gefunden habe werde ich mich melden.
Danke übrigens an alle die an dem Adapter weiter werkeln :-)

closed 2023-08-18 by aruttkamp