aruttkamp/ioBroker.rct #29: battery.soc Einheit verschoben

OnkelFD, 2023-08-14:
Hi,
ich habe heute auf die v1.1.0 aktualisiert nun zeigt er den battery.soc Status allerdings um zwei Stellen verschoben an anstatt 71% zeit er nur 0,7 an wodurch die nur noch mit 10% Schritten gearbeitet werden kann was das ganze ein wenig ungenau macht.

aruttkamp, 2023-08-15:
Das schaue ich mir an. 
Rein der Logik folgend müsste es dann diese 3 Werte betreffen. 

battery.soc
battery.soh
battery.soc_target

Kannst Du das bestätigen ? Hab leider (noch) keine Battery und kann es nicht testen.

NCIceWolf, 2023-08-15:
Moin!

Scheint mir alle Werte der Batterie zu betreffen.
Dazu hab ich noch seltsame Einträge im Log gefunden.
Da scheint noch was mit den "states" und "names" nicht zu passen.

VG
-- 
Gesendet mit pEp für Android.

aruttkamp, 2023-08-15:
Moin ! 
Was soll ich denn damit anfangen ? 
>>Dazu hab ich noch seltsame Einträge im Log gefunden.
Bitte mal Logs zur Verfügung stellen. Ohne kann ich nichts mit der Aussage anfangen.

aruttkamp, 2023-08-15:
Idee:
Es ist möglich dass sich Einheiten verschoben haben. Bitte löscht doch mal die Werte die falsch angezeigt werden in den Objekten. 
gerne auch den ganzen RCT Baum. 
Adapter einmal neu starten, dann sind alle wieder da. Stimmen die Werte dann immer noch nicht ? 
@NCIceWolf @OnkelFD

NCIceWolf, 2023-08-15:
Ich hatte den Baum gestern noch komplett gelöscht, die Werte waren aber nach wie vor verschoben.
Da einige meiner Skripte dadurch Amok gelaufen sind und der Adapter wiederholt abgestürzt ist, musste ich gestern Abend eine Sicherung zurückspielen und bin daher wieder auf der Vorgängerversion.
@OnkelFD : Falls du die Version noch laufen hast, kannst du das zur Sicherheit noch einmal bei dir prüfen?

OnkelFD, 2023-08-15:
Guten morgen,
erster Fehler das mit den Objekten konnte ich nur beheben durch ein komplettes löschen des Adapters.
Das löschen des Objektbaumes brachte allerdings auch keine Besserung.

aruttkamp, 2023-08-17:
Die Werte sind mit der 1.1.1 wieder "normal" .

closed 2023-08-17 by aruttkamp