aruttkamp/ioBroker.rct #30: rct.0.info.connection / Fehler mit Terminierung des Adapters

OnkelFD, 2023-08-15:
Guten Morgen,

gestern hatte ich nach dem Update auf die v1.1.0 den Fehler dass der das Objekt rct.0.info.connection nicht finden kann dies Habe ich manuell angelegt jetzt kommt allerdings folgender Fehler.

rct.0	2023-08-15 10:21:41.787	info	RCT: disconnected from server
rct.0
2023-08-15 10:21:41.785	warn	Terminated (UNCAUGHT_EXCEPTION): Without reason
rct.0
2023-08-15 10:21:41.784	info	terminating
rct.0
2023-08-15 10:21:41.781	error	Exception-Code: ERR_BUFFER_OUT_OF_BOUNDS: Attempt to access memory outside buffer bounds
rct.0
2023-08-15 10:21:41.780	error	RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readFloatBackwards [as readFloatBE] (node:internal/buffer:520:5) at parseResponse (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:258:28) at handleData (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:147:20) at handleData (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:169:3) at handleData (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:169:3) at Socket.<anonymous> (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:111:3) at Socket.emit (node:events:514:28) at addChunk (node:internal/streams/readable:324:12)
rct.0
2023-08-15 10:21:41.778	error	uncaught exception: Attempt to access memory outside buffer bounds
rct.0
2023-08-15 10:20:41.636	info	RCT: connected to server at 192.168.178.135
rct.0
2023-08-15 10:20:41.547	info	config rct_ip: 192.168.178.135
rct.0
2023-08-15 10:20:41.525	info	starting. Version 1.1.0 (non-npm: aruttkamp/ioBroker.rct#146cce57638a1afe6020e1acdec853c89018350e) in /opt/iobroker/node_modules/iobroker.rct, node: v18.17.1, js-controller: 4.0.24

NCIceWolf, 2023-08-15:
Ah, das passt zu dem was bei mir gestern zu den Abstürzen geführt hat.
Dann hatte ich genau den gleichen Fehler.

aruttkamp, 2023-08-15:
*grummel* das sagt mir grad so gar nix... versuche das mal nachzuvollziehen.

NCIceWolf, 2023-08-15:
Hier meine Übersicht:

CPU: 2,93 % RAM: 23,5 % 
JS-Controller:
Verfügbar: 4.0.24 Installiert: 4.0.23
Plattform: docker 
Betriebssystem: linux 
Architektur: x64 
CPUs: 4 
Geschwindigkeit: 2395 MHz 
Modell: AMD GX-424CC SOC with Radeon(TM) R5E Graphics 
RAM: 14.6 GB 
Node.js: v18.16.0 
NPM: 9.5.1 
Datenträgergröße: 10994.0 GB 
Freier Festplattenspeicher: 3627.0 GB

Vielleicht komme ich am Wochenende mal dazu ein neues Docker Image mit der aktuellen js Controller Version zu installieren.
Dann könnte ich sagen ob es daran liegt.
-- 
Gesendet mit pEp für Android.

OnkelFD, 2023-08-15:
Ja die Versionen passen soweit.

Node.js - v18.17.1
JS-Controller - 4.0.24

aruttkamp, 2023-08-17:
Den Fehler konnte ich nur zum Teil reproduzieren. Im Test ist der Adapter ist gestern bei mir den ganzen Tag fehlerfrei gelaufen. 
Bitte mal mit 1.1.1 beobachten.

OnkelFD, 2023-08-17:
Also sobald ich nun den Adapter starte bringt er folgende Meldungen jetzt meckert er auch noch etwas im host an.

host.ioBroker
2023-08-17 11:07:27.497	warn	Do not restart adapter system.adapter.rct.0 because restart loop detected

host.ioBroker
2023-08-17 11:07:27.497	error	instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)

rct.0
2023-08-17 11:07:26.928	warn	Terminated (UNCAUGHT_EXCEPTION): Without reason

rct.0
2023-08-17 11:07:26.924	error	Exception-Code: ERR_BUFFER_OUT_OF_BOUNDS: Attempt to access memory outside buffer bounds

rct.0
2023-08-17 11:07:26.924	error	RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readFloatBackwards [as readFloatBE] (node:internal/buffer:520:5) at parseResponse (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:261:28) at handleData (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:147:20) at Socket.<anonymous> (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:111:3) at Socket.emit (node:events:514:28) at addChunk (node:internal/streams/readable:324:12) at readableAddChunk (node:internal/streams/readable:297:9) at Readable.push (node:internal/streams/readable:234:10)

rct.0
2023-08-17 11:07:26.922	error	uncaught exception: Attempt to access memory outside buffer bounds

host.ioBroker-Server
2023-08-17 11:06:19.315	warn	States 127.0.0.1:47838 Error from InMemDB: Error: CONFIG-UNSUPPORTED for ["SET","notify-keyspace-events","Exe"]

host.ioBroker-Server
2023-08-17 11:06:19.208	warn	Objects 127.0.0.1:35500 Error from InMemDB: Error: CONFIG-UNSUPPORTED for ["SET","lua-time-limit","10000"]

aruttkamp, 2023-08-17:
Mmhh.. kannst du mir bitte die Liste der Parameter schicken die du abrufst ?

2023-08-17 11:07:26.924 error RangeError [ERR_BUFFER_OUT_OF_BOUNDS]: Attempt to access memory outside buffer bounds at new NodeError (node:internal/errors:405:5) at boundsError (node:internal/buffer:86:11) at Buffer.readFloatBackwards [as readFloatBE] (node:internal/buffer:520:5) at parseResponse (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:261:28) at handleData (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:147:20) at Socket. (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:111:3) at Socket.emit (node:events:514:28) at addChunk (node:internal/streams/readable:324:12) at readableAddChunk (node:internal/streams/readable:297:9) at Readable.push (node:internal/streams/readable:234:10)

OnkelFD, 2023-08-17:
battery.inv_cmd,
battery.bat_status,
battery.max_cell_temperature,
battery.bat_status,
battery.soc,battery.soh,
battery.stored_energy,
battery.temperature,
battery.used_energy,
db.core_temp,
db.power_board.status,
dc_conv.dc_conv_struct[0].p_dc_lp,
dc_conv.dc_conv_struct[1].p_dc_lp,
energy.e_ac_total,
energy.e_dc_total[0],
energy.e_dc_total[1],
energy.e_grid_feed_total,
energy.e_grid_load_total,
energy.e_load_total,
g_sync.p_ac_load_sum_lp,
g_sync.p_ac_sum_lp,
g_sync.p_acc_lp,
g_sync.u_l_rms[0],
g_sync.u_l_rms[1],
g_sync.u_l_rms[2],
power_mng.bat_next_calib_date,
power_mng.battery_power

aruttkamp, 2023-08-17:
ok. Ich erhalte den Fehler jetzt auch. Leider ist er nur schwer reproduzierbar. Will heissen , er tritt mal direkt nach dem Starten auf, mal erst nach ner Stunde. Ich muss sehen was ich dagegen tun kann. Kann so natürlich nicht bleiben.

aruttkamp, 2023-08-17:
Fehler ist gefunden. Der Wechselrichter gibt teilweise komische Daten zurück. 
Sollte jetzt wieder absturzfrei laufen. Kann ich selbst erst morgen feststellen. 
hab den letzten Test kurz vor Sonnenuntergang positiv abgeschlossen ;-)

OnkelFD, 2023-08-17:
Sehr gut 👍
bei mir läuft es bis jetzt ohne Probleme 😘

aruttkamp, 2023-08-18:
Was mich noch interessieren würde... welche Softwareversion auf dem WR ist im Einsatz ?
Ich glaube die spielt auch noch eine Rolle.

OnkelFD, 2023-08-18:
Also der Wechselrichter hat die neuste Version (v2.3.5661) aufgespielt welche mir angeboten wird.

Der Adapter ist heute Morgen mit der v1.1.3 noch einmal abgestürzt ist aber schon deutlich länger gelaufen, ich musste allerdings das komplette System neu starten (ist ziemlich träge geworden).

Heute morgen das Update auf v1.1.4 aufgespielt, bis jetzt keinerlei Probleme gehabt.

Gruß

OnkelFD, 2023-08-18:
Stürzt leider immer noch alle paar Stunden ab bzw. startet von selbst neu 😥.

aruttkamp, 2023-08-18:
Gibt gleich noch ein Release.
Die Fehler Meldung wär noch cool

OnkelFD, 2023-08-18:
Mehr Infos als die kommen leider nicht ich hoffe du kannst etwas damit anfangen.

2023-08-18 14:03:05.747  - error: host.ioBroker instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-08-18 14:03:05.747  - info: host.ioBroker Restart adapter system.adapter.rct.0 because enabled
2023-08-18 14:03:05.747  - warn: host.ioBroker Do not restart adapter system.adapter.rct.0 because restart loop detected

aruttkamp, 2023-08-18:
Alles klar. Vielen Dank für Eure Geduld. Ich selbst habe den Adapter heute ziemlich krass arbeiten lassen und es gab immer ein und denselben Abbruch. Diesen habe ich nun ebenfalls abgefangen.
Ich hoffe damit ist Ruhe und wir ( ich ) können uns nun endlich den Erweiterungen widmen ;-)

OnkelFD, 2023-08-21:
Also ich habe es jetzt mal eine längere Zeit laufen lassen um zu schauen wie sich der Adapter verhält aber er Stürzt leider immer noch von Zeit zu Zeit ab 😢. 

Das kommt bei mir immer wenn der Adapter Startet:

host.ioBroker-Server | 2023-08-21 13:27:39.843 | warn | States 127.0.0.1:38558 Error from InMemDB: Error: CONFIG-UNSUPPORTED for ["SET","notify-keyspace-events","Exe"]
host.ioBroker-Server | 2023-08-21 13:27:39.669 | warn | Objects 127.0.0.1:60802 Error from InMemDB: Error: CONFIG-UNSUPPORTED for ["SET","lua-time-limit","10000"]

Und das wenn er abstürzt:

2023-08-21 11:39:15.239  - error: host.ioBroker instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
2023-08-21 11:39:15.240  - info: host.ioBroker Restart adapter system.adapter.rct.0 because enabled
2023-08-21 11:39:15.240  - warn: host.ioBroker Do not restart adapter system.adapter.rct.0 because restart loop detected

Danke für deine zeit und die Arbeit die du in das Projekt steckst.

aruttkamp, 2023-08-21:
Wie kommst Du darauf, dass die Meldungen zusammenhängen ? 
Ich erkenne bei den oberen keinerlei Zusammenhang zum Adapter. 
Schickst du nochmal deine Versionen ,bitte? 
Zuletzt hattest du gesagt : 
Node.js - v18.17.1
JS-Controller - 4.0.24

OnkelFD, 2023-08-22:
Also sobald ich den RCT Adapter Starte kommt diese Meldung bei mir also die Host Meldung.
 Die Versionen passen soweit:
Node.js - v18.17.1
JS-Controller - 4.0.24
RCT 1.1.5

Habe auf meiner Teststation den JS-Controller - 5.0.11 installiert dort kommt es nicht.
Muss aber mal noch schauen ob es dort auch zum Absturz kommt der Test läuft gerade.

aruttkamp, 2023-08-22:
ok. ich kann es aktuell nicht nachvollziehen. Weder im Live- noch im Testsystem.

NCIceWolf, 2023-08-22:
So, meine Versionen sind wie folgt:

Node.js - v18.17.1
JS-Controller - 4.0.24
NPM: 9.6.7
RCT 1.1.5

Der Adapter startet bei mir mit den folgenden Log-Einträgen:

<html><body>

rct.0 | 2023-08-22 19:34:27.054 | info | RCT: connected to server at 192.168.178.70
rct.0 | 2023-08-22 19:34:27.049 | info | rct state not defined:
rct.0 | 2023-08-22 19:34:27.048 | warn | Invalid RCT name:
rct.0 | 2023-08-22 19:34:27.037 | info | rct state not defined: power_mng.is_grid
rct.0 | 2023-08-22 19:34:27.036 | warn | Invalid RCT name: power_mng.is_grid
rct.0 | 2023-08-22 19:34:27.036 | info | rct state not defined: power_mng.battery_type
rct.0 | 2023-08-22 19:34:27.035 | warn | Invalid RCT name: power_mng.battery_type
rct.0 | 2023-08-22 19:34:27.025 | info | rct state not defined: dc_conv.dc_conv_struct[1].enabled
rct.0 | 2023-08-22 19:34:27.024 | warn | Invalid RCT name: dc_conv.dc_conv_struct[1].enabled
rct.0 | 2023-08-22 19:34:27.021 | info | rct state not defined: dc_conv.dc_conv_struct[0].enabled
rct.0 | 2023-08-22 19:34:27.020 | warn | Invalid RCT name: dc_conv.dc_conv_struct[0].enabled
rct.0 | 2023-08-22 19:34:26.995 | warn | State "rct.0.info.connection" has no existing object, this might lead to an error in future versions
rct.0 | 2023-08-22 19:34:26.978 | info | config rct_ip: 192.168.178.70
rct.0 | 2023-08-22 19:34:26.943 | info | starting. Version 1.1.5 in /opt/iobroker/node_modules/iobroker.rct, node: v18.17.1, js-controller: 4.0.24

Speziell die Meldung "State "rct.0.info.connection" has no existing object, this might lead to an error in future versions"
kommt bei mir also weiterhin. Bin mal gespannt ob und wie lange der Adapter ohne Murren läuft.

aruttkamp, 2023-08-22:
> Speziell die Meldung "State "rct.0.info.connection" has no existing object, this might lead to an error in future versions"
> kommt bei mir also weiterhin. Bin mal gespannt ob und wie lange der Adapter ohne Murren läuft.

Die Meldung ist aktuell "normal" und nur eine Warnung. hat im Moment keine Auswirkungen. Das Forschungsprojekt "Was muss ich tun" ist bereits angelaufen. 
Die werte die er anmeckert müsstest du aus deiner Konfig entfernen damit die Meldung verschwindet. diese werte werden aktuell nicht unterstützt. 
Wenn da etwas bei ist was man wirklich benötigt, dann müsste es eingebaut werden. 'Dann bitte genaue Angaben dazu.

aruttkamp, 2023-08-22:
> host.ioBroker-Server | 2023-08-21 13:27:39.843 | warn | States 127.0.0.1:38558 Error from InMemDB: Error: CONFIG-UNSUPPORTED for ["SET","notify-keyspace-events","Exe"]
> host.ioBroker-Server | [...]

Dies sind keine Meldungen die vom RCT Adapter erzeugt werden..

NCIceWolf, 2023-08-22:
> Die Meldung ist aktuell "normal" und nur eine Warnung. hat im Moment keine Auswirkungen. Das Forschungsprojekt "Was muss ich tun" ist bereits angelaufen. Die werte die er anmeckert müsstest du aus [...]

Ist ja glatt wie Magie...
Kaum nimmt man die raus, schon sind die Meldungen weg :-D

Müssten wir nur mal drüber nachdenken die aus den "Defaults" in der Readme raus zu nehmen.

aruttkamp, 2023-08-22:
> Müssten wir nur mal drüber nachdenken die aus den "Defaults" in der Readme raus zu nehmen.
*autsch* 
erwischt. nicht nur aus der Readme.. vor allem auch aus dem Code... 
das mache ich mal direkt.

NCIceWolf, 2023-08-22:
Also jetzt mal ganz im Ernst:

Was du da an Zeit und Expertise rein steckst ist schon echt der Hammer!!
Alleine in den letzten 2 Wochen gab es mehr Updates als in der gesamten Lebenszeit des Adapters!

Wirklich, beide Daumen hoch und vielen vielen Dank für deine Arbeit!! 💯 👍 🥇

aruttkamp, 2023-08-22:
> Alleine in den letzten 2 Wochen gab es mehr Updates als in der gesamten Lebenszeit des Adapters!

Danke - War aber auch meinem "lernen" geschuldet. Jeder hat so seinen Programmierstil und man erkennt manchmal nicht direkt warum etwas so gemacht ist wie es gemacht ist. 
Kommen noch ein paar Anpassungen und mir schwebt noch eine einfachere Konfiguration vor ;-)

aruttkamp, 2023-08-22:
> Speziell die Meldung "State "rct.0.info.connection" has no existing object, this might lead to an error in future versions"
> kommt bei mir also weiterhin. Bin mal gespannt ob und wie lange der Adapter ohne Murren läuft.

Ist mit nächster Version dann auch weg.

NCIceWolf, 2023-08-26:
So, bei mir sind inzwischen weder irgendwelche Fehlermeldungen sichtbar, noch ist der Adapter seit dem letzten Update bei mir abgestürzt.

@OnkelFD : Stürzt bei dir mit der aktuellen Version noch was ab? Oder können wir diesen Punkt abschließen?

OnkelFD, 2023-08-26:
Der Adapter läuft bei mir nun stabil und hatte keine Abstürze mehr.
Der Fehler der mir beim starten angezeigt wird (vom host.ioBroker-Server) kommt zwar immer noch aber hat anscheinend keinen Einfluss auf die Funktion. Ich hatte meinen aktive ioBroker mal testweise auf die Controller Version 5.0.12 angehoben und danach ist der Fehler verschwunden.

Die folgenden Versuche hatten keine Besserung gebracht
-den Controller 4.0.24 erneut zu installieren
-ioBroker neu installieren
-Adapter löschen & neu installieren

nur das update Auf eine neuere Controller Version brachte Abhilfe

da es so aber keine weiteren Probleme Verursacht warte ich einfach auf eine neue Controller Version.

Wobei der Adapter auf Orange bleibt "Verbunden mit Gerät oder Dienst", das hat ja etwas mit der info.connection zu tun.
Ist das noch in Arbeit oder Habe nur ich den Fehler?
Im Protokoll sind keine Fehler vom RCT Adapter zu finden. 

Gruß und danke für die tolle Arbeit, freue mich so dass an dem Adapter Weitergearbeitet wird, leider kenne ich mich damit nicht wirklich aus sonst würde ich meine Hilfe beim Programmieren anbieten.

NCIceWolf, 2023-08-26:
Schau mal auf was bei dir der "rct.0.info.connection" steht.
Sollte der auf "false" stehen(von früher), ändere den mal manuell auf "true" und starte den Adapter neu.
-- 
Gesendet mit pEp für Android.

OnkelFD, 2023-08-26:
Das habe ich auch schon gemacht ist aber wieder auf false gesprungen.
Habe ihn auch schon gelöscht und er wurde neu angelegt das passt soweit er wird/bleibt aber nicht true.

aruttkamp, 2023-08-27:
Bleibt auch false in der 1.0.6 ;-)

NCIceWolf, 2023-09-01:
Und nun auch hier:
Mit der 1.1.7 bekomme ich hier keinen Fehler mehr, der Wert steht korrekt auf "true".
Abstürze hatte ich schon seit der 1.1.6 nicht mehr, von daher kann für mich dieser Punkt zu :-)

@OnkelFD: Wie steht es bei dir? Läuft alles mit der 1.1.7?

OnkelFD, 2023-09-01:
Leider habe ich dennoch von zeit zu zeit abstürze mit der v1.1.6 und v1.1.7 hält mal einen oder Zwei tage aber dann stürzt er wieder ab 😢 kann es jetzt aber auch nicht ausschließen dass es an meinem System liegt.

Setting:
Proxmox
VM mit Ubuntu 22.04.3
Node.js v18.17.1
Controller 4.0.24
RCT- Adapter  v1.1.7

also das ist mal ein log von heute morgen:

2023-09-01 10:50:03.292  - info: rct.0 (11745) config rct_ip: 192.168.178.135
2023-09-01 10:50:03.305  - silly: rct.0 (11745) States system redis pmessage system.adapter.rct.0.logLevel/system.adapter.rct.0.logLevel:{"val":"silly","ack":true,"ts":1693558203296,"q":0,"from":"system.adapter.rct.0","lc":1693470938732}
2023-09-01 10:50:03.398  - info: rct.0 (11745) RCT: connected to server at 192.168.178.135
2023-09-01 10:50:03.404  - warn: rct.0 (11745) Read-only state "rct.0.info.connection" has been written without ack-flag with value "true"
2023-09-01 10:55:23.476  - error: rct.0 (11745) uncaught exception: write ECONNRESET
2023-09-01 10:55:23.478  - error: rct.0 (11745) Error: write ECONNRESET
    at afterWriteDispatched (node:internal/stream_base_commons:160:15)
    at writeGeneric (node:internal/stream_base_commons:151:3)
    at Socket._writeGeneric (node:net:931:11)
    at Socket._write (node:net:943:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Writable.write (node:internal/streams/writable:337:10)
    at /opt/iobroker/node_modules/iobroker.rct/rct/rct.js:73:15
    at Array.forEach (<anonymous>)
    at Timeout.requestElements [as _onTimeout] (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:71:16)
2023-09-01 10:55:23.478  - error: rct.0 (11745) Exception-Code: ECONNRESET: write ECONNRESET
2023-09-01 10:55:23.482  - info: rct.0 (11745) terminating
2023-09-01 10:55:23.483  - warn: rct.0 (11745) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-09-01 10:55:23.985  - info: rct.0 (11745) terminating
2023-09-01 10:55:55.036  - debug: rct.0 (11777) Redis Objects: Use Redis connection: 127.0.0.1:9001
2023-09-01 10:55:55.091  - debug: rct.0 (11777) Objects client ready ... initialize now
2023-09-01 10:55:55.094  - warn: rct.0 (11777) Unable to increase LUA script timeout: Error CONFIG-UNSUPPORTED for ["SET","lua-time-limit","10000"]
2023-09-01 10:55:55.095  - debug: rct.0 (11777) Objects create System PubSub Client
2023-09-01 10:55:55.096  - debug: rct.0 (11777) Objects create User PubSub Client
2023-09-01 10:55:55.191  - debug: rct.0 (11777) Objects client initialize lua scripts
2023-09-01 10:55:55.197  - debug: rct.0 (11777) Objects connected to redis: 127.0.0.1:9001
2023-09-01 10:55:55.199  - silly: rct.0 (11777) redis psubscribe cfg.o.system.user.*
2023-09-01 10:55:55.229  - silly: rct.0 (11777) redis psubscribe cfg.o.enum.*
2023-09-01 10:55:55.231  - silly: rct.0 (11777) objectDB connected
2023-09-01 10:55:55.232  - debug: rct.0 (11777) Redis States: Use Redis connection: 127.0.0.1:9000
2023-09-01 10:55:55.255  - warn: rct.0 (11777) Unable to enable Expiry Keyspace events from Redis Server: Error CONFIG-UNSUPPORTED for ["SET","notify-keyspace-events","Exe"]
2023-09-01 10:55:55.255  - debug: rct.0 (11777) States create System PubSub Client
2023-09-01 10:55:55.256  - debug: rct.0 (11777) States create User PubSub Client
2023-09-01 10:55:55.336  - debug: rct.0 (11777) States connected to redis: 127.0.0.1:9000
2023-09-01 10:55:55.337  - silly: rct.0 (11777) statesDB connected
2023-09-01 10:55:55.594  - info: rct.0 (11777) starting. Version 1.1.7 (non-npm: aruttkamp/ioBroker.rct#refs/tags/v1.1.7) in /opt/iobroker/node_modules/iobroker.rct, node: v18.17.1, js-controller: 5.0.12
2023-09-01 10:55:55.625  - info: rct.0 (11777) config rct_ip: 192.168.178.135
2023-09-01 10:55:55.640  - silly: rct.0 (11777) States system redis pmessage system.adapter.rct.0.logLevel/system.adapter.rct.0.logLevel:{"val":"silly","ack":true,"ts":1693558555629,"q":0,"from":"system.adapter.rct.0","lc":1693470938732}
2023-09-01 10:55:55.737  - warn: rct.0 (11777) Read-only state "rct.0.info.connection" has been written without ack-flag with value "true"
2023-09-01 10:56:02.902  - info: rct.0 (11777) RCT: connected to server at 192.168.178.135
2023-09-01 11:02:02.992  - error: rct.0 (11777) uncaught exception: write ECONNRESET
2023-09-01 11:02:02.993  - error: rct.0 (11777) Error: write ECONNRESET
    at afterWriteDispatched (node:internal/stream_base_commons:160:15)
    at writeGeneric (node:internal/stream_base_commons:151:3)
    at Socket._writeGeneric (node:net:931:11)
    at Socket._write (node:net:943:8)
    at writeOrBuffer (node:internal/streams/writable:392:12)
    at _write (node:internal/streams/writable:333:10)
    at Writable.write (node:internal/streams/writable:337:10)
    at /opt/iobroker/node_modules/iobroker.rct/rct/rct.js:73:15
    at Array.forEach (<anonymous>)
    at Timeout.requestElements [as _onTimeout] (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:71:16)
2023-09-01 11:02:02.993  - error: rct.0 (11777) Exception-Code: ECONNRESET: write ECONNRESET
2023-09-01 11:02:02.998  - info: rct.0 (11777) terminating
2023-09-01 11:02:02.999  - warn: rct.0 (11777) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-09-01 11:02:03.502  - info: rct.0 (11777) terminating

eventuell hilft das weiter, was komisch ist er zeigt den Controller v5.0.12 was nicht stimmt.

aruttkamp, 2023-09-01:
Dieser LOG stammt nicht von der 1.1.7 !!
bitte installier die Version von NPM !

OnkelFD, 2023-09-01:
Okay ich Probiere es mal aus.
Hab es immer über den den GitLink (https://github.com/aruttkamp/ioBroker.rct/archive/refs/tags/v1.1.7.tar.gz) installiert, was ist da der unterschied?

NCIceWolf, 2023-09-01:
Hm, für mich liest sich das eher nach einem Timeout in deinem Setting.
Leider wird nicht genau aufgeführt was genau da das Problem ist.
Datenbank? Redis? Oder die Verbindung zum WR?

Läuft das alles bei dir auf der gleichen Maschine oder sind Redis und Datenbank extra?

NCIceWolf, 2023-09-01:
Zum Glück liest sich jeder Log anders, aber diese Zeile hier:
at Timeout.requestElements [as _onTimeout] (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:71:16)

Hier werden die Elemente vom WR abgefragt. Und wenn das in einen Timeout läuft, antwortet entweder dein WR nicht (rechtzeitig) oder deine Maschine hängt und kriegt den Request nicht raus.
Welches Intervall hast du denn für die Abfragen? Die 10 Sekunden Standardeinstellung?

NCIceWolf, 2023-09-05:
Ich würde zum Test vielleicht einfach mal die Anzahl der Elemente z.b. auf 5 reduzieren und schauen ob der Fehler weiter auftritt. Oder auch das Intervall verlängern.

Ansonsten bleiben die basics, wie anderes Netzwerkkabel, anderer Port am Switch und im dümmsten Fall müsste man mitschneiden wo die Daten denn hängen bleiben.

Scheint kein generelles Thema zu sein, sonst scheint das Thema ja niemand zu haben.

aruttkamp, 2023-09-09:
Wenn hier noch etwas sein sollte bitte ein neuen Issue aufmachen. 
info.connection und unerwartete Terminierung des Adapters sind mit der 1.1.7 definitiv erledigt. Das waren beides von mir versehentlich eingebaute Fehler.

closed 2023-09-09 by aruttkamp

OnkelFD, 2023-09-13:
Hi,
ich habe gerade eben die neue Version v1.2.2 installiert und damit ist der Fehler ("lua-time-limit") welcher beim Starten erzeugtet wurde erledigt.

😊