aruttkamp/ioBroker.rct #31: battery.cycles wird mit 0 angezeigt

BenS89, 2023-08-22:
Hallo zusammen,

bei mir wird der Parameter battery.cycles immer mit 0 zurückgegeben, was nicht korrekt ist. Interessanterweise klappen andere werte, die SOC oder SOH ohne Probleme. 

Gibt es bei dem Parameter battery.cycles aktuell ein Problem?

BenS89, 2023-08-22:
Was ich dazu sagen muss: ich bin aktuell noch auf Version 1.0.1. Bisher habe ich keine Möglichkeit gefunden, den Adapter zu aktualisieren, ohne die zugehörigen Instanzen löschen zu müssen. Da ich recht viel über InfluxDV logge, möchte ich die individuellen Einstellungen der Datenpunkte nicht verlieren. 

Gibt es eine Möglichkeit, von 1.0.1 auf die aktuellste Variante zu gehen, ohne die Adapter-Instanzen zwischendrin löschen zu müssen?

BenS89, 2023-08-22:
Bisher hatte ich den Adapter immer löschen müssen, um dann eine Instanz mit der neuen Version starten zu können. Nach einem npm install iobroker.rct war sonst immer noch die alte Version 1.0.1 zu sehen. Vermutlich mache ich aber noch einen Fehler bei der Aktualisierung des Adapters

aruttkamp, 2023-08-22:
v1.0.1 war vor meiner Zeit ;-) 
Hatte ein solches Problem noch nicht. 
Mach ne Sicherung und probier es :-D

camast, 2023-08-22:
Zuerst auch von meiner Seite vielen Dank für die Arbeit,  die Du in den Adapter steckst.
Zu battery.cycles und battery.SoH:
Wenn man die Instanz vor der Installation löscht, werden diese beiden Datenpunkte (bei mir) nicht mit der  Neuinstallation (v1.0.5) angelegt.
Für mich sind die auch nicht spannend,  weil sehr statisch (bei mir ca. 250 Cycles p.a. und SoH immer 100 - über rctclient ausgelesen).

aruttkamp, 2023-08-22:
Vielen Dank camast. Wenn du die beiden Werte in die Konfiguration geschrieben hast , dann müssen auch die Datenpunkte angelegt werden ! Sehe nicht wie er das umgehen sollte.

BenS89, 2023-08-22:
Nach dem Update bekomme ich ebenfalls eine Zahl angezeigt: 309. Das deckt sich aber nicht mehr der Zyklen-Zahl, die mir der RCT Support genannt hat (469) Zyklen. Ich werde beim Support mal nachfragen, wo diese Differenz herkommen könnte. 

Auch von meiner Seite ein großes Dankeschön an dich, dass du die Entwicklung des Adapters weiter vorantreibst!! Er ist bei mir die Basis für das komplette Logging über Influx und Grafana.

camast, 2023-08-22:
erwischt ... ich habe gar keine Einträge bei den RCT-Elements übergeben. Die Datenpunkte die h brauche, waren in dem Umfang in diesem Fall angelegt wird enthalten.

aruttkamp, 2023-08-22:
> erwischt ... ich habe gar keine Einträge bei den RCT-Elements übergeben. Die Datenpunkte die h brauche, waren in dem Umfang in diesem Fall angelegt wird enthalten.

Standard sind diese hier : 
battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].enabled,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].enabled,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.battery_type,power_mng.is_grid,power_mng.is_heiphoss,power_mng.state,power_mng.use_grid_power_enable,power_mng.u_acc_mix_lp,prim_sm.island_flag,prim_sm.state';

aruttkamp, 2023-08-22:
> Ich werde beim Support mal nachfragen, wo diese Differenz herkommen könnte.
Kannst Du diesen Wert nicht in der APP sehen ? Einen Fehler beim auslesen würde ich ausschliessen. Das sähe anders aus.

BenS89, 2023-08-22:
Dass die Ladezyklen in der App eingesehen werden können, wäre mir neu. Geht das?

aruttkamp, 2023-08-22:
Was passiert , wenn du in der APP auf den Speicher „drückst“ ?

BenS89, 2023-08-22:
Bei iOS leider nix :-(

NCIceWolf, 2023-08-22:
> Leider habe ich keinen Akku , weshalb der Wert 0 bei mir sehr plausibel ist. @NCIceWolf wie ist es bei Dir ?

So, bin leider auch eben erst dazu gekommen die aktuellste Version zu installieren.
Der Wert "rct.0.battery.cycles" gibt bei mir "117" zurück.
Schaue ich in der RCT App unter "Gerät" -> "Messwerte" -> "Batterie" sehe ich da ganz unten den exakt gleichen Wert.
Ich meine hier muss man als "Service" eingeloggt sein um dort hin zu kommen.

Meiner Meinung nach passt das also einwandfrei 🥇

BenS89, 2023-08-22:
Danke für eure schnellen Kommentare. Die Android App kann da offenbar deutlich mehr als die iOS Variante. Ich habe es eben auf einem Emulator nachvollzogen: App und Adapter sind sich einig --> das scheint also zu passen. Folglich ist die Info vom RCT Service nicht korrekt. 

Danke euch!

aruttkamp, 2023-08-22:
ok. Android : Apple 1:0
Ich mach dann hier zu - weil gelöst.

aruttkamp, 2023-08-24:
> @aruttkamp Bekommen wir diese Batterie-Werte auch über den Adapter ausgelesen? Dort sieht man alle Infos pro Batterie-Modul, und eben auch die korrekten Ladezyklen.

Ich denke schon. Hab aber immer noch keine Batterie. 
teste mal "battery.cycles"

BenS89, 2023-08-24:
> > @aruttkamp Bekommen wir diese Batterie-Werte auch über den Adapter ausgelesen? Dort sieht man alle Infos pro Batterie-Modul, und eben auch die korrekten Ladezyklen.
> 
> Ich denke schon. Hab aber immer noch keine Batterie. teste mal "battery.cycles"

battery.cycles zeigt den Wert des Wechselrichters an, also quasi den falschen. Es müsste pro Batteriemodul dann nochmal einen separaten Cycle-Wert geben, der laut RCT immer richtig ist

aruttkamp, 2023-08-24:
ja, dann nimm
battery.stack_cycles[0]
battery.stack_cycles[1]
..
..
battery.stack_cycles[6]

BenS89, 2023-08-24:
> ja, dann nimm
> battery.stack_cycles[0]
> battery.stack_cycles[1]
> ..
> ..
> battery.stack_cycles[6]

War auch mein Gedanke, leider kommen dort falsche Werte zurück:

[image]

Richtig wären eigentlich 475 Zyklen:

[image]

BenS89, 2023-08-24:
Liegt‘s ggf am Datentyp? `battery.cycles` ist ein `INT32`, das liest er richtig aus. Die battery.stack_cycles scheinen als unsigned Int definiert zu sein

aruttkamp, 2023-08-25:
> Liegt‘s ggf am Datentyp?

Davon gehe ich aus. 
Heute gab es Gute Nachrichten ! Ich bekomme in 1 Woche meinen Speicher ! 
d.h. zum einen ich bekomme endlich echte Werte und zum anderen ... der WR geht abends nicht mehr aus ! :-D
kann jetzt nämlich schon wieder nix machen weil der WR aus ist. 
Also, hab bitte noch ein wenig Geduld !

OnkelFD, 2023-08-26:
Also bei mir wird der Wert battery.cycles auch korrekt wiedergegeben der Abgleich mit der App passt, die Werte sin identisch.

BenS89, 2023-08-26:
> Also bei mir wird der Wert battery.cycles auch korrekt wiedergegeben der Abgleich mit der App passt, die Werte sin identisch.

Der Wert aus battery.cycles passt auch bei mir zu dem Wert in der App unter Gerät —> Messwerte —> Batterie. Beides liefert bei mir 309 Zyklen zurück. Dieser Wert ist in meinem Fall aber falsch (durch WR Tausch vor einigen Wochen).

Die richtige Zyklenzahl bekommt man in der App, wenn man im selben Menü ganz unten auf das BMS klickt. Dort sind dann die Zyklen pro Batterie-Modul aufgeführt. Laut RCT sind diese Werte IMMER korrekt. Der Wert unter Gerät —> Messerte —> Batterie kommt vom WR und kann beliebig falsch sein.

Die tatsächlichen Zyklen liegen vermutlich in den Arrays battery.stack_cycles[n], wobei Duft im Adapter aktuell noch der Datentyp falsch ist

aruttkamp, 2023-09-12:
Ja, Du kannst die Datentypen in der Datei rct_core2.js anpassen. 
Aber... damit unterstellst Du , dass der Wechselrichter ein anderes Datenformat liefert als von RCT dokumentiert. 
Das hatte ich - bisher - ausgeschlossen. Kann es auch erst heute Abend probieren. 

Batterie ist da und klappt auch soweit. muss mich jetzt noch mit den Einstellungen ( soc ) beschäftigen.
Aktuell laüft alle auf Standard.
7.5 kw hab ich und das reicht mal so gerade ;-)

aruttkamp, 2023-09-13:
So wie es aussieht ist der korrekte Datentyp INT16 und nicht UINT16 wie in der Doku angegeben. 
Zumindest bekomme ich damit die korrekten Werte angezeigt. 
In Version 1.2.2 habe ich das jetzt angepasst. 
Bitte um Feedback !

OnkelFD, 2023-09-14:
Also wenn ich bei mir schaue stimmen die Werte von stack_cycles_0-3 mit der App überein.

BenS89, 2023-09-14:
> So wie es aussieht ist der korrekte Datentyp INT16 und nicht UINT16 wie in der Doku angegeben.
> Zumindest bekomme ich damit die korrekten Werte angezeigt.
> In Version 1.2.2 habe ich das jetzt angepasst.
> Bitte um Feedback !

Jap, passt jetzt auch bei mir. Vielen Dank für´s Anpassen. Bei mir sind `battery.cycle` und `battery.stack_cycles[n]` vermutlich deswegen unterschiedlich, weil mein Wechselrichter zwischendrin getauscht wurde. Laut RCT haben die Werte in `battery.stack_cycles[n]` immer Recht, wohingegen `battery.cycle` vom Wechselrichter kommt und ggf. falsch sein kann. 

Top, dann kann das Issue zu. 

Vielen Dank

NCIceWolf, 2023-09-14:
Bestätige ebenfalls.
Mal wieder vielen Dank @aruttkamp 😃

BenS89, 2024-05-16:
@aruttkamp Seit dem Update auf v. 1.2.7. leider wieder kaputt. Ich vermute, dass der Datantyp aus Versehen überschrieben wurde. Statt 127 stehen jetzt bei mir 32k Cycles

[image]

aruttkamp, 2024-05-16:
@BenS89 
You are right ! 
War wieder der falsche Datentyp hinterlegt. Ich weiss warum es passiert ist ;-) 
corrected with 1.2.8

BenS89, 2024-05-16:
@aruttkamp Wow, das ging schnell :-)
Vielen Dank für den schnellen Fix

Viele Grüße
Ben

BenS89, 2024-05-17:
@aruttkamp Hat sich auch etwas am `power_mng.bat_next_calib_date` geändert?
Auch hier wird seit dem Update kein plausibler Timestamp mehr ausgegeben. 

Mit v. 1.2.4 konnte ich den Wert x 1000 verwenden, um das nächste Kalibrierungsdatum abzufragen. Nach dem Update erhalte ich den Wert 18.04.1997

NCIceWolf, 2024-05-17:
Jo, das hatte ich die Woche schon gesehen.
Shame on me, wollte das schon einmal ein melden 😅

@aruttkamp: immer diese Datentypen...

aruttkamp, 2024-05-17:
ok. alles klar. Korrektur kommt .... mit 1.2.9
@BenS89 @NCIceWolf  - Bitte macht so einen Issue wieder auf , wenn ihr was reinschreibt ;-)

aruttkamp, 2024-05-17:
> @aruttkamp Danke dir
> 
> Ich wollte das Issue wieder öffnen, hatte aber keine Schaltfläche dafür. Kann es sein, dass mir dazu die Rechte fehlen?

puhhh. Das könnte tatsächlich sein. dann hätten NCIceWolf oder ich das tun müssen :-P

NCIceWolf, 2024-05-17:
> > @aruttkamp Danke dir
> > Ich wollte das Issue wieder öffnen, hatte aber keine Schaltfläche dafür. Kann es sein, dass mir dazu die Rechte fehlen?
> 
> puhhh. Das könnte tatsächlich sein. dann hätten NCIceWolf oder ich das tun müssen :-P

Ich gelobe Besserung!
Passt mit der 1.2.9 wieder!

BenS89, 2024-05-17:
Super, vielen Dank für den schnellen Fix :-)
Schaut jetzt wieder gut aus

 Viele Grüße

aruttkamp, 2024-05-17:
> Super, vielen Dank für den schnellen Fix :-) Schaut jetzt wieder gut aus
> 
> Viele Grüße

Gerne !

closed 2024-05-17 by aruttkamp