aruttkamp/ioBroker.rct #33: Adapter stürzt nach Update auf V1.1.7 wiederholt ab

NCIceWolf, 2023-08-31:
Nach dem Update auf Version 1.1.7 stürzt der Adapter bei mir leider wiederholt ab.
Musste deshalb wieder zurück auf die 1.1.6.

Folgendes sagt der Debug-Log:

host.iobroker | 2023-08-31 04:02:55.260 | warn | Do not restart adapter system.adapter.rct.0 because restart loop detected
-- | -- | -- | --
host.iobroker | 2023-08-31 04:02:55.259 | error | instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
rct.0 | 2023-08-31 04:02:54.700 | warn | Read-only state "rct.0.info.connection" has been written without ack-flag with value "false"
rct.0 | 2023-08-31 04:02:54.697 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
rct.0 | 2023-08-31 04:02:54.696 | info | terminating
rct.0 | 2023-08-31 04:02:54.691 | error | this.setState is not a function
rct.0 | 2023-08-31 04:02:54.690 | error | TypeError: this.setState is not a function at Socket.<anonymous> (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:83:8) at Object.onceWrapper (node:events:628:28) at Socket.emit (node:events:526:35) at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1480:10)
rct.0 | 2023-08-31 04:02:54.687 | error | uncaught exception: this.setState is not a function
rct.0 | 2023-08-31 04:02:54.685 | info | RCT: connected to server at 192.168.178.70
rct.0 | 2023-08-31 04:02:54.605 | info | config rct_ip: 192.168.178.70
rct.0 | 2023-08-31 04:02:54.567 | info | starting. Version 1.1.7 (non-npm: aruttkamp/ioBroker.rct) in /opt/iobroker/node_modules/iobroker.rct, node: v18.17.1, js-controller: 4.0.24
rct.0 | 2023-08-31 04:02:54.359 | debug | States connected to redis: redis:6379

**Versions:**  
 - Adapter version: <1.1.7>
 - JS-Controller version: <4.0.24>
 - Node version: <18.17.1> 
 - Operating system: <Plattform: docker / Betriebssystem: linux>

aruttkamp, 2023-08-31:
Das sieht so aus als ob du manuell von github installiert hast, kann das sein ?

NCIceWolf, 2023-08-31:
Moin,

die 1.1.7 über NPM, die 1.1.6 hab ich dann manuell per URL drüber gejagt.
Sollte das einen Unterschied machen?
-- 
Gesendet mit pEp für Android.

aruttkamp, 2023-08-31:
Jein. Den Fehler den du beschreibst habe ich gestern abend noch "eingebaut". Allerdings ist der nicht in der 1.1.7 die bei NPM liegt sondern in der "latest" in github. DEshalb hatte ich befürchtet dass du den Stand erwischt hast. Siehst du daran , dass in der Readme der letzte Eintrag unter "work in progress" steht. 
Bei mir stürzt er nicht ab. 
Aber egal - es kommt im Laufe des Tages noch eine neue Version.

NCIceWolf, 2023-08-31:
Achso!
Bin halt gerne Beta-Tester :)

Auf das "work in progress" hab ich heute Nacht nicht wirklich geachtet, wollte nur die Chance nutzen wenn ich schon wach bin.
Dann warte ich einfach gespannt die nächste Version ab.

aruttkamp, 2023-08-31:
Das NPM Package ist i.O. nimm das und alles ist schön.

NCIceWolf, 2023-09-01:
So, frisch drauf und alles läuft wieder wie es soll:

rct.0 | 2023-09-01 07:51:11.425 | warn | Read-only state "rct.0.info.connection" has been written without ack-flag with value "true"
rct.0 | 2023-09-01 07:51:11.420 | info | RCT: connected to server at 192.168.178.70
rct.0 | 2023-09-01 07:51:11.352 | info | config rct_ip: 192.168.178.70
rct.0 | 2023-09-01 07:51:11.320 | info | starting. Version 1.1.7 (non-npm: aruttkamp/ioBroker.rct) in /opt/iobroker/node_modules/iobroker.rct, node: v18.17.1, js-controller: 4.0.24

closed 2023-09-01 by NCIceWolf