aruttkamp/ioBroker.rct #34: rct_refresh

wiese-computer-brb, 2023-09-05:
Hallo, deine Version 1.1.7 läuft ja seid dem erscheinen ohne Abstürze, ganz toll. Aber meine Frage den Wert "rct_refresh" habe ich versucht auf meine Bedürfnisse anzupassen, habe aber egal welchen Wert ( 1 - 200 ) keinen ersichtlichen Unterschied gesehen. Die Objekte werden aller 2-3 Sekunden aktualisiert. Gibt es da eine Trick? Es wäre nicht schlimm aber da ich die Werte an die Homematic übergebe, und diese nicht so oft abholt kommt es zu Fehlermeldungen. Oder muss ich eine Wartezeit einbauen für die Übergabe?

aruttkamp, 2023-09-05:
Moin ! 
Wie übergibst du an Homematic ? 
Über welche Fehlermeldung reden wir ?

wiese-computer-brb, 2023-09-05:
Die Sache mit den Fehlermeldungen ich nicht das entscheidende, nur die Häufigkeit der Änderung der Werte. hat der rct_refresh eine Funktion?

wiese-computer-brb, 2023-09-05:
Ich habe in der Homematic Systemvariablen erstellt und errechne die Werte die ich brauche mit einem Blockly Script und Steuere damit die Objekte (von hm-rega) im iobroker.

wiese-computer-brb, 2023-09-06:
Es ist schade das es keine Antwort auf die Frage mit dem rct_refresh git, dann müss wohl die Homematic damit leben das sich die systemvariablen alle 2 Secunden verändern und auch das Netzwerk freut sich. Stream vom Wechselrichter und zu Homematic. Eventuell gibt es ja doch eine Lösung. Danke

aruttkamp, 2023-09-06:
> Es ist schade das es keine Antwort auf die Frage mit dem rct_refresh git, 
Stimmt. und das obwohl schon 24h um sind.

Sorry - aber sowas macht mich sprachlos.

NCIceWolf, 2023-09-06:
@aruttkamp : Einfach ignorieren, gibt halt Leute die haben das Gefühl dich hierfür fürstlich zu entlohnen und deshalb Anrecht auf zeitnahe Reaktionen und Lösungen zu haben.
Wird es leider immer geben 😕

wiese-computer-brb, 2023-09-06:
Entschuldigung, das ist dann aber völlig falsch rüber gekommen.

wiese-computer-brb, 2023-09-06:
Das ich die Arbeit die hinter dem Adapter nicht würdige, möchte ich nicht sagen. Meine Anfrage begann mit dem Satz "Hallo, deine Version 1.1.7 läuft ja seid dem erscheinen ohne Abstürze, ganz toll. "

NCIceWolf, 2023-09-08:
Nachdem ich da jetzt etwas geforscht habe, sieht das für mich so aus:

Der Refresh Timer wird entweder per Konfig gesetzt, oder alternativ auf Default (10 Sekunden) gesetzt.
Im Code habe ich diesen Schnipsel gefunden, der aus meiner Sicht für das aktualisieren zuständig ist (requestElements):
__refreshTimeout = setTimeout(requestElements, (1000 * iobInstance.config.rct_refresh) || 15000);

Soweit ich den Code verstehe startet er also die Abfrage der Daten nach der eingestellten Zeit neu, jedoch durch das "|| 15000" spätestens alle 15 Sekunden.

@aruttkamp: War das Absicht und hat irgendwelche anderen Abhängigkeiten? Oder könnte man das rausnehmen und damit auch längere Intervalle zulassen?

aruttkamp, 2023-09-09:
> Nachdem ich da jetzt etwas geforscht habe, sieht das für mich so aus:

Erstmal vielen Dank an @NCIceWolf  für´s forschen. Du bist schon ziemlich richtig ;-) 

Ich habe die Pflege dieses Adapters übernommen und da jeder Entwickler so seine eigene Handschrift / seine eigenen Techniken hat muss man Code den man selbst nicht geschrieben hat verstehen ;-) 
Wenn dann für einen selbst auch noch Neuland hinzukommt ( nein ich meine nicht das Internet ) dann dauern gewissen Dinge einfach länger. Da der Adapter fehlerfrei läuft und das ganze ein Hobby ist ( bezahlten Support und bezahlte Programmierung gibts bei mir jeden Tag - das ist mein "Haupt" Job ) arbeite ich so daran wie es mir in den Kram passt ( sehr platt formuliert - ich weiss ) 

So.. aber nun zu diesem Timeout. 
Der Timeout wird hier genutzt um alle x Millisekunden die Daten zu holen. 
Standardeinstellung hierbei ist 10 - was bedeutet es wird alle 10x1000 = 10000 Millisekunden ( was 10 Sekunden entspricht ) der Abruf der Daten vom WR getriggert. Klappt auch - Ich habe testweise mal einen Logeintrag in die Routine reingebaut : 
[image]

Ändere ich das Intervall auf 60 Sekunden sieht der Log so aus : 
[image]

Klappt also alles wie es soll. Ich sehe keinen Bedarf für eine Anpassung des Adapters. 
Das "|| 15000" greift nur , wenn der vordere Teil , also die Berechnung, fehlerhaft ist. 

Da meine Fragen oben nicht konkret beantwortet wurden sondern nur sehr allgemein sehe ich auch keine Chance hier weiter zu helfen.

aruttkamp, 2023-09-15:
@wiese-computer-brb - können wir hier zumachen oder benötigst Du noch infos ?

closed 2023-09-22 by aruttkamp