aruttkamp/ioBroker.rct #46: Verbindung zu Wechselrichter bricht ab

Corneys, 2023-09-27:
Hallo, 

ich habe folgendes Problem, was ich mir nicht so richtig erklären kann.
Kurz zum Hintergrund der Anlage:
Ich habe zwei Wechselrichter inclusive PowerSwitch und Speicher. 
Der Speicher hängt logischerweise an dem Master-Wechselrichter und bleibt dementsprechend an, auch wenn kein Strom produziert wird.
Der Slave schaltet sich Abends ab, weswegen ich dann das Skript von [https://github.com/aruttkamp/ioBroker.rct/discussions/27](url) nutze um den Adapter des Slaves auszuschalten.

Jedoch verliert auch der Master, sobald es dunkel ist, in unregelmäßigen Abständen ,die Verbindung zum Adapter bis der Adapter deaktiviert wird.
Über den Ping-Adapter sehe ich aber, dass die ganze Zeit eine Verbindung besteht, weswegen leider auch das Skript zum neu starten nicht funktioniert.

Was ich dabei seltsam finde ist, dass nachdem nach einem Verbindungsabbruch, manchmal sofort und manchmal erst nach 10 Minuten erneut die Verbindung verliert (siehe Screenshot) 
In der Zeit des Screenshots habe ich den Adapter nicht angerührt, würde ich jetzt aber neu starten, würde er vermutlich innerhalb in den nächsten 15-60 Minuten wieder drei Mal die Verbindung verlieren und  wegen der Loop-detection deaktiviert werden.

[image]

aruttkamp, 2023-09-28:
Moin ! 

Erste Frage ... Adapter ist die Version 1.2.2 ?
Du hast 2 Adapterinstanzen laufen ? 

Welche Software Versionen setzt Du ein ? ( node, npm, Admin ) 

Sind beide WR identisch ? Welche? 
Welche SW Versionen werden verwendet ? 

Das sind erstmal so die Punkte die mir vor dem Kaffee einfallen ;-)

Corneys, 2023-09-28:
Guten Morgen,

ja beide Adapter sind auf Version 1.2.2
Genau für jeden WR eine Instanz, da es für mich so aussieht als müsste ich aus den Instanzen Werte addieren, um auf das zu kommen was mir in der RTC-Power App angezeigt wird.

node - 18.16.1
npm - 9.5.1
Admin - 6.10.1

Der Master-WR ist ein "RCT Power Storage DC 10.0" und der Slave ein "RCT Power Storage DC 8.0"
Beide Wechselrichter laufen auf dem SW-Control-Update: 2.3.5661

Ich hoffe, dass hilft erst mal weiter und lass dir deinen Kaffee schmecken ;-)

aruttkamp, 2023-09-28:
Das sieht alles gut aus. 
Bitte poste mal die Parameter die du abrufst.

Corneys, 2023-09-28:
Die Parameter in rct_elements sind bei mir leer. Es werden also die Standardparameter aus abgerufen. 
Mit weiteren Parametern werde ich mich erst beschäftigen wenn es richtig läuft :-) 

Genau, das ganze tritt erst auf, sobald es dunkel ist. Sprich, der zweite Wechselrichter abgeschaltet hat.

aruttkamp, 2023-09-29:
Ich habe das Problem verstanden. Dies wird scheinbar durch einen Netzwerkfehler hervorgerufen. Hier bei mir kann ich es nicht reproduzieren und einen Grund, warum das auftritt wenn der 2. WR "aus" ist sehe ich auch nicht. 
Ich werde im Adapter eine Fehlerbehandlung implementieren die dafür sorgt, dass der Adapter nicht abstürzt. 
Theoretisch ist es "im Kopf" fertig, aber ich brauche noch Zeit, da ich gerade ein größeres Projekt zu Ende bringen muss. 
Also bitte noch ein wenig Geduld.

Corneys, 2023-09-30:
Kein Stress wegen mir :-) 
Wenn ich noch irgendetwas tun kann um das Problem irgendwie einzugrenzen, dann helfe ich gerne. 
Ansonsten weiß ich, dass die Anlage mit den zwei Wechselrichtern vermutlich eher nicht die Regel ist.

Im schlimmsten Fall, falls es nicht klappt, würde ich mich damit behelfen die Adapter rechtzeitig vor Sonnenuntergang über den Helligkeitssensor der Wetterstation zu deaktivieren und morgens wieder einzuschalten. Einziger Wermutstropfen wäre dann nur, dass ich nachts den Ladestand des Speichers nicht abrufen kann. 
Damit Leben könnte ich aber. 

Vielen Dank aber trotzdem schon mal für deine Hilfe.

duejo46, 2023-10-04:
Moin Andreas, moin Corney,
erstmal vielen, vielen Dank an Andreas für die Betreuung und Weiterentwicklung des Adapters...Wenn ich mich recht erinnere, hatten wir hinsichtlich V1.0.3 mal ein paar  Mails ausgetauscht...
Die von Corney beschriebenen "Kommunikationsunterbrechungen" kenne ich und habe ich auf meinem System seit ich den RCT-Adapter erstmals installiert habe, also ab V1.0.0 im Feb./März23. Ich betreibe einen RCT DC6.0 mit 5,7kWh-Speicher. Hinsichtlich IOB habe ich zwei RPi4, eines als Produktivsystem, eines als Anfänger-Test-System. Alles auf aktuellen Stand, also Node 18.18.0, NPM 9.8.1, Admin 6.10.1 und RCT 1.2.2.
Der Fehler tritt bei mit völlig stochastisch auf, manchmal im Minutentakt, machmal ist auch stundenlang Ruhe. Ich habe kein Master/Slavesystem und es hat bei mir auch nichts mit der Tageszeit zu tun.
Die Warn/Alarmmeldungen sind exakt, wie oben von Corney beschrieben.
Meine Krücke, um den Adapter betreibbar zu halten, ist ein scheduled restart. Ich starte also den Adapter z.B. alle 15 Minuten automatisch neu. Das ist natürlich keine gute Lösung, aber alle anderen Versuche (mehrfacher Kabeltausch in der Anbindung an mein Heimnetz, Variation der Abrufzeiten, Variation der abgerufenen Elemente uswusw.) haben bei mir nichts nachhaltiges gebracht. 
Wenn Du jetzt einen anderen Lösungsansatz hast, bin ich sehr interessiert und könnte bei Bedarf auch im Rahmen meiner Kenntnisse unterstützen und im Testsystem probieren. Skripten kann ich nur für den einfachen Hausgebrauch, da bin ich ziemlich wenig im Detail.
Beste Grüße
Duejo46

aruttkamp, 2023-10-22:
Ich hatte heute exakt die oben in rot dargestellten Fehlermeldungen. 
Auch in der App wurden keine sinnvollen Daten mehr angezeigt. 
Lösung des Rätsels : Der Wechselrichter hatte offenbar eine neue IP bekommen. Warum auch immer. 
Neue IP konfiguriert - alles wieder gut.

Vermutlich nicht Deine Lösung - aber ich wollte das mal hier festhalten .. wer weiss zu welchen Schlüssen das am Ende führt.

duejo46, 2023-10-23:
Moin, danke für die Info. Dieses Thema hatte ich auch schon mal und seither teile ich via Fritzbox dem WR eine bestimmte IP zu.
Ich kann aktuell berichten, dass seit etwa 2 Wochen der Adapter V1.2.2 sowohl im Test- als auch im Produktivsystem völlig stabil und ohne jede Fehlermeldung läuft. Was habe ich geändert?
Meine beiden Systeme fragen seit jeher parallel ab. Durch das Herumprobieren waren die rct-elements unterschiedlich in Anzahl und Reihenfolge. Ich habe also mal aufgeräumt und die rct-elements geordnet und dann via copy/paste auf beide Systeme eingetragen. Seither ist Ruhe!
Ich habe das bis jetzt einmal verifiziert, also im Testsystem einfach einige elements gelöscht und die Reihenfolge an einer Stelle geändert. Nach etwa einer Stunde kam die bekannte Fehlermeldung. Ursprünglichen Zustand wieder eingetragen, keine Meldungen.
Das ist alles nur eine Beobachtung. Im Moment keine Probleme, und das ist erstmal positiv...
Beste Grüße und weiterhin gutes Gelingen...

Corneys, 2023-10-23:
Guten Morgen, 
direkt nachdem meine WRs installiert wurden habe ich bei mir auch den beiden über die FritzBox feste IPs vergeben.
Bei mir ist das Feld für die rct-elements leer, es werden also die Standard-elemets abgefragt.

Einen Versuch ist es aber mal Wert, dass ich bei mir nur das für mich nötigste abrufe und schaue was passiert.
Ich werde euch berichten.

Corneys, 2023-10-26:
Guten Morgen zusammen,

also an der Reduzierung der rct_elements und dem Abgleichen der beiden Adapter hat bei mir leider keine Besserung gebracht.
Jedoch ist bei mir durch Zufall etwas anderes interessantes passiert.

Nachdem ich vorgestern mit den Elementen rumgetestet habe, vergas ich mein Script wieder zu aktivieren, welches mir meine Adapter zur Dämmerung ein und ausschaltet. 
Und siehe da, der Adapter lief die ganze Nacht ohne einen Verbindungsabbruch durch, was aber für mich sehr seltsam war, da ich den Tag in meinem Produktivsystem nichts verändert habe.
Den Tag über hatte ich nur in meinem Testsystem mit den rct_elements rumgespielt um zu schauen, was mir ausgegeben wird.

In der Regel habe ich die Adapter im Testsystem dieses mal auch vergessen zu deaktivieren.
Und tatsächlich konnte ich gestern das ganze mehrmals reproduzieren. 

Läuft der RCT-Adapter, der auf den selben WR zugreift im Testsystem, funktioniert der Adapter im Produktivsystem, auch Nachts ohne Probleme.
Sobald ich den Adapter im Testsystem deaktiviere dauert es keine 15 Minuten, bis mir der Adapter im Produktivsystem drei mal abschmiert und per Loop-Detection deaktiviert wird.

Erklären kann ich mir das allerdings nicht
Ich hoffe du kannst damit was anfangen :-)

Viele Grüße und einen schönen Tag

aruttkamp, 2024-05-05:
Bitte mit Version 1.2.7 prüfen. Ich könnte mir vorstellen das die Änderungen dazu führen dass dies nicht mehr auftaucht. 
@Corneys @duejo46

Corneys, 2024-05-07:
Guten Abend,

also ich habe die letzen Abende den Adapter auf der Version 1.2.7 laufen lassen und seitdem bricht nachts die Verbindung zum Master Wechselrichter nicht mehr ab!
Fur mich ist das Problem damit also gelöst.
Vielen Dank!

closed 2024-05-07 by Corneys