aruttkamp/ioBroker.rct #53: android_description lässt Adapter abstürzen

aruttkamp, 2023-10-04:
Danke für den Link. Ich werde mir das anschauen. Da sind ja viele Beschreibungen drin, die ich aber nicht alle deuten kann.

> Der Datenpunkt "android_description" gibt den Device Namen aus. Wenn du beiden WR unterschiedliche Namen gegeben hast sollte das darüber ausgegeben werden.

Wenn ich "android_description" hinzufüge, startet die Instanz nicht mehr:

```
2023-10-04 19:20:25.910  - info: rct.0 (24001) starting. Version 1.2.2 in /opt/iobroker/node_modules/iobroker.rct, node: v18.17.1, js-controller: 5.0.12
--
2023-10-04 19:20:25.943  - info: rct.0 (24001) config rct_ip: 192.168.nnn.mmm
2023-10-04 19:20:26.005  - warn: rct.0 (24001) Invalid RCT name: power_mng.use_grid_power_enable
2023-10-04 19:20:26.007  - info: rct.0 (24001) rct state not defined: power_mng.use_grid_power_enable
2023-10-04 19:20:26.015  - error:  rct.0 (24001) Unhandled promise rejection. This error originated either  by throwing inside of an async function without a catch block, or by  rejecting a promise which was not handled with .catch().
2023-10-04 19:20:26.016  - error: rct.0 (24001) unhandled promise rejection: Cannot read properties of undefined (reading 'replace')
2023-10-04 19:20:26.017  - error: rct.0 (24001) TypeError: Cannot read properties of undefined (reading 'replace')
at rct.getStateInfo (/opt/iobroker/node_modules/iobroker.rct/rct/rct.js:35:25)
at Rct.onReady (/opt/iobroker/node_modules/iobroker.rct/main.js:60:26)
2023-10-04 19:20:26.018  - error: rct.0 (24001) Cannot read properties of undefined (reading 'replace')
2023-10-04 19:20:26.026  - info: rct.0 (24001) terminating
2023-10-04 19:20:26.027  - warn: rct.0 (24001) Terminated (UNCAUGHT_EXCEPTION): Without reason
2023-10-04 19:20:26.031  - warn: rct.0 (24001) Read-only state "rct.0.info.connection" has been written without ack-flag with value "false"
2023-10-04 19:20:26.530  - info: rct.0 (24001) terminating
2023-10-04 19:20:26.586  - error: host.iobroker instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
```

_Originally posted by @peperoni60 in https://github.com/aruttkamp/ioBroker.rct/discussions/52#discussioncomment-7189788_

aruttkamp, 2023-10-04:
Absturz ist nachvollziehbar. Bitte den Parameter "android_description" erstmal nicht verwenden

aruttkamp, 2024-07-02:
android_description wird nicht unterstützt !

closed 2024-07-02 by aruttkamp

mcm1957, 2024-07-03:
Na ja
ABSTÜRZEN sollte dercAdaptercaber auch da nicht...

aruttkamp, 2024-07-03:
Wird er auch nicht. Der Parameter fliegt erstmal raus.