aruttkamp/ioBroker.rct #54: g_sync.p_ac_sum_lp aktualisiert nicht sauber

BenS89, 2023-10-16:
**Describe the bug**  
Die Variable rct.0.g_sync.p_ac_sum_lp dient in verschiedenen Skripten als Basis für die Smart Home Steuerung (z.B. Warmwasser-Erzeugung mit PV-Überschuss)

Aus mir nicht bekannten Gründen wird die Variable vom Adapter nicht regelmäßig aktualisiert, wohingegen andere Variablen regelmäßig aktualisiert werden.

2. Fragen: 
1. Tritt das bei euch auch auf?
2. Habt ihr eine Idee, woran das liegen könnte?

Reproduzierbar, indem der Objektbaum im ioBroker einfach eine Weile offen gelassen wird (bei geschlossener RCT App). Sobald ich die RCT App parallel öffne, scheint ein Stream zustandet zu kommen und die besagte Variable wird aktualisiert. Sobald die Ap geschlossen wird, werden die restl. Variablen aktualisiert, nicht aber p_ac_sum_lp 

[image]

aruttkamp, 2023-10-16:
Moin ! 

Da war ich sofort neugierig und hab den Objektbaum beobachtet. 
Der Wert wird regelmäßig aktualisiert. Allerdings nicht so oft wie z.B. p_ac_load_0 / 1 / 2 ..
p_ac_sum_lp aktualisiert etwa jedes 3. oder 4. mal mit. 
Die RCT App habe ich gar nicht geöffnet. 

Andreas

BenS89, 2023-10-16:
Das ist seltsam, ich hatte den Baum jetzt bestimmt eine Stunde auf. Der Wert aktualisiert sich nur, wenn ich den Adapter Neustarte. Hast du eine Idee, woran das liegen könnte?

aruttkamp, 2023-10-16:
Poste mal deine konfigurierten Parameter bitte. 
ich habe den refresh auf 10 stehen. bedeutet natürlich , bevor bei dir 1x aktualisiert wird sind es bei mir schon 6 mal.

BenS89, 2023-10-16:
Aktuell sind diese Werte hier konfiguriert:

`battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,battery.cycles,battery.ah_capacity,battery.stack_cycles[0],battery.stack_cycles[1],battery.stack_cycles[2],battery.stack_cycles[3],battery.stack_cycles[4],battery.stack_cycles[5],battery.stack_cycles[6],battery.soh,dc_conv.dc_conv_struct[0].enabled,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].enabled,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],energy.e_load_day,energy.e_load_month,energy.e_load_total,energy.e_load_total_sum,energy.e_load_year,energy.e_dc_day[0],energy.e_dc_day[1],energy.e_dc_month[0],energy.e_dc_month[1],energy.e_dc_year[0],energy.e_dc_year[1],energy.e_grid_feed_day,energy.e_load_day,energy.e_grid_load_day,energy.e_grid_feed_month,energy.e_grid_feed_year,energy.e_grid_load_month,energy.e_grid_load_year`

BenS89, 2023-10-16:
Ich hab das Objekt mal aus Spaß aus dem Objektbaum gelöscht und anschließend den RCT Adapter neu gestartet. Jetzt ist der Wert null, offenbar bekommt er den Wert nicht mehr. Interessanterweise ist nur dieser Wert betroffen

[image]

aruttkamp, 2023-10-16:
Über welchen Wechselrichter ( Power Storage 10 ?? ) und welche Softwareversion sprechen wir bei dir ?

BenS89, 2023-10-16:
Das ist der DC 6.0; Formware-Version ist die aktuellste, also 2.3.5661
Interessanterweise ging es mit älteren Versionen bisher ohne Probleme, das betroffene Script lief seit ca. 1 Jahr ohne Probleme. Daher fehlt mir derzeit die Erklärung.

aruttkamp, 2023-10-16:
Dann muss ich nochmal nachhaken. Speicher sind bei dir 12.5 ? 
Software 2.3.5661 setze ich auch ein.

BenS89, 2023-10-16:
Ja, auch mit dem aktuellsten BMS (Version 5599)
Unser Stack müsste recht gleich sein

aruttkamp, 2023-10-16:
Ich kann mal versuchen das im Debug nachzuvollziehen. Mir fehlt im Moment nur der Ansatz. Was könnte das verursachen ? 
Forsche gerade bei mir warum sich mein Speicher heute aus dem Netz geladen hat ... ( Kalibirerung ist erst am 31.10. wieder dran ) 
Kämpfe auch noch mit dem Notstrom .. das liegt wohl noch bei RCT als pot. Fehler.

BenS89, 2023-10-16:
Das wäre super, vielen Dank. Soll ich den Adapter mal auf "Debug" stellen und dir das Log zukommen lassen?

Ja, das Thema "Software" ist bei RCT nicht so ohne. Ich hatte ständig SOC-Sprünge (die Batterie sollte ja nie unter 7% und nicht über 97% gehen --> außer wenn kalibriert oder gelalanced wird)....

aruttkamp, 2023-10-16:
> Das wäre super, vielen Dank. Soll ich den Adapter mal auf "Debug" stellen und dir das Log zukommen lassen?

Können wir gerne probieren. Ich hätte dann gerne Log von 10x deinem refresh Wert. 
Das sollte reichen.

BenS89, 2023-10-16:
Hab das Debug Log 2 Minuten laufen lassen:

`
2023-10-16 21:53:58.407 - info: host.iobroker stopInstance system.adapter.rct.0 (force=false, process=true)
--
2023-10-16 21:53:58.410 - info: rct.0 (1754) Got terminate signal TERMINATE_YOURSELF
2023-10-16 21:53:58.411 - info: rct.0 (1754) terminating
2023-10-16 21:53:58.411 - info: rct.0 (1754) Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
2023-10-16 21:53:58.413 - info: rct.0 (1754) RCT: disconnected from server
2023-10-16 21:53:58.414 - warn: rct.0 (1754) Read-only state "rct.0.info.connection" has been written without ack-flag with value "false"
2023-10-16 21:53:58.421 - info: javascript.0 (1016) script.js.PV-Anlage.RCT_Health_Check: exec: iobroker restart rct.0
2023-10-16 21:53:58.452 - info: host.iobroker stopInstance system.adapter.rct.0 send kill signal
2023-10-16 21:53:58.913 - info: rct.0 (1754) terminating
2023-10-16 21:53:58.941 - info: host.iobroker instance system.adapter.rct.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
2023-10-16 21:53:58.994 - info: host.iobroker instance system.adapter.rct.0 started with pid 1779
2023-10-16 21:53:59.344 - debug: rct.0 (1779) Redis Objects: Use Redis connection: 127.0.0.1:9001
2023-10-16 21:53:59.358 - debug: rct.0 (1779) Objects client ready ... initialize now
2023-10-16 21:53:59.358 - debug: rct.0 (1779) Objects create System PubSub Client
2023-10-16 21:53:59.359 - debug: rct.0 (1779) Objects create User PubSub Client
2023-10-16 21:53:59.392 - debug: rct.0 (1779) Objects client initialize lua scripts
2023-10-16 21:53:59.401 - debug: rct.0 (1779) Objects connected to redis: 127.0.0.1:9001
2023-10-16 21:53:59.416 - debug: rct.0 (1779) Redis States: Use Redis connection: 127.0.0.1:9000
2023-10-16 21:53:59.422 - debug: rct.0 (1779) States create System PubSub Client
2023-10-16 21:53:59.422 - debug: rct.0 (1779) States create User PubSub Client
2023-10-16 21:53:59.472 - debug: rct.0 (1779) States connected to redis: 127.0.0.1:9000
2023-10-16 21:53:59.540 - info: rct.0 (1779) starting. Version 1.2.2 (non-npm: aruttkamp/ioBroker.rct) in /opt/iobroker/node_modules/iobroker.rct, node: v18.15.0, js-controller: 5.0.12
2023-10-16 21:53:59.549 - info: rct.0 (1779) config rct_ip: 192.168.178.66
2023-10-16 21:53:59.564 - warn: rct.0 (1779) Invalid RCT name: dc_conv.dc_conv_struct[0].enabled
2023-10-16 21:53:59.565 - info: rct.0 (1779) rct state not defined: dc_conv.dc_conv_struct[0].enabled
2023-10-16 21:53:59.566 - warn: rct.0 (1779) Invalid RCT name: dc_conv.dc_conv_struct[1].enabled
2023-10-16 21:53:59.566 - info: rct.0 (1779) rct state not defined: dc_conv.dc_conv_struct[1].enabled
2023-10-16 21:53:59.579 - info: rct.0 (1779) RCT: connected to server at 192.168.178.66
`

BenS89, 2023-10-16:
Ja, total komisch. Kann es sein, dass die AC-Werte nur dann vom Wechselrichter aktualisiert werden, wenn vom Dach Strom kommt? (das ist jetzt nur ein verzweifelter Guess)--> oder aktualisiert sich der Wert p_ac_sum_lp bei dir jetzt gerade (also abends, wenn kein Strom erzeugt wird) trotzdem?

aruttkamp, 2023-10-16:
Genau.. aktualisiert sich bei mir auch jetzt - und es ist verdammt dunkel draussen ;-)

aruttkamp, 2023-10-16:
Das sollte passen. am "Abruf" der Daten ist auch nichts geändert und es ist für alle Parameter gleich. 
Das macht es so strange

BenS89, 2023-10-16:
echt super komisch; der Wert bleibt bei mir null bis ich die RCT App auf dem Handy öffne (also bis es quasi einen Datenstream zum Handy gibt) und dann den RCT Adapter neu starte. 

Sobald dich die App wieder schließe, aktualisiert die Variable nicht mehr

aruttkamp, 2023-10-22:
Ich habe das nun ein paar Tage beobachtet und die Aktualisierung ist bei mir völlig unabhängig von der APP.

BenS89, 2023-10-23:
> Ich habe das nun ein paar Tage beobachtet und die Aktualisierung ist bei mir völlig unabhängig von der APP.

Danke für‘s Beobachten. Bei mir ist der Wert nach wie voll null, super strange. Ich hab meine Scripte aber inzwischen umgestellt und arbeite jetzt mit der Summe der DC-Werte, die vom Dach kommen. Das klappt sehr gut und ist inhaltlich sogar noch einen ticken präziser.

Mal schauen, ob sich das mit dem nächsten Firmware Update ggf wieder bessert.

closed 2023-10-23 by BenS89