aruttkamp/ioBroker.rct #55: "Warn" "Invalid RCT Name" bei Adapterstart

Madaraxa, 2023-10-22:
Hi.
Admin 6.11.0
NodeJS 18.18.2
RCT Adapter 1.2.2

Beim Start des Adapters kommt im LOG:

`<html>
<body>

rct.0 | 2023-10-22 13:47:54.449 | warn | Invalid RCT name: prim_sm.state
-- | -- | -- | --
rct.0 | 2023-10-22 13:47:54.412 | warn | Invalid RCT name: power_mng.use_grid_power_enable

</body>
</html>`

Auch, nachdem ich alle Objekte einmal gelöscht habe vor dem Adapterstart.
Ist nicht schlimm, da alles funktioniert.
Wollte es nur mitteilen.

aruttkamp, 2023-10-22:
Wenn du die von dir benötigten Parameter konfigurierst und dabei den prim_sm.state weglässt ist auch der Fehler weg. 
ich werde ihn in der nächsten Version in den defaults einfach rausnehmen. Gleiches gilt auch für power_mng.use_grid_power_enable
Sind noch "Relikte" aus Vorversionen.

Madaraxa, 2023-10-22:
Hmm,
wo kann ich Parameter definieren? Bei den "Einstellungen" ist bei mir nichts ausgewählt / eingegeben.
Erste Zeile bei den Einstellungen ist die IP, Darunter der Refresh (10), darunter in grau "rct_elements", da ich dort nichts eingegeben habe. Das wars :)
Ich habe den Adapter allerdings immer "nur" geupdatet und nie gelöscht und neu installiert. Soll ich das mal probieren?

aruttkamp, 2023-10-22:
> darunter in grau "rct_elements", da ich dort nichts eingegeben habe. Das wars :)

Genau da kannst du die Parameter eingeben. Wenn da nix drinsteht nimmt er default und da sind noch die Parameter drin die er anmeckert. 
Nein. du brauchst nicht neu zu installieren.

Madaraxa, 2023-10-22:
Hmm, sorry für die doofe Frage: wie nehme ich denn was raus, was ich nicht eingegeben habe?
muss ich alle "Objekte" eingeben, die er anlegen soll, um den WARN zu entfernen? 
Also "SOC, SOC_target, soc_target_high, ...." und das für Batterie, Wechselrichter, etc?

aruttkamp, 2023-10-22:
Du musst nichts rausnehmen. Der Default wird nur genommen wenn du NICHTS einträgst. 
In dem Moment wo du was einträgst wird genau das genommen was du eingetragen hast.
Du trägst genau die Werte ein die du bekommen möchtest !!

Madaraxa, 2023-10-22:
Das habe ich (denke ich) verstanden. Mit "Werte" meinst du das, was später als "Objekt" angezeigt wird?
Also ich müsste die objekte eingeben, die ich brauche? oder halt leer lassen, damit alle verfügbaren Objekte angezeigt werden (und mit dem "WARN" leben)?

aruttkamp, 2023-10-22:
Nein. nicht ganz. Ich zitiere mal aus der Beschreibung ...
>>Other elements can be found in the code (file "rct/rc_core2.js"). But this is not self descriptive at all (even not really tested)

Dort stehen alle Parameter die aktuell unterstützt werden. Jeder Parameter spiegelt einen Wert wieder, der als Objekt in ioBroker landet. 
Die "Default" Belegung ist eigentlich nur drin , damit auch ohne dass sich jemand damit beschäftigen muss Werte ausgegeben werden.

Madaraxa, 2023-10-22:
Ah top. Vielen, vielen Dank für die Erklärung. 
Hab da gerade mal reingeschaut und das sind ja Unmengen an Daten :) Da passt das Default für meine Zwecke gut :)

closed 2023-10-22 by Madaraxa