aruttkamp/ioBroker.rct #56: Objektbaum "energy" wird nicht aktualisiert bzw. nicht angelegt

petermeter2000, 2023-10-26:
Sämtliche Werte des Objektbaums "energy" werden bei mir seit einiger Zeit nicht mehr aktualisiert. Alle anderen Werte schon.
Habe den Adapter gestoppt, Ordnerbaum RCT/0 gelöscht und den Adapter neu gestartet. Alle Unterordner werden neu erzeugt - bis auf "energy".
Die Werte werden in der RCT Power App richtig angezeigt.
Hat noch jemand das Problem?

 - Adapter version: 1.22
 - RCT-Control version: 2.3.5661
 - RCT BMS version: 5599
 - RCT Power Storage DC 8.0

Danke für Eure Hilfe!

aruttkamp, 2023-10-26:
Bitte poste deine Parameter .. 
Also den Inhalt des Feldes rct_elements aus der Konfiguration.

petermeter2000, 2023-10-26:
[image]

petermeter2000, 2023-10-27:
Irgendeine Idee? Bin ich der Einzige mit dem Problem?
Kann ich noch irgendwelche nützlichen Infos schicken?
rct_elements war und ist leer.
Danke für Eure Hilfe!

NCIceWolf, 2023-10-29:
Hatte ich so auch noch nie. Gib doch die Parameter bei dir mal explizit in den rct_elements an und schau ob sie dann kommen.

petermeter2000, 2023-10-29:
Wie sieht denn die Syntax aus?
Da mir der gesamte Baum fehlt, weiß ich auch nicht mehr, wie die einzelnen Werte dort heißen.

NCIceWolf, 2023-10-29:
Ich zitiere aus der Readme:

Default is "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.use_grid_power_enable,power_mng.u_acc_mix_lp"

petermeter2000, 2023-10-29:
Vielen Dank, ich werde es am Donnerstag ausprobieren!

aruttkamp, 2023-10-29:
> Default is [...]

Und dieser Default wird genutzt , wenn KEIN Eintrag in rct_elements vorgenommen wird. 
Da in diesem default aber nicht ein parameter aus dem Ast "energy" angesprochen wird wird da auch nix aktualisiert. 

Also... rct_elements mit den Parametern füllen die ausgelesen werden sollen - dann klappt das auch. 

Und bevor die Frage kommt... 
Zitat aus Readme:
Other elements can be found in the code (file "rct/rc_core2.js"). But this is not self descriptive at all (even not really tested).

NCIceWolf, 2023-10-29:
Kommt davon wenn man das nur eben am Handy rum kopiert...
Danke für das wache Auge!!

aruttkamp, 2023-10-31:
@petermeter2000 - soweit alles geklärt ?

petermeter2000, 2023-11-01:
Ich konnte es noch nicht testen, da ich erst heute Abend aus dem Urlaub zurück bin :-)
Theoretisch ist mir alles klar, unklar ist mir nur, warum beim erstmaligen installieren des Adapters alle Werte da waren. Und jetzt eben nicht mehr. Ich versuche es heute Abend mal mit der expliziten Eingabe einzelner Werte.
Ich habe auch schon überlegt, ob das daran liegt, dass mein Wechselrichter ausgetauscht werden musste...?
Ich halte Euch auf dem Laufenden. Vielen Dank auf alle Fälle schon mal für Euren Einsatz!!!

petermeter2000, 2023-11-01:
Hallo,
juhu, es hat geklappt. Habe die Default -Werte reinkopiert und noch die "Energy"-Werte aus der rct_core2.js dazu. Wunderbar!
Warum die ursprünglich mal drin waren und jetzt nicht mehr - keine Ahnung...
Nochmals vielen Dank für die Hilfe!!!

aruttkamp, 2023-11-01:
> Hallo, juhu, es hat geklappt. Habe die Default -Werte reinkopiert und noch die "Energy"-Werte aus der rct_core2.js dazu. Wunderbar! Warum die ursprünglich mal drin waren und jetzt nicht mehr - keine [...]

Sehr gut ! 
Dann schliesse ich hier ab ;-)

closed 2023-11-01 by aruttkamp