aruttkamp/ioBroker.rct #63: register prim_sm.state not included anymore

FreakyJ, 2023-11-26:
I am on the latest version 1.2.2 and realized that the register prim_sm.state isn't included anymore on rct_core2.js
https://rctclient.readthedocs.io/en/latest/inverter_registry.html#prim-sm

OID: 0x5F33284E

Can you add it again?

FreakyJ, 2023-12-01:
Yes, I would just return it as an int 0 to 13 like you did in the older version :)

aruttkamp, 2024-02-09:
added with 1.2.3

timoglaess, 2024-02-20:
Unfortunalty, it is still not updating the value (prim_sm.state) on 1.2.24 
--> Object of state "rct.0.prim_sm.state" is missing the required property "common.type"

I tried to manually delete the object and it was automatically re-created by the adapter but still not updating.
--> State "rct.0.prim_sm.state" has no existing object, this might lead to an error in future versions

If I remenber well, "prim_sm.state" was working until ~1.1.7

For me, just the value (1-13) is perfectly ok.
I still have an active JS script, to convert the value to text (inverter status) to show it in the vis.

```
// RCT Power Storage Statusmeldungen in TEXT umsetzen
on({ id: [].concat(['rct.0.prim_sm.state']), change: 'any' }, async (obj) => {
  let value = obj.state.val;
  let oldValue = obj.oldState.val;
  if (getState('rct.0.prim_sm.state').val == 0) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Standby', true);
  }
  if (getState('rct.0.prim_sm.state').val == 1) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Initialization', true);
  }
  if (getState('rct.0.prim_sm.state').val == 2) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Standby', true);
  }
  if (getState('rct.0.prim_sm.state').val == 3) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Efficiency', true);
  }
  if (getState('rct.0.prim_sm.state').val == 4) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Isolationsprüfung', true);
  }
  if (getState('rct.0.prim_sm.state').val == 5) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Inseltest', true);
  }
  if (getState('rct.0.prim_sm.state').val == 6) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Leistungstest', true);
  }
  if (getState('rct.0.prim_sm.state').val == 7) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Symmetrisierung', true);
  }
  if (getState('rct.0.prim_sm.state').val == 8) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Relaistest', true);
  }
  if (getState('rct.0.prim_sm.state').val == 9) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Passivnetz', true);
  }
  if (getState('rct.0.prim_sm.state').val == 10) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Prepare Bat passive', true);
  }
  if (getState('rct.0.prim_sm.state').val == 11) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Battery passive (off-grid)', true);
  }
  if (getState('rct.0.prim_sm.state').val == 12) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Hardwaretest', true);
  }
  if (getState('rct.0.prim_sm.state').val == 13) {
    setState('0_userdata.0.RCT-state-Meldung' /* RCT-state-Meldung */, 'Einspeisung', true);
  }
});
```

aruttkamp, 2024-02-20:
ok. its a general problem with enum type. i fix it in a couple of days.

NCIceWolf, 2024-05-03:
Has this issue been fixed by one of the releases since then?
Can anyone confirm/test this (@timoglaess / @FreakyJ )?

aruttkamp, 2024-05-05:
fixed with 1.2.7

closed 2024-05-05 by aruttkamp

timoglaess, 2024-05-05:
Läuft Tip Top. Vielen Dank Andreas.