aruttkamp/ioBroker.rct #75: Adapter stürzt ab

Floatcher, 2024-01-09:
**Describe the bug**  
Adapter stürzt ab, sobald der RCT Wechselrichter sich Abends automatisch abschaltet. 

**To Reproduce**  
Steps to reproduce the behavior:
Wechselrichter offline setzen

**Expected behavior**  
Wenn der Wechselrichter nicht erreichbar ist, soll der Adapter warten bis dieser wieder online ist und dann neu connecten. ggf. Werte in diesem Zeitraum auf 0 setzen.

**Screenshots & Logfiles**  
rct.0
2024-01-09 07:45:09.491	warn	Read-only state "rct.0.info.connection" has been written without ack-flag with value "false"

rct.0
2024-01-09 07:45:09.482	warn	Terminated (UNCAUGHT_EXCEPTION): Without reason

rct.0
2024-01-09 07:45:09.470	error	Exception-Code: EHOSTUNREACH: connect EHOSTUNREACH 192.168.1.223:8899

rct.0
2024-01-09 07:45:09.470	error	Error: connect EHOSTUNREACH 192.168.1.223:8899 at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1495:16)

rct.0
2024-01-09 07:45:09.469	error	uncaught exception: connect EHOSTUNREACH 192.168.1.223:8899

rct.0
2024-01-09 07:45:06.407	warn	Invalid RCT name: prim_sm.state

rct.0
2024-01-09 07:45:06.395	warn	Invalid RCT name: power_mng.use_grid_power_enable

**Versions:**  
 - Adapter version: <adapter-version>
 - JS-Controller version: <js-controller-version> 
 - Node version: <node-version> 
 - Operating system: <os-name>

**Additional context**  
Add any other context about the problem here.

dermueller98, 2024-01-09:
kommt bei mir auch ab und zu vor, sollte sich der WR nachts aus irgendwelchen Gründen abschalten, mit Speicher bleibt er ja normalerweise immer im Stand-By.
Ich hab mir einfach ein Java-Script geschrieben, das bei Verbindungsfehler des adapters drei mal versucht den adapter neu zu starten und dann den adapter stoppt, wenn es nicht geklappt hat. Ab Sonnenaufgang wird dann wieder regelmäßig ein neustart des adapters versucht, bis die Verbindung wieder hergestellt wurde.

Floatcher, 2024-01-09:
Ich habe keine Batterie. Bei mir schaltet er sich jedes mal ab, wenn die Sonne untergeht. Das mit dem Skript wäre jetzt der Workaround. Aber schöner ist, wenn dies der Adapter selbst in seinem Code beherrscht. Auf die Excption richtig reagiert. Wenn das sogar mit Batterie passiert, würde ich mich über ein entsprechendes Feature freuen :)

aruttkamp, 2024-01-09:
Hallo Floatcher, 

aktuell fehlt mir hier die Zeit das einzubauen. Eine Lösung findest du hier : 
https://github.com/aruttkamp/ioBroker.rct/discussions/27

Problem hatte ich selbst 1 Jahr lang. seit der Speicher dran ist passiert es nicht mehr. Wobei das eher am PowerSwitch liegt als am Speicher ;-) 

Eine Erweiterung in dieser Hinsicht habe ich aktuell nicht in der Planung !

closed 2024-01-16 by aruttkamp