aruttkamp/ioBroker.rct #87: Min SOC Zielwert 2 stellig nach Komma

jomahol, 2024-02-14:
Bisher hat der Min SOC Zielwert (power_mng.soc_min) nur eine Stelle nach dem Komma. Ist es möglich den auch auf 2 Stellen nach dem Komma zu bekommen?

aruttkamp, 2024-02-15:
Moin ! 
Ich denke nicht dass es möglich ist. Die Werte die in der APP eingestellt werden können reichen von 0,01 bis 1,00
wo sollen da weitere Nachkommastellen herkommen ? 
0,07 in der APP ergeben die 7% in der Anzeige im ioBroker. 

Vielleicht erklärst du was du meinst - bitte !

jomahol, 2024-02-15:
Auch Moin.
Wenn ich den Min SOC Zielwert in der RCT Power APP  z.B. auf 0,35 einstelle, kommt im iobroker Objekt 'power_mng.soc_min' als Wert  0,3 an, die zweite Nachkommastelle 5 wird einfach unterschlagen. Eine Prozent Anzeige habe ich hier nicht!

Vielleicht das falsche Objekt?
Den Datenpunkt 'power_mng.soc_min' habe ich der Liste https://rctclient.readthedocs.io/en/latest/inverter_registry.html# entnommen, einen anderen habe ich dafür nicht gefunden. Hab' jetzt aber keine Ahnung inwieweit die Liste noch aktuell bzw. vollständig ist?

Letztendlich geht es mir nur darum dass mein eingestellter Wert auch im iobroker auftaucht.  Ob in % oder mit zweiter Nachkommastelle ist dabei egal.

aruttkamp, 2024-02-16:
Welche Version des Adapters setzt Du ein ?Ich habe das erst in der letzten Version angepasst. 
Wenn es bei dir nicht so angezeigt wird - stopp den adapter - lösch die Objekte und starte den Adapter wieder. dann sollte es passen

jomahol, 2024-02-17:
Uuups echt jetzt? Tatsächlich die 1.4.2  ist mir die doch glatt durch die Lappen gegangen. ;-)  Sorry für die Aufregung... alles gut jetzt und vielen Dank für deine Unterstützung. :-)

closed 2024-02-17 by jomahol