aruttkamp/ioBroker.rct #90: Adapter Absturz bei Anlagen Abschaltung

jomahol, 2024-02-21:
Ich bin da zwar dank Akku nur marginal davon betroffen aber es gibt ja auch Anlagen ohne Akku die sich Nachts abschalten k.A. was da dann passiert.

Schalte ich hier die Anlage ab, erscheinen folgende Fehler im Protokoll, danach ist der Adapter stillgelegt und startet auch nicht neu selbsttätig neu wenn die Anlage wieder eingeschaltet ist:

host.Server(SmartHome)
2024-02-21 10:48:50.921	error	instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
rct.0
2024-02-21 10:48:50.266	error	Exception-Code: ETIMEDOUT: connect ETIMEDOUT 192.168.116.15:8899
rct.0
2024-02-21 10:48:50.265	error	Error: connect ETIMEDOUT 192.168.116.15:8899 at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16)
rct.0
2024-02-21 10:48:50.264	error	uncaught exception: connect ETIMEDOUT 192.168.116.15:8899
host.ServerSmartHome)
2024-02-21 10:48:23.615	error	host.Server(SmartHome) instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
rct.0
2024-02-21 10:48:22.897	error	Exception-Code: ETIMEDOUT: connect ETIMEDOUT 192.168.116.15:8899
rct.0
2024-02-21 10:48:22.896	error	Error: connect ETIMEDOUT 192.168.116.15:8899 at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16)
rct.0
2024-02-21 10:48:22.894	error	uncaught exception: connect ETIMEDOUT 192.168.116.15:8899
host.ServerSmartHome)
2024-02-21 10:47:56.416	error	host.Server(SmartHome) instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
rct.0
2024-02-21 10:47:55.713	error	Exception-Code: ECONNRESET: read ECONNRESET
rct.0
2024-02-21 10:47:55.712	error	Error: read ECONNRESET at TCP.onStreamRead (node:internal/stream_base_commons:217:20)
rct.0
2024-02-21 10:47:55.711	error	uncaught exception: read ECONNRESET

Adapter: 1.4.2
js-controler: 5.0.17
npm: 10.2.5
OS: Windows 11

aruttkamp, 2024-02-21:
Bitte hier schauen : 
https://github.com/aruttkamp/ioBroker.rct/discussions/27

jomahol, 2024-02-21:
OK... prüft alive auch ob die Anlage überhaupt über WLAN erreichbar ist?  Ansonsten würde das Script ja dauerhaft bis zum wieder einschalten im Kreis laufen, oder wie ist das?

aruttkamp, 2024-02-21:
Ich verstehe Deine Frage nicht. 
PING prüft ob eine Adresse erreichbar ist oder nicht und gibt das als Status zurück.
Darauf reagiere ich mit dem Blockly.

jomahol, 2024-02-21:
Nee alles klar mich hat nur der Trigger 'Alive RCT WR' etwas verwirrt, ich wusste nicht  das dies die Ping Abfrage ist, dachte das wäre der Datenpunkt 'system.adapter.rct.0.alive'.

NCIceWolf, 2024-02-21:
> Bitte hier schauen : #27

Workaround gibt es also schon :)