aruttkamp/ioBroker.rct #93: Abfrageintervall wird nicht korrekt eingehalten

BenS89, 2024-02-23:
Kennt ihr das Problem, dass der Adapter manchmal nicht im konfigurierten Intervall neue Daten vom Wechselrichter erhält? 

Beispiel: der Adapter soll alle 30 Sekunden refreshen (rct_refresh=30).
Ich möchte so z.B. unseren Hausverbrauch (g_sync.p_ac_load_sum_lp) jede Minute in meine InfluxDB schreiben. 

Aus mir nicht bekannten Gründen bekommt der Adapter manchmal ganze 20 Minuten keine neuen Werte vom Wechselrichter. Im Log sind keine Fehler zu sehen, auch gibt es keinerlei Netzwerkprobleme.  

In dem Screenshot sieht man z.B., dass nach 21.08 Uhr erst wieder um 21.21 Uhr ein neuer Wert übertragen wurde, obwohl der rct_refresh auf 30 Sekunden steht. Kennt jemand von euch das Problem? Installierte Adapter Version ist 1.2.2

[image]

kbb444, 2024-02-24:
Ich habe auch noch die 1.2.2 installiert und kenne den von Dir beschriebenen Effekt.
Da ich die Daten für das PV-Überschussladen aktuell brauche, überwache ich mit einem Blockly-Script den Zeitstempel.
Falls es hier zu "Hängern" kommt, wird die Instanz neu gestartet.

BenS89, 2024-02-24:
Hab denselben Use Case. Möchte mit dem PV-Überschuss meine Warmwassererzeugung steuern. Durch das unregelmäßige Update der zugehörigen Variablen klappt das aber nicht mehr zuverlässig.

Zuerst hatte ich die Wechselrichter-Firmware im Verdacht. Nach einem Neustart des WR aber bisher keine Besserung. 

Behebt ein Neustart der Instanz das Problem temporär? Das würde dann ja für ein Problem beim Datenstream vom WR zum Adapter sprechen

aruttkamp, 2024-02-24:
Fragen über Fragen....
@BenS89 - Du hast aber schon den Haken gesetzt das auch gleiche Werte aufgezeichnet werden, oder? 
Ja ich weiss - wahrscheinlichkeit ist eher gering.. aber zumindest eine mögliche Fehlerquelle.

@BenS89 - Du schreibst es klappt nicht mehr zuverlässig. Wann klappte es denn zuverlässig ? 
Ich hatte vor ein paar Monaten schon mal getestet und da wurde das Intervall zuverlässig berücksichtigt.

BenS89, 2024-02-25:
Ja, doppelte Werte sind erlaubt. Man sieht schon im Objektbaum, dass die Updates manchmal sehr unregelmäßig kommen (Werte werden in der UI ja grün eingefärbt, sobald sie aktualisiert werden). Das passiert manchmal minütlich so wie es sein soll und manchmal erst alle 10 Minuten. 

Über den Winter habe ich es kaum bemerkt, da ja in den dunklen Monaten ohnehin wenig PV Überschuss zustande kommt. Letztes Jahr im Sommer lief es aber zuverlässig. Ich kann daher keinen exakten Zeitpunkt ausmachen.

Was mir aufgefallen ist: das Verhalten liegt nicht an einzelnen Variablen, sondern ist bei mehreren Werten zu sehen (zB Hausverbrauch, Ladestand Batterie, PV Überschuss)

Sobald ich die RCT App öffne, scheint ein konstanter Datenstream zustande zu kommen. Dann werden alle Werte im Objektbaum im Sekundentakt aktualisiert. 

Sobald die RCT App wieder zu ist, klappen die Intervalle des Adapters manchmal, aber nicht immer. Ein hängen von mehr als 10 Minuten (so wie die letzten Tage) ist mir bisher aber früher noch nicht aufgefallen

NCIceWolf, 2024-02-29:
Kenne ich so bei mir nicht. Man sieht nur dass es ab und an mal etwas "hängt".
Würde ich aber eher auf den WR schieben wollen, oder weil auf dem System gerade viel Last ist.

Was mir aber bei genauerem Hinsehen aufgefallen ist dass manche Werte wohl vom Wechselrichter nicht öfter aktualisiert werden.
Bei mir steht das Intervall z.B. auf 10 Sekunden und bei fast allen Werten macht er das auch so, den "g_sync.p_ac_sum_lp" aktualisiert er aber z.B. nur alle 30 Sekunden, bis dahin gibt es schlicht keinen neuen Wert.

Und ein anderes Thema was noch anschauen/prüfen kann: Der Wechselrichter ist wohl nicht so gut bei parallel Abfragen.
Gibt es also noch eine andere Instanz die den WR abfragt kann es sein dass er den Stream zu diesem Adapter trennt. Würde vor allem Sinn machen wenn gleich alle Werte betroffen sind.

Sonst sieht das eigentlich recht konsistent aus:
[image]

aruttkamp, 2024-02-29:
Es scheint definitiv so zu sein, dass es Probleme mit gleichzeitigen Zugriffen von mehreren Clients auf den WR gibt. 
Grundsätzlich erhält wohl jeder Client jede Antwort eines beliebigen Clients. 
Mir ist noch nicht klar ob und wie ich da was dran machen kann. 
Aktuell habe ich nicht die Zeit dies genau zu untersuchen

NCIceWolf, 2024-03-01:
Ich denke nicht dass du da von Adapter-Seite her etwas tun kannst.
Höchstens eine Prüfung ob denn überhaupt Daten empfangen werden, gekoppelt an einen Reconnect wenn z.B. im Intervall x 5 nix ankommt.

Die Probleme von Ben scheinen mir eher woanders her zu kommen.
@BenS89 Versuch es mal mit der neuen Version, hat aruttkamp seit heute auf stable gesetzt.
Dazu die Fragen:
- Ist dein WR per WLAN oder LAN angebunden?
- Control Software Version ist die aktuelle (2.3.5673)?
- BMS-Version ebenfalls aktuell (5599)?
- Andere Instanzen/Adapter/Geräte die irgendwie auf die Daten des WRs zugreifen?

BenS89, 2024-03-01:
Hallo zusammen,

danke für eure Ideen. Als Client habe ich lediglich die ioBroker Instanz und die iOS App. 

- WR ist über LAN angebunden, es gibt keinerlei Netzwerk-Fehler o.Ä. (weder im ioBroker Log noch im WR)
- Ja, Firmware ist die aktuellste 2.3.5673
- Ja, BMS ist 5599
- Nein, nur 1x ioBroker und 1x iOS App, die lediglich sporadisch mal aufgemacht wird

So gesehen habe ich eigentlich dasselbe Setup wie ihr --> die Hänger sind bei mir auch nicht immer so lang wie in dem Bild. Manchmal hängt es auch nur 2-3 Minuten statt ganze 15-20. Merkwürdig finde ich es in jedem Fall. Kann es ggf. sein, dass sich App und Adapter stören könnten?

aruttkamp, 2024-04-05:
Adapterseitig kann ich an der Thematik nichts tun. Da wäre RCT gefragt ;-) 
Daher schliesse ich dieses Issue

closed 2024-04-05 by aruttkamp