aruttkamp/ioBroker.rct #106: Datatype info

Mutzeli, 2024-04-04:
Hallo zusammen,
ich richte derzeit den tollen RCT Adapter ein, vielen Dank schonmal dafür!

In den logs sehe ich nun Infomeldungen wie diese hier:
rct.0 | 15913 | 2024-04-04 17:04:28.296 | info | State value to set for "rct.0.battery.stored_energy" has to be type "FLOAT" but received type "number"

Das führt zwar nicht zu einem Fehler, aber es ist mir aufgefallen. Vielleicht kann dies ja angepasst werden.
Wenn ich mehr Informationen hierzu liefern soll, bitte mit Hinweis wo ich diese finde.

Besten Gruß

maba-zep, 2024-04-09:
Dasselbe gilt für mich ... hab den Adapter erstma auf WARN gestellt, da sonst die Anzahl der Meldungen zu hoch war ...

modmax, 2024-04-09:
Dasselbe bei mir ... hab den Log-Level auf WARN gesetzt, da sonst zuviele Meldungen kamen.

Ursächlich ist das hier:
Object rct.0.g_sync.p_acc_lp is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
Also müsste alles, was FLOAT/UINT8/UINT16/UINT32 ist auf "number" gesetzt werden, "STRING" sollte in "string" geändert werden.

Mutzeli, 2024-04-10:
Auch bei den energy.xx Variablen tritt dieser Fehler auf.
Der IoBroker seht dies schon selber als Warn-Nachricht.

admin.0 | 2024-04-10 15:13:14.381 | warn | This object will not be created in future versions. Please report this to the developer.
admin.0 | 2024-04-10 15:13:14.379 | warn | Object  rct.0.energy.e_grid_feed_day is invalid: obj.common.type has an invalid  value (FLOAT) but has to be one of number, string, boolean, array,  object, mixed, file, json

aruttkamp, 2024-04-12:
@modmax @Mutzeli 
Könnt Ihr mich bitte mal mit Euren Versionen versorgen ? 
Admin , RCT , NodeJs

DANKE !

modmax, 2024-04-12:
Admin 6.3.16
Rct 1.2.4
JScontroller 5.0.19
NodeJS 18.9.1

modmax, 2024-04-12:
Siehe auch:
https://www.iobroker.net/docu/index-43.htm?page_id=5385&lang=de#Datentypen

Es gibt in Javascript keine Unterscheidung in 8stellige, 16stelluge oder 32stelluge integer oder in float. Es ist alles eine number.

Ebenso sollten nur kleinbuchstaben für die typen verwendet werden ...

MfG modmax

aruttkamp, 2024-04-12:
> Admin 6.3.16
> Rct 1.2.4
> JScontroller 5.0.19
> NodeJS 18.9.1

Wäre es möglich dass Du NodeJS mal updatest ? Ich habe hier aktuell 18.19.0 im Einsatz. 
Auch Dein Admin ist recht alt. Da wäre 6.17.3 aktuell. 

Mein Problem ist - ich kann den beschriebenen Fehler nicht nachvollziehen

aruttkamp, 2024-04-12:
@modmax
Object rct.0.g_sync.p_acc_lp is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json

Also... der Typ der vom Wechselrichter kommt ist FLOAT - das Datenfeld welches im ioBroker angelegt wird ist number. 
bei verwendung einer aktuellen NodeJS Version und des aktuellen Admins.

Mein LogLevel steht auf Silly und ich habe keinerlei solcher Meldungen im LOG

modmax, 2024-04-12:
Ich ipdate mal ... aber aktuell ist mein sydtem gewesen ... weil alles vom stable repo kam und nicht vom beta repo ...
Ich würde auf stabke erwarten, daß es da immer läuft ...beta ist halt beta ...

aruttkamp, 2024-04-12:
NodeJS ist im Stable bei 18.9 ? Und Admin noch bei 6.3 ?

mcm1957, 2024-04-12:
> > Admin 6.3.16
>  Auch Dein Admin ist recht alt. Da wäre 6.17.3 aktuell.

Admin ist aktuell auf 6.13.16 (im stable). Ich vermute mal 6.3.16 ist ein Tippfehler (vergessener 1er)

modmax, 2024-04-12:
Admin war 6.13.16 ... sorry.
NodeJS idt bei mitc18.9.3 ... openduse system  mit deren paketen ....
Ich teste gerade mit neuem admin ...

NCIceWolf, 2024-04-12:
Hab den Fehler bei mir auch nicht, hab extra mal auf Debug hoch gedreht...

Hab folgendes im Einsatz:

starting. Version 1.2.4 in /opt/iobroker/node_modules/iobroker.rct, node: v18.20.2, js-controller: 5.0.19

Plattform: docker (official image - v9.1.2)
Betriebssystem: linux
Node.js: v18.20.2
NPM: 10.5.2

An den Daten kann das eigentlich nicht liegen, den "rct.0.g_sync.p_acc_lp" frage ich bei mir auch ab...
-- 
Gesendet mit pEp für Android.

modmax, 2024-04-12:
Ich habe eigene rct_rlements definiert ...vielleicht liegt es daran ....
Nebenbei noch node auf 18.20 aktualisieren

aruttkamp, 2024-04-12:
@modmax 
>>Ich habe eigene rct_rlements definiert ...vielleicht liegt es daran ....
Poste deine Einträge doch einfach mal hier

Mutzeli, 2024-04-12:
js-controller: 5.0.19
Node.js: v18.20.0
NPM: 10.5.0
admin v6.13.16
rct 1.2.4

Habe die folgenden elements drin, muss aber demnächst mal aussortieren.

energy.e_ac_day,energy.e_dc_day[0],energy.e_dc_day[1],energy.e_grid_feed_day,energy.e_grid_load_day,energy.e_ext_day,energy.e_load_day,energy.e_ac_month,energy.e_dc_month[0],energy.e_dc_month[1],energy.e_grid_feed_month,energy.e_grid_load_month,energy.e_ext_month,energy.e_load_month,energy.e_ac_total,energy.e_dc_total[0],energy.e_dc_total[1],energy.e_grid_feed_total,energy.e_grid_load_total,energy.e_ext_total,energy.e_load_total,energy.e_ac_year,energy.e_dc_year[0],energy.e_dc_year[1],energy.e_grid_feed_year,energy.e_grid_load_year,energy.e_ext_year,energy.e_load_year,battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.use_grid_power_enable,power_mng.u_acc_mix_lp,prim_sm.island_flag,battery.stored_energy,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_sum,g_sync.p_ac_load_sum_lp,power_mng.bat_next_calib_date,battery.used_energy,battery.stored_energy,battery.efficiency

maba-zep, 2024-04-12:
So .. nun auch auf NodeJS 18.20.0 geupdated.
Rechner neu gestartet; alles States untere rct.0 gelöscht und rcp-Adapter neu gestartet.
Ergebnis: Es hat sich nichts geändert.

Meine rct_elements sind:
battery.bat_status,battery.soc,battery.soc_target,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,energy.e_grid_feed_day,energy.e_grid_feed_month,energy.e_grid_feed_year,energy.e_grid_feed_total,energy.e_grid_load_day,energy.e_grid_load_month,energy.e_grid_load_year,energy.e_grid_load_total,energy.e_load_day,energy.e_load_month,energy.e_load_year,energy.e_load_total,energy.e_ac_day,energy.e_ac_month,energy.e_ac_year,energy.e_ac_total,dc_conv.dc_conv_struct[0].p_dc,dc_conv.dc_conv_struct[1].p_dc

Ich weiß ja nicht, woher die Datentypen kommen .. aber es sind die, die in der Datei rct_core2.js stehen ...

modmax, 2024-04-12:
So .. nun auch das System auf NodeJs 18.20.0 geupdated, die Datenpunkte rct.0 gelöscht und iobroker neu gestartet.
Ergebnis: Es bleibt bei dem Fehler

Datenpunkte in rct_elements sind:
battery.bat_status,battery.soc,battery.soc_target,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,energy.e_grid_feed_day,energy.e_grid_feed_month,energy.e_grid_feed_year,energy.e_grid_feed_total,energy.e_grid_load_day,energy.e_grid_load_month,energy.e_grid_load_year,energy.e_grid_load_total,energy.e_load_day,energy.e_load_month,energy.e_load_year,energy.e_load_total,energy.e_ac_day,energy.e_ac_month,energy.e_ac_year,energy.e_ac_total,dc_conv.dc_conv_struct[0].p_dc,dc_conv.dc_conv_struct[1].p_dc

Im Ergebnis werten die Datenpunkte mit den Typen gesetzt, wie sie auch in rtc_core2.js beschrieben sind.

mcm1957, 2024-04-13:
Also ich habe den RCT grad ERSTMALs auf einem System installiert. Da ich keine passende HW habe sehe ich nur States die der Adapetr anlegt:

```
![image](https://github.com/aruttkamp/ioBroker.rct/assets/28575778/03a0069f-c100-4be6-9396-3551c45d3535)
```

UND da wird schon mal ein common,type UINT32 angelegt den es nicht gibt :-(

mcm1957, 2024-04-13:
@modmax 
I would suggest to check you states. most likely an invalid type will be used there too.

mcm1957, 2024-04-13:
The adapter creates many invalid states at first startup:

```
rct.0 | 2024-04-13 07:29:06.365 | warn | Read-only state "rct.0.info.connection" has been written without ack-flag with value "false"
rct.0 | 2024-04-13 07:29:06.350 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
rct.0 | 2024-04-13 07:29:06.349 | info | terminating
rct.0 | 2024-04-13 07:29:06.341 | error | Exception-Code: EHOSTUNREACH: connect EHOSTUNREACH 10.17.2.26:8899
rct.0 | 2024-04-13 07:29:06.340 | error | Error: connect EHOSTUNREACH 10.17.2.26:8899 at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16)
rct.0 | 2024-04-13 07:29:06.339 | error | uncaught exception: connect EHOSTUNREACH 10.17.2.26:8899
rct.0 | 2024-04-13 07:29:03.252 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:03.251 | warn | Object rct.0.prim_sm.island_flag is invalid: obj.common.type has an invalid value (UINT16) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:03.203 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:03.202 | warn | Object rct.0.power_mng.u_acc_mix_lp is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:03.172 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:03.172 | warn | Object rct.0.power_mng.state is invalid: obj.common.type has an invalid value (UINT8) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:03.151 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:03.150 | warn | Object rct.0.power_mng.is_heiphoss is invalid: obj.common.type has an invalid value (UINT8) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:03.110 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:03.110 | warn | Object rct.0.io_board.s0_external_power is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:03.073 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:03.072 | warn | Object rct.0.g_sync.u_sg_avg_1 is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:03.040 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:03.039 | warn | Object rct.0.g_sync.u_sg_avg_0 is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:03.017 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:03.016 | warn | Object rct.0.g_sync.p_acc_lp is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:02.995 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:02.994 | warn | Object rct.0.g_sync.p_ac_sum_lp is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:02.969 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:02.968 | warn | Object rct.0.g_sync.p_ac_load_sum_lp is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:02.945 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:02.945 | warn | Object rct.0.g_sync.p_ac_grid_sum_lp is invalid: obj.common.type has an invalid value (FLOAT) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:02.909 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:02.909 | warn | Object rct.0.fault.3_flt is invalid: obj.common.type has an invalid value (UINT32) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:02.889 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:02.888 | warn | Object rct.0.fault.2_flt is invalid: obj.common.type has an invalid value (UINT32) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:02.870 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:02.870 | warn | Object rct.0.fault.1_flt is invalid: obj.common.type has an invalid value (UINT32) but has to be one of number, string, boolean, array, object, mixed, file, json
[... cut ...]
rct.0 | 2024-04-13 07:29:02.675 | warn | This object will not be created in future versions. Please report this to the developer.
rct.0 | 2024-04-13 07:29:02.674 | warn | Object rct.0.battery.bat_status is invalid: obj.common.type has an invalid value (INT32) but has to be one of number, string, boolean, array, object, mixed, file, json
rct.0 | 2024-04-13 07:29:02.605 | info | config rct_ip: 10.17.2.26
rct.0 | 2024-04-13 07:29:02.580 | info | starting. Version 1.2.4 in /opt/iobroker/node_modules/iobroker.rct, node: v18.20.1, js-controller: 5.0.19

```

modmax, 2024-04-13:
Alle number states sind ungültig, wenn sie vom adapter angelegt werden. Da hilft nur manuelles ändern des typs.

mcm1957, 2024-04-13:
Das ist ein massiver Fehler im Code:

States werden hier angelegt:
https://github.com/aruttkamp/ioBroker.rct/blob/7cb51e8ad4cdde8a23467f886e0572a06a0165bf/main.js#L77

Und als Typ werden die Definitionen von hier verwendet:
https://github.com/aruttkamp/ioBroker.rct/blob/7cb51e8ad4cdde8a23467f886e0572a06a0165bf/rct/rct_core2.js#L691

Der Datentyp der für die externe Kommunikation verwendet wird ist NICHT geeignet als common.type verwendet zu werden.

@aruttkamp 
Zum Testen bitte den Adapter mal am Testsystem KOMPLETT löschen und auch alle Objekte löschen und dann neu installieren.
Ev. auch zunächst ohne Verbindung mit der Hardware (= mit unbenutzer IP addresse).

NCIceWolf, 2024-04-13:
Das macht dann sogar Sinn...
@aruttkamp: wir hatten doch irgendwann mal die Datentypen manuell angepasst wegen einem anderen Thema.
Deswegen haben wir den Fehler nicht!

aruttkamp, 2024-04-17:
>>Das` macht dann sogar Sinn...
>>@aruttkamp: wir hatten doch irgendwann mal die Datentypen manuell angepasst wegen einem anderen Thema.
>>Deswegen haben wir den Fehler nicht!`
Du irrst @NCIceWolf - Änderungen ja ... aber ich hab den Verdacht das es hier Änderungen am ioBroker gibt der jetzt meckert, weil.. seit ich den Adapter betreue muss er schon die "falschen" Datentypen anlegen. 
Genauso wie die anderen Dinge die da grad hochpoppen. 
Ich kümmere mich darum. Braucht aber ein wenig zeit.

NCIceWolf, 2024-04-18:
Ja klar, den Fehler gibt es sicher schon seit Erstellung des Adapters.
Durchaus möglich dass ein Update von Javascript, Admin oder dem js-controller zu gab.

Ich meinte übrigens dieses Thema hier, da waren aber strings anstatt numbers hinterlegt:
https://github.com/aruttkamp/ioBroker.rct/issues/16

OnkelFD, 2024-04-28:
Hier passt die Rolle leider auch nicht.

State value to set for "rct.0.power_mng.soc_min" has to be type "FLOAT" but received type "number"

NCIceWolf, 2024-04-29:
> Das ist ein massiver Fehler im Code:
>
> States werden hier angelegt:
>
> https://github.com/aruttkamp/ioBroker.rct/blob/7cb51e8ad4cdde8a23467f886e0572a06a0165bf/main.js#L77
>
> Und als Typ werden die [...]

Bin jetzt nicht so tief in JS drin, aber dann in etwa so?

Anstatt 
https://github.com/aruttkamp/ioBroker.rct/blob/7cb51e8ad4cdde8a23467f886e0572a06a0165bf/main.js#L77

das hier:
if (rct.cmdReverse[rct_id].type == STRING) {
	const type = string;
} else {
	const type = number;
}

Oder die ganz billige Klamotte:
const type = mixed;

Wäre ja eine einfache Lösung 😄

mcm1957, 2024-04-29:
> Du irrst @NCIceWolf - Änderungen ja ... aber ich hab den Verdacht das es hier Änderungen am ioBroker gibt der jetzt meckert, weil.. seit ich den Adapter betreue muss er schon die "falschen" [...]

Ja da gibt es sicher Änderungen in js-controller 5. Fehlerhafte Objekte (und das sind z,B. States mit einem nicht existenten Typ) werden zwischenzeitlich angemahnt und voraussichtlich mit js-controller 6 dann nicht mehr angelegt. Wobei die letztere Aussage völlug unverbindlich nur meinen Wissensatnd wiedergibt.

mcm1957, 2024-04-29:
> Oder die ganz billige Klamotte: const type = 'mixed';
> Wäre ja eine einfache Lösung 😄

Ist hoffentlich nur mit dem Smily als Scherz gemeint.

Warum nicht einfach in der Tabelle den Typ für den State ablegen und verwenden? Macht mehr Sinn als irgendwelche on-the-fly Konverteierungen die dann bei einer Anpssung auf einmal nicht mehr passen,,,

aruttkamp, 2024-04-29:
Ich kümmere mich darum ;-)

aruttkamp, 2024-04-30:
Die Tests mit den Types sind durch. das läuft jetzt wie es sollte. Datentypen quelle und Datentypen Ziel sind unterschiedlich handelbar und es kommen keine Fehler mehr beim Anlegen der Felder im ioBroker
Vermutlich gibts heute abend eine neue Version. Schaue mir noch #114 an.

closed 2024-05-03 by aruttkamp