aruttkamp/ioBroker.rct #107: [BUG] Read-only state "rct.0.info.connection" has been written without ack-flag with value "false"

mcm1957, 2024-04-13:
**Describe the bug**  
Read-only state "rct.0.info.connection" has been written without ack-flag with value "false"

**To Reproduce**  
Configure an instance and use an ip-address which is unused 

**Expected behavior**  
info.connectio  shoudl always be set correctly

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: 1.2.4
 - JS-Controller version: 5.0.19
 - Node version: 18.20.1
 - Operating system: Raspbian

**Additional context**  
Add any other context about the problem here.

NCIceWolf, 2024-05-03:
Auch dieses Thema sollte mit 1.2.5 erledigt sein.
Bitte um kurze Bestätigung 😃

aruttkamp, 2024-05-03:
Taucht nicht mehr im Log auf !

closed 2024-05-03 by aruttkamp