aruttkamp/ioBroker.rct #108: [BUG] Adapter crashes if host unreachable

mcm1957, 2024-04-13:
**Describe the bug**  
Adapter crashes if IP not reachable

```
rct.0 | 2024-04-13 07:30:19.628 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
rct.0 | 2024-04-13 07:30:19.627 | info | terminating
rct.0 | 2024-04-13 07:30:19.619 | error | Exception-Code: EHOSTUNREACH: connect EHOSTUNREACH 10.17.2.26:8899
rct.0 | 2024-04-13 07:30:19.618 | error | Error: connect EHOSTUNREACH 10.17.2.26:8899 at TCPConnectWrap.afterConnect [as oncomplete] (node:net:1555:16)
rct.0 | 2024-04-13 07:30:19.617 | error | uncaught exception: connect EHOSTUNREACH 10.17.2.26:8899
rct.0 | 2024-04-13 07:30:16.448 | info | config rct_ip: 10.17.2.26
rct.0 | 2024-04-13 07:30:16.417 | info | starting. Version 1.2.4 in /opt/iobroker/node_modules/iobroker.rct, node: v18.20.1, js-controller: 5.0.19
host.pi4 | 2024-04-13 07:30:14.034 | info | instance system.adapter.rct.0 started with pid 21396
host.pi4 | 2024-04-13 07:29:43.941 | info | Restart adapter system.adapter.rct.0 because enabled
```

**To Reproduce**  
Configure a new instance and use an ip-address which is unused.
Adapter crashes.

NOTE: IP 10.17.2.26 is not used, no device reacts at this address.

**Expected behavior**  
There shouldn't be a crash but a clear error message (EHOSTUNREACH is sufficient) followed by retrys at some intervalls.

**Screenshots & Logfiles**  
If applicable, add screenshots and logfiles to help explain your problem.

**Versions:**  
 - Adapter version: 1.2.4
 - JS-Controller version: 5.0.19
 - Node version: 18.20.1
 - Operating system: Raspbian

**Additional context**  
Add any other context about the problem here.

````

NCIceWolf, 2024-05-03:
> **Describe the bug** Adapter crashes if IP not reachable
>
> ```
> rct.0 | 2024-04-13 07:30:19.628 | warn | Terminated (UNCAUGHT_EXCEPTION): Without reason
> rct.0 | 2024-04-13 07:30:19.627 | info | [...]

Bitte auch dieses Thema mit der neuen Version 1.2.5 noch einmal testen.
Sollte damit auch nicht mehr auftreten, bzw. eine Error-Message erzeugen.

aruttkamp, 2024-05-03:
Fehler wird in Log Datei vermerkt. Kein Absturz mehr. 
Meldung lautet : RCT: connection error, please check ip address and network!

closed 2024-05-03 by aruttkamp