aruttkamp/ioBroker.rct #114: [BUG] loosing of

mosandlt, 2024-04-26:
The rct adapter within iobroker is losing connection without logging sporadically and receives then only a short amount of data.

DK4NS, 2024-04-26:
Hallo. 
Das kann ich leider bestätigen. Der Adapter aktualisiert nur noch sporadisch. Kann es mir momentan nicht erklären. 
Setup: 
RCT v.1.2.4
iobroker v 6.13.16
Node.js : v 18.20.2
NPM: v 10.5.0

Habe den Adapter gelöscht und neu installiert.  Leider ohne Erfolg. Im Log sieht man neben den bekannten Warnungen zu den falschen Datenformaten einige Aktualisierungen und dann schweigt der Adapter. Ebenfalls wenn ich den Adapter neu starte...Läuft ca. 1 Minute mit eingestellten Intervall von 10 Sekunden. Sporadisch dann wieder einige Aktualisierungen nach ein paar Minuten.

mosandlt, 2024-04-26:
Welche Version vom wechselrichter ? Evtl liegt dort das Problem

mosandlt, 2024-04-27:
aktueller workaround.
- adapter Feld manuell auf schreibbar eingestellt
- adapter cron restart alle 2 minuten.

so bekomme ich wenigstens ein paar logs

habe das logging auf warn umgestellt um weniger logs zu bekommen.

duejo46, 2024-04-27:
Hallo,
vor einer Woche habe ich im Rahmen einer Speicherergänzung die WR-Firmware 2.3.5674 aufgespielt bekommen, vorher war die Version 2.3.5661 drauf. Seither habe ich das oben beschriebene Problem. Der WR arbeitet normal, in der rct-power App kann ich verfolgen, dass die Daten übermittelt und aktualisiert werden. Beim Start des iob-rct Adapters ist die Aktualisierung eine Zeitlang synchron, dann aktualisiert der iob nach recht kurzer Zeit nicht mehr. Beispielsweise gut am Farbumschlag in den "iob objecten" zu beobachten. Keine Fehlermeldung dazu in den logs. 
RCT Adapter V1.2.4, Nodejs 18.20.2, NPM 10.5.0, Admin 6.13.16, Pi4-System, alles aktuell ohne Probleme am laufen.
Ping auf dem RCT Adapter überwacht Verbindung, keine Fehlermeldung. 
Workaround: automatischer Neustart des Adapters alle 2 Minuten
Hat jemand eine Idee?
BG duejo

DK4NS, 2024-04-27:
Wechselrichter PS 8.0 FW: 2.3.5674 bisher die aktuelle laut der App.

Komisch ist auch, dass seit Aktualisierung der FW die APP ebenfalls sporadisch lange braucht um Daten zu erhalten. Sieht man schön oben links an dem fehlenden grünen Rahmen um das RCT Symbol. Dauert dann manchmal bis zu 15 Sekunden. Kann man auch im Verbindungsstatus unter "Geräteauswahl" in der App sehen. Da ist dann ein roter Punkt.

Hatte aber den Adapter letze Woche eigentlich mehrere Tage ohne Schwierigkeiten in Betrieb, da ich gerade an der Erstellung einer Überschussladeregelung für das BEV dran bin. Berechne mir daher die gleitenden Mittelwerte von Erzeugung, Verbrauch usw. Da lief es eigentlich. Allerdings hatte ich mir nicht den ganzen Log angeschaut. Ich hatte es dann gestern bemerkt. Da war der PI aber einige Tage aus.... Also gestern frisch gestartet. Warum soll das jetzt auf einmal nicht mehr funktionieren? 

Werde mal den die Verbindung zum RCT Portal probeweise ausschalten in der App und den Wechselrichter neu starten. Vielleicht ballert ja RCT den Socket im Wechselrichter zu....

mosandlt, 2024-04-27:
Ich glaube der Fehler ist in der RCT Firmware 

es wurden Pakete aktualisiert. Hatte dir releasenotes per Mail bekommen 

ebenfalls stimmt die Berechnung der Pv erzeugten Leistung in der Firmware nicht mehr 

Glaube ein Neustart hilft ein paar Wochen. 

Kurzes Update bitte wegen Portal deaktivierung.

mosandlt, 2024-04-27:
Ich glaube der Adapter hängt sich intern auf wenn er kaputte Datensätze bekommt. Evtl kann man reinprogrammieren das man diese einfach verwirft und weiter arbeitet

DK4NS, 2024-04-27:
Danke erst einmal für die Kommentar. Gut das man damit nicht allein ist. Man weiß ja nie, ob das Problem nicht bei einem selbst zwischen den Ohren sitzt ;-)
Verbindung zum Portalserver getrennt. WR flasht und startet neu. Und siehe da - auf einmal läuft es. 
PI lasse ich jetzt mal laufen und prüfe zwischendurch die Logs. Momentan aktualisiert er wieder im angegebenen Refrehintervall von 10 Sekunden. Kann aber noch nicht sagen, ob stabil. Melde mich wieder.

NCIceWolf, 2024-04-27:
Moin!

Ich hatte das ebenfalls bemerkt, seit Mittwoch Abend (ca. 16 Uhr) sah die Statistik in Grafana sehr komisch aus. Da kamen valide Daten nur noch alle 5-10 Minuten an.
Hatte gestern zwar noch etwas gespielt, aber noch keine Lösung gefunden.

Hab jetzt ebenfalls probiert die Verbindung zum RCT Portal zu kappen und siehe da: läuft bis jetzt!

Wäre also erst einmal ein Workaround.
Die Frage ist wie man den Adapter gegen so etwas schützen kann...

VG
Matthias 
-- 
Gesendet mit pEp für Android.

duejo46, 2024-04-27:
Bin gespannt, was wir da noch an Erkenntnissen finden...
Mein Pi läuft 24/7, die RCT App nutze ich normalerweise gar nicht. Aufgefallen ist mir dieses "nicht aktualisieren" erstmals über meine Überschussladung: das Skript hat einfach die Ladeleistung nicht mehr nachgeführt. Kann es ja auch nicht, da quasi ein konstanter (nicht mehr aktueller) Eingangswert verarbeitet wurde. Mit der FW 2.3.5661 habe ich dieses Verhalten seit Frühjahr 2023 (Erstinbetriebnahme) nicht beobachtet. Insofern besteht da für mich ein Zusammenhang zur FW 2.3.5674.
Beobachtet habe ich auch, dass nach längerem "Nicht-Aktualisieren" stochastisch mal wieder eine Zeitlang aktualisiert wurde...
@NCIceWolf Wie hast du die verbindung zum Portal gekappt? Ich habe gestern mal den Haken in den Netzeinstellungen der App weggenommen. Im Portal waren daraufhin keine Daten mehr, das Aktualisierungsproblem bestand aber weiter...

DK4NS, 2024-04-27:
Hatte gestern probeweise nur den WR neu gestartet in dem ich in den Einstellungen das "Displayblinken" deaktiviert hatte. Dann erzwingt man ja einen  Flashvorgang mit Neustart. (Keine Ahnung wozu man dieses Feature "Blinken" benötigt???). Der Portalzugang war da aber noch aktiv. Es hatte auch nichts gebracht. Adapter hing sich wieder auf. 
Weiß jemand, wie man ein Ticket bei RCT diesbezüglich erstellt? Danke.

DK4NS, 2024-04-27:
> Bin gespannt, was wir da noch an Erkenntnissen finden... Mein Pi läuft 24/7, die RCT App nutze ich normalerweise gar nicht. Aufgefallen ist mir dieses "nicht aktualisieren" erstmals über meine [...]

Hat der WR danach den Flashvorgang gestartet?

duejo46, 2024-04-27:
Habe nicht explizit darauf geachtet. Aufgefallen ist es mir nicht.

NCIceWolf, 2024-04-27:
Genau, den Haken in den Netzwerkeinstellungen raus. Dann aber nicht vergessen auf "Next" und dann "Finish" gehen.
Sonst speichert er das nicht.

Seitdem läuft es bei mir wieder absolut sauber (knapp über eine Stunde).
Vorher hat er schon nach spätestens 2 Minuten nichts sinnvolles mehr bekommen.
-- 
Gesendet mit pEp für Android.

duejo46, 2024-04-27:
Bin mir eigentlich sicher, dass ich das gemacht habe. Aber egal, wenn es damit bei euch mit dieser Massnahme stabil läuft, probiere ich das bei Gelegenheit (morgen) auch gerne nochmal.

DK4NS, 2024-04-27:
Also alles zurück auf Los. Nach gut zwei Stunden hängt sich der Adapter wieder auf. Kommen keine aktuellen Werte mehr. Workaround mit Deaktiverung des RCT Portals half nur kurzzeitig.

mosandlt, 2024-04-27:
Dann warten RCT Robe neue Firmware bringt. Reporter jemand den bug bei denen ?

ubf evtl. In den Adapter ei en workaround einprogrammieren?

bei evcc funktioniert es ja auch irgendwie

NCIceWolf, 2024-04-27:
[image]

Seit der Deaktivierung des Portals läuft es immer noch gut. Hätte man spätestens beim Auto-Laden gemerkt.

Ich meine ein Ticket kann man bei RCT erstellen indem man denen ne Mail schreibt ("service@rct-power.com").

aruttkamp, 2024-04-27:
ey ey ey ... hier ist ja was los. 
Ich muss gestehen , dass ich die letzten Tage nur sporadisch zu Hause war. Meine Anzeige hier im Wohnzimmer wo ich mir den aktuellen Hausverbrauch sowie die aktuelle Erzeugung anzeigen lasse hat mich bisher nicht stutzig gemacht. Immer wenn ich geguckt habe waren plausible Zahlen da. 
Da es am Adapter und an der RCT App kein Update gab scheint es so zu sein dass die Version der WR Software das Problem darstellt. @NCIceWolf - kannst Du sagen ob es seit dem einspielen der SW so ist ? Du weisst doch bestimmt wann du die installiert hast ....

mosandlt, 2024-04-28:
Mit deaktivierten Portal sieht’s besser aus. Bei mir ein paar Wochen nach dem letzten Firmware Update

NCIceWolf, 2024-04-28:
> ey ey ey ... hier ist ja was los. Ich muss gestehen , dass ich die letzten Tage nur sporadisch zu Hause war. Meine Anzeige hier im Wohnzimmer wo ich mir den aktuellen Hausverbrauch sowie die [...]

Ich schau heute mal was zu dem Zeitpunkt passiert ist, vielleicht ist aus dem iobroker Log was auszulesen.
Die Firmware hab ich auf jeden Fall schon viel länger drauf, an der alleine kann es nicht liegen.
Meine erste Vermutung ging eher in Richtung JavaScript oder so, da ich das "latest" repository nutze.

Bis jetzt läuft es übrigens auch weiterhin einwandfrei, nachdem ich die Verbindung zum Portal gestern gekappt habe. Vielleicht also doch eine Änderung bei RCT, da können wir uns hier natürlich einen Wolf suchen...

DK4NS, 2024-04-28:
Leider läuft es trotz Abschalten zum Portalserver nicht stabil. Habe in den Aktualisierungszeiten der RCT Objekte teilweise große Lücken. Jetzt aktuell hängt er wieder seit über 15 Minuten.

@NCIceWolf 
Welche JS Version wird genutzt?

OnkelFD, 2024-04-28:
Stürzt bei mir jetzt nicht mehr ab allerdings starte ich den Adapter immer neu wenn sich länger als eine Minute der wert energy.e_load_total nicht geändert hat.

Ist in letzter zeit nicht so oft vorgekommen im Moment macht er dies aber alle 5-10 Minuten.

rct v1.2.4
iobroker v6.13.16
nodejs v18.20.2
js-controller: 5.0.19
npm v10.5.0

NCIceWolf, 2024-04-29:
> Leider läuft es trotz Abschalten zum Portalserver nicht stabil. Habe in den Aktualisierungszeiten der RCT Objekte teilweise große Lücken. Jetzt aktuell hängt er wieder seit über 15 Minuten.
> 
> @NCIceWolf Welche JS Version wird genutzt?

Meine Versionen:
Plattform: docker (official image - v9.1.2)
ioBroker Admin; 6.17.3
Node.js: v18.20.2
js-controller: 5.0.19
NPM: 10.5.2
Javascript: 8.0.3

RCT Adapter: v1.2.4
RCT Control Software: 2.3.5674
RCT BMS: 5599

Im Log finde ich zum fraglichen Zeitpunkt (24.04.2024 ca. 20Uhr) rein gar nichts. An dem Tag habe ich zwar um 10Uhr Javascript auf die 8.0.3 gehoben, die Ausfälle sehe ich in Grafana aber erst ab ziemlich genau 19:50Uhr.
Zum Test habe ich die Portal-Verbindung vorhin noch einmal aktiviert -> Nach ca. 4 Minuten kamen wieder keine Daten...

Ich gehe also nach wie vor davon aus dass hier an der Anbindung WR <-> RCT-Server etwas angepasst wurde und daher die aktiven Datenströme beeinflusst werden.
Ein Neustart der Verbindung löst das Problem dann natürlich kurzfristig, bei länger bestehender Verbindung kommt das jedoch immer wieder.

@aruttkamp: Ich hatte ohnehin schon einmal die Idee ob es nicht cleverer wäre die Datenverbindung nicht dauerhaft aktiv zu lassen, sondern nach erfolgtem Datenabruf wieder zu kappen und beim nächsten Intervall neu aufzubauen.
Laut Log des Brokers und den Einträgen in der influxDB sind das so um die 0,2 Sekunden.
Man könnte die "netConnection" also nach 2 Sekunden kappen und bei Ablauf des Intervalls neu starten.

DK4NS, 2024-04-29:
@NCIceWolf  
Logge jetzt auch mit fluxdb/Grafana und sehe nun kleine Lücken bis zu <2 Minuten. Scheint sich etwas beruhigt zu haben. 

Wenn man den Log-Lever auf Debug stellt, sieht man dann die Terminierungen von ioBroker.

Wie hast du das mit dem Intervall-Öffnen der Datenverbindung hinbekommen? Möchte das Portal eigentlich wieder aktivieren. Meine bessere Hälfte ist Controllerin und zieht sich immer die Daten für die Umsatzsteuererklärung.....

Danke und Grüße

rct.0
2024-04-29 17:09:01.602	debug	Objects client ready ... initialize now
rct.0
2024-04-29 17:09:01.584	debug	Redis Objects: Use Redis connection: 127.0.0.1:9001
host.raspberrypi
2024-04-29 17:09:00.934	info	instance system.adapter.rct.0 started with pid 15399
admin.0
2024-04-29 17:09:00.522	info	<== Disconnect system.user.admin from ::ffff:192.168.178.106
host.raspberrypi
2024-04-29 17:08:58.411	info	instance system.adapter.rct.0 terminated with code 11 (ADAPTER_REQUESTED_TERMINATION)
rct.0
2024-04-29 17:08:58.387	info	terminating
host.raspberrypi
2024-04-29 17:08:57.913	info	stopInstance system.adapter.rct.0 send kill signal
rct.0
2024-04-29 17:08:57.897	warn	Read-only state "rct.0.info.connection" has been written without ack-flag with value "false"
rct.0
2024-04-29 17:08:57.892	info	RCT: disconnected from server
rct.0
2024-04-29 17:08:57.886	info	Terminated (ADAPTER_REQUESTED_TERMINATION): Without reason
rct.0
2024-04-29 17:08:57.885	info	terminating
rct.0
2024-04-29 17:08:57.873	info	Got terminate signal TERMINATE_YOURSELF
host.raspberrypi
2024-04-29 17:08:57.854	info	stopInstance system.adapter.rct.0 (force=false, process=true)
admin.0
2024-04-29 17:08:46.585	info	==> Connected system.user.admin from ::ffff:192.168.178.106

NCIceWolf, 2024-04-30:
> @NCIceWolf Logge jetzt auch mit fluxdb/Grafana und sehe nun kleine Lücken bis zu <2 Minuten. Scheint sich etwas beruhigt zu haben.
> 
> Wenn man den Log-Level auf Debug stellt, sieht man dann die Terminierungen von ioBroker.
> 
> Wie hast du das mit dem Intervall-Öffnen der Datenverbindung hinbekommen? Möchte das Portal eigentlich wieder aktivieren. Meine bessere Hälfte ist Controllerin und zieht sich immer die Daten für die Umsatzsteuererklärung.....
> 
> Danke und Grüße
> 

Hab mir dazu eine fork erstellt und in der einmal von "Timeout" auf "Interval" für die Abfragen umgestellt.
Und dazu wird 1s nach jeder Abfrage die TCP-Verbindung abgebaut und erst mit dem neuen Intervall wieder aufgebaut.

Für mich funktioniert das so zwar sehr gut, aber ob @aruttkamp das Konzept des Datenabrufs generell überarbeiten will, muss er entscheiden.
Dazu kommen ja noch viele Dinge, wie
- Verhalten dieser Lösung bei Wifi? (bei mir: Ethernet)
- Verhalten bei zu langsamem Datenabruf?
- Performanceimpact? Oder sogar Verbesserung?
- Erhöhter Netzwerk-Traffic durch mehr Overhead? (Wobei das sogar immer besser wird, sofern die Intervalle länger konfiguriert sind)

Ist also aktuell eher eine kleine Spielerei, die für mich jetzt das beobachtete Problem gelöst hat.
Ob das aber Massentauglich ist? Das ist dann eine Frage des künftigen Designs.

mcm1957, 2024-04-30:
Ohne jetzt den Code amgesehen zu gaben

Intervall ist bei externer Kommunikation immer bedenklich da durch eine hängende Verbindung mehrfache Verbindungen aufgebaut werden könnten die ggF das gesammte System z.B wg Speicherüberlastung zum Absturz bringen könnten.

Ich würde hier zuerst mal checken ob die Kommunikation ein Timeout hat. Hängen über längere Zeit könnte ein Indiz sein dass der Adapter bei der Kommunikation keinnoder kein passendes Timeout setzt. Axios z.B. hat per Default kein Timeout.

mosandlt, 2024-04-30:
ich habe bei mir ein script erstellt welches alle 1 minute den adapter neustartet.. jetzt hab ichs geschaft das ich gar keine daten mehr bekomme. auch nicht über die app. habe den WR hart neugestartet.

NCIceWolf, 2024-04-30:
Soweit ich das verstanden habe ist es aktuell so:

Bei Adapter-Start wird eine (TCP-)Session aufgebaut und dann alle konfigurierten Elemente abgefragt.
Gleichzeitig wird ein Timeout gesetzt um nach Ablauf die Elemente wieder abzufragen.
Session bleibt immer bestehen.

Und ich denke da hat es sich dann auch aufgehangen wenn der WR mit anderem (z.B. der Portal-Kommunikation) beschäftigt ist oder die Session irgendwie hängen bleibt.

Ich habe das nun etwas abgewandelt:
Zu Beginn wird eine Session aufgebaut und alle Elemente abgefragt. Soweit gleich.
Parallel wird ein Intervall angelegt um den kompletten Prozess (rct.process) alle x Sekunden neu zu starten, inklusive des Aufbaus einer Session.

Vor Abfrage der Elemente setze ich dann einen Timeout um die Verbindung nach 2 Sekunden zu trennen.
Dadurch verhindere ich dass die Session oder Abfragen hängen bleiben. Die Abfragen brauchen laut meinen Logs in meinem Fall immer so um die 200ms, also genug Zeit alle Daten zu bekommen.

Ist vielleicht nicht die beste oder Endlösung, aber auf die Schnelle ist mir nichts besseres eingefallen um die Abfrage wieder stabil zu bekommen. Das Auto wollte dringend per Überschuss geladen werden.

mosandlt, 2024-04-30:
> Soweit ich das verstanden habe ist es aktuell so:
>
> Bei Adapter-Start wird eine (TCP-)Session aufgebaut und dann alle konfigurierten Elemente abgefragt. Gleichzeitig wird ein Timeout gesetzt um nach [...]

ich nehme jetzt mal deinen adapter her. Sieht aus meiner sicht recht gut aus.

OnkelFD, 2024-04-30:
Also ich deine Version seit gestern installiert und es läuft seit dem durch 😀

aruttkamp, 2024-04-30:
> Also ich deine Version seit gestern installiert und es läuft seit dem durch 😀
Von welchem Adapter ist denn jetzt hier die Rede ? So langsam blicke ich nicht mehr durch.

OnkelFD, 2024-04-30:
bei mir auch die von NCIceWolf

NCIceWolf, 2024-04-30:
> > Also ich deine Version seit gestern installiert und es läuft seit dem durch 😀
> > Von welchem Adapter ist denn jetzt hier die Rede ? So langsam blicke ich nicht mehr durch.

Meine Änderungen findest du alle in der rct.js, dort die rct.reconnect / rct.end und rct.process
Beschreibung dazu weiter oben. Ist wie gesagt nicht schön geschrieben, sollte nur auf die Schnelle einen Fix bringen.

aruttkamp, 2024-04-30:
@NCIceWolf - Alles klar. verstanden. Das schaue ich mir dann heute abend noch an bevor ich eine Version auf den Weg bringe.

aruttkamp, 2024-04-30:
Wenn Ihr einhellig der Meinung seit, dass diese Version erstmal tut was sie soll, dann würde ich die Änderungen von @NCIceWolf erstmal so übernehmen und in die Version einfliessen lassen. Performance Tests etc. würden jetzt nochmal ein paar Tage in Anspruch nehmen und darauf würde ich dann jetzt zu Gunsten einer fehlerbereinigten Version verzichten.

OnkelFD, 2024-04-30:
Ich würde es auch erstmal ein paar Tage so laufen lassen bevor es eingepflegt wird.

NCIceWolf, 2024-04-30:
> Ich würde es auch erstmal ein paar Tage so laufen lassen bevor es eingepflegt wird.

So lange haben alle anderen dann natürlich noch das Problem...

DK4NS, 2024-04-30:
Ein Hallo in die Runde. Wenn ich mich nicht ganz blöd angestellt habe, so konnte ich das Update von @NCIceWolf  einspielen. RCT-Portal ist seit einer Stunde wieder aktiv und bisher rennt hier alles wie erwartet. In Grafana sieht man schön die regelmäßigen Aktualisierungen. Die Objektfehler wegen den Datentypen sind auch weg. Kann jetzt mit der Wallboxsteuerung für Überschussladen weiter machen und bedanke mich bei allen. Melde mich nochmal für Statusupdate. Grüße

NCIceWolf, 2024-05-02:
Ich hab heute Morgen noch einmal einen Blick auf das Thema Performance geworfen.
Auch wenn ich schlecht mit anderen Usern vergleichen kann, kann ich ja wenigstens die Prozesse untereinander vergleichen.
Und da läuft der RCT Adapter aus meiner Sicht aktuell schön ressourcenschonend, sowohl was Speicher, als auch was CPU-Last angeht:

PID     VSZ   %CPU COMMAND
11347  239M   1.5 iobroker.js-con
12039  200M   1.3 io.javascript.0
11791  182M   1.2 io.admin.0
14565   84M   0.5 io.telegram.0
12953   84M   0.5 io.web.0
 5708    78M   0.5 io.shelly.0
20100   75M   0.5 io.influxdb.0
21951   57M   0.3 io.easee.0
24103   54M   0.3 io.vw-connect.0
12714   47M   0.3 io.melcloud.0
13678   45M   0.3 io.pvforecast.0
11148   _**44M   0.2 io.rct.0 <======**_
12388   40M   0.2 io.ping.0
28248   37M   0.2 io.iqontrol.0
12858   32M   0.2 io.discovery.0

JonasReuter, 2024-05-02:
Danke für eure Mühe. Ich habe auch den Umweg über einen Cronjob versucht, damit jedoch die Portal-Verbindung abgeschossen.

@aruttkamp Danke für den baldigen Release. Kann man sich an dem Aufwand (gemeint Kosten) für den Release beteiligen?

aruttkamp, 2024-05-02:
Die Änderungen von @NCIceWolf sind in den Stand 1.2.5 eingeflossen. 
Ich denke hier können wir bald schliessen

NCIceWolf, 2024-05-03:
Würde ich jetzt nicht unbedingt hier mit einfließen lassen, scheint ja ein anderes/neues Problem zu sein.
Hast du den Adapter mal auf "debug" oder "alles" gestellt und mal in den Log rein geschaut? Ohne diese Infos wird es schwierig...

Von den Farben her würde ich tippen dass der Adapter keine Verbindung zum WR aufbauen kann, hast du da mal die hinterlegte IP-Adresse geprüft (Stichwort DHCP)?

timoglaess, 2024-05-03:
Also an der IP liegt es nicht. Die Version 1.2.4 hat ja auch Verbindung bekommen. Auf die bin ich inzwischen wieder runtergegangen und hab einen cron gesetzt, damit er alle 2 Min neu startet. Die Version lief auf anhieb wieder mit den gleichen Einstellungen.

Auf die Logs hatte ich gar nicht geschaut (*schame on me*). Werd nochmal die 1.2.5 installieren und schauen was im Log ankommt.

timoglaess, 2024-05-03:
Selbes Bild. 1.2.5 läuft (bei mir) nicht:

```
host.iobroker 2024-05-03 13:48:30.636	error	instance system.adapter.rct.0 terminated with code 6 (UNCAUGHT_EXCEPTION)
rct.0 2024-05-03 13:48:30.017	error	Cannot read properties of undefined (reading 'description')
rct.0 2024-05-03 13:48:30.017	error	TypeError: Cannot read properties of undefined (reading 'description') at Rct.onReady (/opt/iobroker/node_modules/iobroker.rct/main.js:75:41)
rct.0 2024-05-03 13:48:30.016	error	unhandled promise rejection: Cannot read properties of undefined (reading 'description')
rct.0 2024-05-03 13:48:30.015	error	Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch().
```

aruttkamp, 2024-05-03:
@timoglaess Bitte poste deine Parameter die du abrufst.
Ich vermute das da einer bei ist der nicht vorgesehen ist ;-)

NCIceWolf, 2024-05-03:
Ooooooder, fehlt da nicht was in der Zeile 75 (Klammern und so?):

const rct_id = rct.cmd[e].id;
---> const name = rct.cmdReverse[rct_id].description || stateName; <---
const unit = (rct.cmdReverse[rct_id].unit || '').trim();
const type = (rct.cmdReverse[rct_id].ioBrokerType);

timoglaess, 2024-05-03:
Ich les ja nur die paar Parameter aus ;-) :

`battery.soh,grid_pll[0].f,battery.efficiency,energy.e_dc_day[0],energy.e_dc_day[1],energy.e_dc_total[0],energy.e_dc_total[1],energy.e_load_day,energy.e_load_month,energy.e_load_year,energy.e_load_total,energy.e_grid_load_total,energy.e_grid_load_day,energy.e_grid_load_month,energy.e_grid_load_year,energy.e_dc_month[0],energy.e_dc_month[1],energy.e_dc_year[0],energy.e_dc_year[1],battery.cycles,battery.bat_status,battery.status,battery.status2,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].enabled,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].enabled,dc_conv.dc_conv_struct[1].p_dc_lp,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],g_sync.p_ac_load[0],g_sync.p_ac_load[1],g_sync.p_ac_load[2],io_board.s0_external_power,power_mng.bat_next_calib_date,power_mng.is_grid,power_mng.is_heiphoss,power_mng.state,power_mng.use_grid_power_enable,power_mng.u_acc_mix_lp,prim_sm.state,energy.e_grid_feed_day,energy.e_grid_feed_month,energy.e_grid_feed_year,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,energy.e_ext_day,energy.e_ext_month,energy.e_ext_year,energy.e_ext_total`

aruttkamp, 2024-05-03:
@timoglaess - 
Sofortige Lösung : Entferne diesen Parameter
prim_sm.state

Korrektur erfolgt voraussichtlich mit 1.26

timoglaess, 2024-05-03:
OK, hab den "prim_sm.state" rausgenommen und die 1.2.5 funktioniert nun auch mit der eingestellten Refresh rate.

Hatte mich schon gefreut, dass der Parameter nun auch wieder funktioniert, aber das die Werte überhaupt aktualisiert werden ist definitiv wichtiger ;-) Ohne den RCT Adapter wäre meine komplette iobroker Installation überflüssig. Alles stützt sich auf die Werte aus dem Wechselrichter...

aruttkamp, 2024-05-03:
@timoglaess sorry mein Fehler. ich hatte mit dem Parameter getestet und leider eine Codezeile nicht wieder entfernt. 
Daher wurde der Parameter akzeptiert - es waren aber keine Werte definiert. 

Aber ... heute ist nicht alle Tage - Er kommt wieder - Keine Frage :-D

closed 2024-05-03 by aruttkamp