aruttkamp/ioBroker.rct #124: RCT Adapter auf 100% CPU auslastung

OnkelFD, 2024-05-17:
Hatte gestern Abend das Problem dass der RCR Adapter das System zu 100% ausgelastet hatte und auch den Ram vollgeschrieben hat.
Ich habe den Arbeitsspeicher daraufhin erweitert wodurch ich wenigstens den Adapter Stoppen konnte wodurch die Auslastung sich wieder Gefangen hat.
Laut Protokoll fand ein ständiger reconnect zum Wechselrichter statt.
Ein Neustart des kompletten Containers brachte letztendlich erst die richtige Abhilfe.
Will mir das aber noch einmal genauer anschauen.
Zu dem Zeitpunkt ist die v1.2.7 installiert gewesen.

NCIceWolf, 2024-05-17:
Da wäre es natürlich spannend das Log zu sehen.
Der Adapter verbindet sich zwar nach jeder Abfragen neu, aber da die alte Verbindung nach 2 Sekunden getrennt wird, sollte das kein Problem sein.
Außer die Verbindungen konnten bei dir nie getrennt werden...

Dazu kommt ja auch noch nächste Woche ein Update von RCT:

Sehr geehrte Kunden,

wir freuen uns, Ihnen mitteilen zu dürfen, dass mit dem Software Release am 20.05.2024 der RCT Power Doppelturm verfügbar wird. Damit ist der Power Storage DC 8.0 und der Power Storage DC 10.0 für die Verwendung von zwei parallelen Batterietürmen freigegeben.

Was ist im Release enthalten?

Mit dem Update der Power Storage Control Software auf X.3.5676 und Power Battery BMS Master Software auf 5607 werden folgende Verbesserungen eingeführt:

Einführung der Möglichkeit, zwei Batterietürme anzuschließen, um die Speicherkapazität zu verdoppeln
Behebung eines Fehlers, der das BMS-Update in seltenen Fällen blockiert hat
Verbesserungen in der Kommunikation/Verbindung mit der Power Battery
Verbesserte Erkennung von externen CAN-Kanalfehlern während des Systemstarts
Optimierung der SOC-Berechnung in Parallelmodus
Diverse Verbesserungen und Optimierungen der Batterieverwaltung und Kommunikationsprotokolle.
Anpassungen im RCT Power Portal und der RCT Power App

aruttkamp, 2024-05-17:
@OnkelFD - Bitte Logfiles zur Verfügung stellen.

OnkelFD, 2024-05-18:
Mist ich hatte das LOG auf Log-Stufe Silly stehen da hat er mir irgendwie überhaupt nichts aufgezeichnet. 😢
Aber angezeigt hatte er es. 🤷‍♂️ 
Habe es nun umgestellt, nun zeichnet er es auch wieder auf.
Wenn es noch einmal zu Problemen kommt melde ich mich mit einem Log zurück.

NCIceWolf, 2024-05-21:
Kurzes Update:

Mit der neuen Adapter-Version musste ich nach dem heutigen Update von Wechselrichter und BMS den Adapter nicht mehr neu starten, die Daten wurden wieder sauber abgeholt nachdem alle Komponenten wieder online waren.

Was mir noch eingefallen ist @OnkelFD:
Wenn du im Log ein Reconnect stehen hattest, dann war der Wechselrichter per IP für den Adapter nicht erreichbar. Dafür hatten wir ja extra eine Prüfung eingebaut. Das sollte aber den Speicherverbrauch auch nicht explodieren lassen...

OnkelFD, 2024-05-22:
RCT v1.2.9

- Also ich habe gerade eben den Test gemacht und das Ethernet Kabel vom Wechselrichter getrennt eine weile gewartet.
Folgen die CPU (ca. 5%) Und RAM Auslastung sind moderat geblieben.

- Ethernet Kabel wider angeschlossen leider kein erneuter Aufbau der Verbindung aber die CPU Auslastung ist deutlich angestiegen (40-50%).

- Erst durch das stoppen und starten des Adapters erfolgte einen Verbindungsaufbau und die CPU Auslastung bewegt sich wieder unter 1%.

Die Auslastungen Beziehen sich auf den Adapter und nicht auf das ganze System. Weiß jetzt halt nicht was Passiert wenn die Verbindung öfters abricht und wieder verbunden wird was dann Passiert.

Könnte man wenn die Verbindung abbricht den Intervall der Meldungen Reduzieren bzw. ist jede Meldung ein Verbindungsversuch? 
Ein Verlust der Verbindung schreibt den Log ohne ende voll.
Eine Meldung je Intervall oder so würde ja vollkommen ausreichen.

rct.0 | 2024-05-22 10:04:09.915 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.911 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.909 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.909 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.906 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.906 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.905 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.905 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.905 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.905 | error | RCT: connection error, please check ip address and network!
rct.0 | 2024-05-22 10:04:09.904 | error | RCT: connection error, please check ip address and network!

NCIceWolf, 2024-05-22:
Habe das bei mir mal nachgebaut und einfach den LAN-Port am Switch de- und wieder aktiviert. Sollte also deinem Szenario sehr ähnlich kommen. Hier gab es wohl ein Problem bei den Intervallen/Timeouts.

Hatte jetzt bei mir folgende Meldungen:
1 mal: Verbindungsfehler, vorhergehender Stream nicht korrekt geschlossen
Danach dann 1 mal pro Minute: Verbindungsfehler, IP Adresse und Netzwerk prüfen!

Damit sollten sich die Meldungen bei dir drastisch reduzieren. Wobei das bei dir ja im Millisekunden-Bereich kam....

Bin gespannt ob es das erledigt, aber ich lasse das lieber heute noch den ganzen Tag laufen und beobachte das.
Und @aruttkamp sollte sich das auch in Ruhe anschauen.

Dann darfst du gerne Beta-Tester sein 🍻

OnkelFD, 2024-05-22:
Hebe es mal getestet mit der Version von dir und nun sieht es wesentlich entspannter aus im log 🤗.

closed 2024-05-22 by aruttkamp