aruttkamp/ioBroker.rct #126: Werte werden nicht aktualisiert g_sync.p_ac_sum_lp g_sync.u_l_rms[2] WR v2.3.5676

OnkelFD, 2024-05-24:
Seit dem Update des Wechselrichters auf v2.3.5676 werden die Werte:

g_sync.p_ac_sum_lp -> nur aktualisiert wenn RCT Power APP aktiv ist und man auf der gesamt Übersicht ist
g_sync.u_l_rms[2] -> nur aktualisiert wenn RCT Power APP aktiv ist und man auf der Netz Übersicht ist

RCT v1.2.9 habe zum Versuch auch mal die v1.2.7 installiert wo es aber auch nicht funktioniert.

NCIceWolf, 2024-05-25:
Kann ich nur halb bestätigen.
Es kommen zwar Werte, aber oftmals nur alle 5 Minuten:

[image]

Da viele/alle anderen Werte regelmäßig kommen, liegt es nicht am Abruf der Daten selbst. Muss also an diesen beiden Werten liegen.
Scheinbar hat RCT da wirklich was "optimiert"...

aruttkamp, 2024-06-11:
Moin @OnkelFD , @NCIceWolf 

gibts hier noch was neues oder können wir den Issue schliessen ?

OnkelFD, 2024-06-11:
Also bei mir hat sich nichts geändert, ich nehme aber an dass dies durch das Update zu den Verzögerungen kommt und nicht direkt was mit dem Adapter zu tun hat. 😢

NCIceWolf, 2024-06-13:
Sehe ich auch so.
Trat direkt nach dem letzten Update vom WR so auf, also wohl Teil der "Optimierung" von RCT.

Da kann man aus meiner Sicht nichts am Adapter machen, die Daten die vom WR kommen verarbeitet er auch.

Bliebe hier also nur ein Ticket bei RCT zu erstellen wenn dieser Wert öfter gebraucht wird.

closed 2024-06-13 by NCIceWolf