aruttkamp/ioBroker.rct #136: RCT: connect error, please check IP adress and network

Corneys, 2024-06-21:
Hallo,

in der jüngeren Vergangenheit kam es bei mir immer mal wieder vor das, vor allem Nachts/Morgens mein Iobroker abgestützt ist.
Leider bin ich eine Zeit lang nicht dazu gekommen nach der Ursache zu suchen, weswegen ich auch nicht genau festlegen kann, seit wann folgendes Verhalten tatsächlich auftritt.

Kurz zum Hintergrund meiner Anlage:
Ich habe zwei Wechselrichter inklusive PowerSwitch und Speicher, der Speicher hängt an dem Master-Wechselrichter
Der Slave schaltet sich Abends ab, weswegen ich dann das Skript von [https://github.com/aruttkamp/ioBroker.rct/discussions/27](https://github.com/aruttkamp/ioBroker.rct/issues/url) nutze um den Adapter des Slaves auszuschalten.

Sobald kein Strom mehr vom Dach kommt dauert es nicht lange, dass mir die Fehlermeldung "RCT: connect error, please check IP adress and network" im Log ausgegeben wird. 
Das Verhalten trifft bei mir auch auf die Instant RCT.1 für den Slave zu, dieser wird aber durch das automatische Skript irgendwann abgeschaltet.
Der RCT.0 Adapter feuert die Meldung aber die ganze Nacht lang durch und das mit 1-3 Meldungen pro tausendstel Sekunde. Was dann ungefähr 7,2 Millionen Logeinträge pro Stunde, und Loggrößen von ca. 3GB verursacht.

Der Adapter Arbeitet, solange mein System von dem Meldungen nicht überfordert ist, auch Nachts tapfer weiter.
Der Datenpunkt rct.0.info.connection bleibt die ganze Nacht lang auf true und auch die Daten wie der Speicherstand werden korrekt ausgegeben.

Versionen habe ich schon Kontrolliert und geupdatet, sowie die Wechselrichter neu ins Netzwerk eingebunden, jedoch hat das leider keine Veränderung gebracht.

Versionsstände IoBroker
RCT Adapter: 1.2.9
Node.js: v20.14.0
NPM: 10.7.0
Admin: 6.13.16

Versionsstände Wechselrichter
RCT Control Version: 2.3.5676
BMS Version: 5607
(Wechselrichter sind per Lan-Kabel ans Netzwerk angeschlossen)

Ich habe ehrlichgesagt, keine Ahnung woran das liegen kann oder was ich noch zur Fehlerbehebung tun könnte, weswegen ich auf Hilfe hoffe :) 

Vielen Dank 
Corneys

[image]

NCIceWolf, 2024-06-23:
Hm, das macht bisher aus sehr vielen Gründen nicht so Recht Sinn...

Eigentlich sollte das bei Verlust der Netzwerkverbindung (habe einfach das Kabel abgezogen) ungefähr so aussehen:

2024-06-23 13:39:20.043 - error: rct.0 (1450) RCT: connection error! Previous stream was not closed correctly!
2024-06-23 13:39:35.886 - error: rct.0 (1450) RCT: connection error, please check ip address and network!
2024-06-23 13:39:42.027 - error: rct.0 (1450) RCT: connection error, please check ip address and network!
2024-06-23 13:40:45.132 - error: rct.0 (1450) RCT: connection error, please check ip address and network!
2024-06-23 13:41:48.236 - error: rct.0 (1450) RCT: connection error, please check ip address and network!
2024-06-23 13:42:48.238 - info: rct.0 (1450) RCT: connected to server at 192.168.178.70

Was ich jedoch bei dir nicht verstehe:
Die Meldung kommt von dem WR der die Batterie dran hat? Der sollte dann doch weiterhin erreichbar sein?
Ist der denn nachts noch anpingbar?

Und dazu: die Meldungsflut sollte so eigentlich nicht vorkommen, ein reconnect wird bei Abbruch der Verbindung erst nach 1 Minute wieder versucht.

Hast du die Version von @aruttkamp im Einsatz oder meinen Fork? Bin mir nicht sicher ob meine letzten Änderungen schon in der 1.2.9 drin sind...

Corneys, 2024-06-24:
Guten Morgen,

vielen Dank für deine Unterstützung! Gestern Abend habe ich das ganze noch einmal beobachtet.
Genau, die Meldung kommt vom WR an dem die Batterie hängt, der Master WR also.
Ich habe keinen Fork im Einsatz, sondern die "normale" 1.2.9

Gestern ist mir dabei aber aufgefallen, dass während der Meldungsflut der WR in der Fritzbox als verbundenes Gerät erkannt wird und auch ein anpingen per Eingabeaufforderung, wird erfolgreich beantwortet.
Der RCT-Adapter meldet währenddessen auch ganz normale und valide Daten des Speicherstandes, sodass der der Adapter (bis auf die vielen Meldungen) eigentlich ganz normal funktioniert.

Aber gestern ist mir dabei, mehr durch Zufall, aufgefallen, dass während der Meldungsflut die RCT-App quasi tot ist, weil sie den Master-WR als offline ansieht.
Sobald ich den RCT-Adapter Deaktivieren, wird der WR wieder als online erkannt und Daten an die App gesendet.

[image]
[image]

NCIceWolf, 2024-06-24:
Ok, das erklärt es ein Stück weit.
In der 1.2.9 sind meine letzten Änderungen noch nicht drin und daher werden bei dir wohl immer wieder neue Verbindungen aufgebaut. Auch wenn mich das im Millisekunden-Bereich schon sehr wunder. Ich hatte das erst kurz nach dem Release der 1.2.9 etwas "resilienter" gestalten können. Da werden Verbindungen immer erst getrennt bevor neue aufgebaut werden.

Da du mit dem Konstrukt scheinbar einer der wenigen bist die dieses spezifische Problem haben, wäre es spannend ob die Verbesserungen bei dir zum Erfolg führen (dann bräuchtest du theoretisch auch kein Skript mehr, hättest nur für den Slave noch jede Minute eine Fehlermeldung). Hier könnte man aber sicher in Zukunft noch an den Timeouts feilen.

Du kannst also entweder auf ein neues Release warten, oder die Github Version von @aruttkamp oder mein fork probieren 😃

Corneys, 2024-06-24:
Vielen Dank für die schnelle Rückmeldung. 
Ja, der Aufbau den wir haben ist etwas umfangreicher und kommt deswegen nicht ganz so oft vor. 

Da ich jetzt die Adapter zur Sicherheit, bis auf weiteres, bei Dunkelheit automatisch deaktiviere, werde ich mal die neuen Versionen testen.
Die RCT-Adapter sind ja alle nur "lesend", oder? Sodass der Betrieb keinen Einfluss auf die Wechselrichter hat, oder sie beschädigen könnte? Sorry bin da immer etwas paranoid/vorsichtig.

Zum Testen werde ich vermutlich dann auch erst wieder am Wochenende kommen, da zur Zeit das testen doch erst recht spät möglich ist :)

NCIceWolf, 2024-06-24:
Stimmt, aktuell gibt es ja zum Glück sehr lange und seit heute auch wieder mehr Strom :)

Aktuell liest der Adapter nur Werte vom WR aus.
Es gibt aber die Idee/Bestrebung zumindest die SOC-Ziel Werte durch den Adapter ändern zu können (#20 ), die Umsetzung wird aber sehr vorsichtig erfolgen und sicher auch noch etwas Zeit brauchen. Wären dann aber auch nur Werte die man jetzt schon über die RCT-App ändern kann.

Aktuell also kein Grund zur Sorge, kaputt machen kann der Adapter nix!

Corneys, 2024-06-30:
So, ich habe mich gestern Abend dem Adaptern angenommen und und habe beobachtet.
Dazu habe ich deinen Fork mit der Version 1.2.11 installiert. 
Nach dem Installieren des Forks, lief der Adapter Tagsüber sauber ohne Fehlermeldungen durch.
Sobald es aber Richtung Dämmerung ging, habe ich folgende Meldung bekommen

2024-06-29 21:49:46.187  - error: rct.0 (935118) RCT: connection error! Previous stream was not closed correctly!

Nach der Meldung blieb der Adapter quasi stehen und lieferte keine neuen Daten mehr.
Ich habe um 23:10 Uhr den Adapter noch einmal neu gestartet, woraufhin der Adapter wieder für 13 Minuten sauber lief. Daraufhin folge wieder die Meldung und der Adapter blieb stehen.
Ansonsten gab es in Logs nichts weiter zu der Instanz, also keine Flut von Fehlermeldungen oder ähnlichem.

NCIceWolf, 2024-06-30:
Dass der Wechselrichter nach Sonnenuntergang nicht mehr erreichbar ist, macht ja auch Sinn.
Zumindest der zweite. Und dass da dann keine Daten mehr kommen ist ja auch klar. Und das jetzt keine Meldungsflut mehr kommt ist auch gut, damit haben die Änderungen ja dieses Problem gelöst.

Im normalen Tagesverlauf bekommst du von beiden Instanzen korrekte Daten? Und sind diese zeitversetzt oder kommen die gleichzeitig?
Eigentlich sollte die Instanz mit dem Haupt-WR weiterlaufen und die mit dem zweiten WR keine mehr liefern.
Zumindest sofern es funktioniert dass zwei Instanzen Datenverbindungen zu zwei WRs aufbauen. Das kann ich natürlich nicht testen und müsste man anhand der Logs auslesen.

Corneys, 2024-06-30:
Hi, das habe ich nicht ganz klar geschrieben. 
Gestern Abend hatte ich bisher noch mein Skript laufen, was die Instanz des Slave-WRs deaktiviert, sobald dieser offline geht.
Deswegen lief auch nur die Instanz rct.0 mit dem Master-WR

Genau Tagsüber passen die Daten zu dem was mir die RCT-App auch sagt und die Daten aus den Instanzen rct.0 (Master) und rct.1 (Slave) kommen dabei zeitversetzt.

Nachdem die Meldung "connection error! Previous stream was not closed correctly!" ausgeworfen wurde und der Master keine Daten lieferte blieb die Instanz trotzdem auf grün.

NCIceWolf, 2024-06-30:
Also die Instanz mit dem WR-Master bringt bei Sonnenuntergang den Fehler?
Wenn der WR erreichbar bleibt, sollte das eigentlich nicht passieren, der Fehler kommt nur wenn die Kommunikation mit dem WR abbricht und/oder nicht mehr neu aufgebaut werden kann.

Bitte teste heute Abend doch mal folgendes:
Instanz rct1 ausschalten und nur die rct0 laufen lassen.
Sollte dann noch einmal die Fehlermeldung kommen, dann bitte das komplette Log (auf debug) herunterladen und hier rein posten.

aruttkamp, 2024-07-01:
@NCIceWolf - Vielleicht spicken wir den Verbindungsprozess mal mit zusätzlichen InfoLogausgaben !?!
Oder führen einen neuen Parameter ein "Debug Modus" - der dann eben diese zusätzlichen Logs schreibt. 

@peperoni60 - Du hast doch auch 2 Wechselrichter - wie ist das bei Dir ?

peperoni60, 2024-07-01:
Ich hatte solche Probleme bisher nicht. Ich habe zwei WR mit je einem Batteriestack. Meine Instanzen liefen alle sauber. Mittlerweile bin ich aber auf den Homeassistant umgestiegen, weil der m.E. benutzerfreundlicher und funktionsreicher ist.

aruttkamp, 2024-07-01:
@peperoni60 - Danke für Deine Schnelle Reaktion. ok. du hast an beiden WR nen Batterieturm. Das macht natürlich nen großen Unterschied. 
So so .. abgewandert zu Homeassist ;-) Dann streiche ich das mit der "android_description" ... :-P

Corneys, 2024-07-01:
Hi, leider komme ich erst jetzt dazu meine Erkenntnisse von letzter Nacht zu teilen.
Gestern Abend habe ich meinen Test-Raspi, mit mit dem IoBroker komplett neu aufgesetzt und eventuelle selbstgemachte Probleme auszuschließen. Auf dem System habe ich nur den Fork 1.2.11 und den History-Adapter laufen um zu sehen wann keine validen Daten mehr ankommen. (Deswegen ist der Log auch recht überschaubar)
Die RCT-Instanzen auf meinem "Produktiv"-System, sowie die Instanz rct.1 für den Slave-WR im Testsystem habe ich gegen 18 Uhr deaktiviert.
Es lief an RCT-Instanzen also nur die Instanz des Master-WRs auf dem Testsystem.
Leider war es wieder so, dass als es dunkel wurde keine Daten mehr geliefert wurden und auch obwohl die Instanz auf dem Log-Level "debug" stand nichts an Fehlermeldungen mitgekommen ist. 
[Log RCT IoBroker.txt](https://github.com/user-attachments/files/16056312/Log.RCT.IoBroker.txt)

aruttkamp, 2024-07-01:
Reden wir hier über immer 22:00 wenn es auftritt ? @Corneys

Corneys, 2024-07-01:
Kann ich mir heute noch einmal anschauen. 
Aber letztes Mal war es so, dass ich die Instanz neu gestartet hatte, wobei sie dann für 12 Minuten wieder lief. Danach gab es wieder den error.

aruttkamp, 2024-07-01:
Was mich stört ist das es nur bei dir auftritt. jetzt ist die Lösung mit 2WR natürlich nicht sooo häufig. 
kann es auch ein Ausfall deiner Infrastruktur sein ? Hast mal auf deinem WLAN Router nach evtl. unterbrechungen geguckt ? 
Oder bei Kabelanbindung .. startet evtl. der Switch neu oder resettet den Port ? 
Ja, alles sehr weit hergeholt - gebe ich zu. aber irgendwas muss es doch sein.

Corneys, 2024-07-01:
Also die RCT-App funktioniert weiterhin ohne Probleme. Würde es am Netzwerk liegen, dürfte da ja auch nichts zu sehen sein.
Außer es ist nur ein kurzer Abbruch, aber sollte der Adapter dann nicht wieder die Verbindung aufnehmen?

NCIceWolf, 2024-07-02:
Ja, da scheint irgendwo das Problem zu liegen.
Im Log ist ja ersichtlich dass die Datenverbindung nicht sauber abgebaut werden konnte.
Also irgendetwas passiert da auf Verbindungsebene, die Frage ist nur was.
Und warum dann der nächste Aufbau nicht mehr funktioniert und scheinbar auch das Intervall nicht weiterläuft...
A propos: Welches Intervall hast du denn eingestellt?

Der Ablauf ist ja inzwischen so:

Intervall zum Abruf der Daten (Standard: 10s)
|-> Abruf der Daten mit Neuverbindung zum WR
||-> Abbau der Verbindung nach 2 Sekunden
|
|
|-> Abruf der Daten mit Neuverbindung zum WR -> Fehlermeldung im Log: vorherige Datenverbindung nicht abgebaut!
-----> XXXXX -> Hier passiert nichts mehr!  <-----
||-> Abbau der Verbindung nach 2 Sekunden
|
|
|-> Abruf der Daten mit Neuverbindung zum WR
||-> Abbau der Verbindung nach 2 Sekunden

NCIceWolf, 2024-07-02:
So, habe in meinen Fork nun ein paar mehr Log-Einträge eingebaut.
@Corneys : Bitte einmal die jetzt aktuelle Version meines Forks installieren und bei Gelegenheit noch einmal testen!

Corneys, 2024-07-03:
Hi also ich habe deinen Fork gestern Abend noch installiert und heute Nacht wurden durchgängig Werte vom Speicher geliefert.
Es gab immer mal wieder die Meldung "connection error! Previous interval connection not successful!", die Instanz blieb aber weiterhin auf grün.

Den Intervall habe ich nicht verändert und stand auf 10

[Log_IoBroker_03.07.2024.txt](https://github.com/user-attachments/files/16078930/Log_IoBroker_03.07.2024.txt)

NCIceWolf, 2024-07-03:
Ok, nun werden wir etwas schlauer!!

Hier scheint tatsächlich Nachts immer mal wieder die Verbindung zwischen Adapter und WR abzubrechen.
Zu sehen ist das daran dass die Verbindung noch aufgebaut wird, dann aber in jedem einzelnen Fall nicht mehr abgebaut werden (können).
Im Zeitraum zwischen 17:35Uhr (Update) und 0:00Uhr ist das bei dir 9mal passiert, in meinem Log hab ich das gar nicht gesehen. Zuerst kam das um 21:45Uhr.

Wenigstens wird der Abbruch jetzt korrekt verarbeitet und nach 1 Minute eine neue Verbindung aufgebaut die dann jeweils auch funktioniert (die ist hauptsächlich wegen dem 30Sekunden Timeout jedes Verbindungsversuchs so gewählt.
Die Instanz bleibt hier noch auf grün, erst wenn diese Neuverbindung nach 1 Minute scheitert geht sie auf gelb.
Ab dann wird eine Neuverbindung nur noch alle 2 Minuten versucht.

Hier das erste Mal als Referenz:

2024-07-02 21:45:00.879 - info: rct.0 (32260) RCT: terminating interval connection to server at 192.168.178.100
2024-07-02 21:45:00.879 - info: rct.0 (32260) RCT: interval connection to server at 192.168.178.100 closed
2024-07-02 21:45:08.879 - info: rct.0 (32260) RCT: interval connection to server at 192.168.178.100 successfully established
--------> Fehler
2024-07-02 21:45:18.880 - error: rct.0 (32260) RCT: connection error! Previous interval connection not successful!
2024-07-02 21:45:18.882 - info: rct.0 (32260) RCT: interval connection to server at 192.168.178.100 closed
2024-07-02 21:46:18.886 - info: rct.0 (32260) RCT: Initial connection successful to server at 192.168.178.100!
2024-07-02 21:46:18.888 - info: rct.0 (32260) RCT: interval connection to server at 192.168.178.100 successfully established
2024-07-02 21:46:20.891 - info: rct.0 (32260) RCT: terminating interval connection to server at 192.168.178.100
2024-07-02 21:46:20.892 - info: rct.0 (32260) RCT: interval connection to server at 192.168.178.100 closed

Corneys, 2024-07-03:
Also die Forks habe ich auf einem eigenen neu aufgesetzten Pi laufen auf dem sonst nichts läuft, weswegen ich die Maschine und den IoBroker fast ausschließen würde, da der selbe Fehler ja auch auf meinem Echtsystem auftritt.
In meinem Netzwerk hängt schon eine Menge, aber nichts (was mir bewusst wäre) was erst bei Sonnenuntergang losrennt und hier Probleme verursachen könnte.

Ich habe hier noch eine ungenutzte FritzBox rumfliegen, die ich am Wochenende mal anschließen kann und nichts außer den Pi und die Wechselrichter dran anschließen kann. Eine Andere Möglichkeit und Probleme auszuschließen sehe ich sonst nicht, außer ihr habt noch Ideen für mich.

NCIceWolf, 2024-07-04:
Interessant wäre nun noch der Log von gestern um zu sehen bis wann die Verbindungsprobleme gingen und ab wann sie wieder angefangen haben.

Laut Log sieht es bisher so aus als würde die Verbindung noch sauber aufgebaut werden können, dann jedoch nicht mehr abgebaut werden. Habe da mal noch einen kleinen Log-Eintrag hinzugefügt der vielleicht zeigen kann ob der Abbau gar nicht erst gestartet wird, oder eben nicht erfolgreich ist (der Adapter schickt hier ein FIN-Paket an den WR).

Sonst bliebe aus meiner Sicht fast nur noch ein Netzwerkmitschnitt (z.B. Port-Mirroring am Pi und/oder WR) um zu sehen wo das hängen bleibt.
Da es ja irgendwie mit dem Sonnenstand zu tun hat, kann ich mir auch interne Timeouts im WR vorstellen, wenn dieser seinen Slave nicht mehr erreichen kann. Sollte sich dann aber auch irgendwie in der RCT-App bemerkbar machen, vielleicht auch nur durch kurze Verzögerungen.

Sonst bleibt nur die Frage ob man hier weiter Aufwand betreiben will. Wenn man 10 Ausfälle pro Stunde annimmt, würden dir in der Nacht (aktuell kürzer, im Winter länger) pro Stunde 10 Minuten an Daten fehlen.
Besser als bisher ist es ja schon, aber da wären halt immer wieder Lücken drin...

Corneys, 2024-07-05:
Gestern Abend um 19:22 habe ich den aktuellsten Fork geladen. Hier der Log von gestern.
[Log_IoBroker_04.07.2024.txt](https://github.com/user-attachments/files/16112664/Log_IoBroker_04.07.2024.txt)
 
Eine positive Sache hatte das ganze bisher trotzdem schon für mich. 
Durch diese ganze Thematik habe ich mich natürlich etwas intensiver mit den Wechselrichtern und auch der RCT-App beschäftigt.
Dabei ist mir aufgefallen, dass in den Einstellungen des Slave-WR die gesamte Watt Peak Leistung in den Einstellungen angegeben war, aber auf dem Master nur genau die Hälfte der Peak-Leistung. 
Durch unsere Ost-West Ausrichtung war mir auch bewusst, dass nie Zeitgleich die gesamte Peak-Leistung vom Dach kommen wird. (Das aber bei Sonnenschein immer nur maximal die Hälfte des Peaks vom Dach kam hätte mich eigentlich schon früher stutzig machen sollen.)
Deswegen habe ich heute Morgen beim RCT-Kundenservice angerufen, der sich auf mein System schalten konnte und die Einstellungen innerhalb ein paar Minuten korrigiert hat. Zack kam direkt 2/3 der Peak Leistung vom Dach.

Ich habe auch ganz Leise die Hoffnung, dass dadurch vielleicht auch das Problem der Verbindungsabbrüche gelöst ist, weil die Einstellungen der beiden Systeme nicht konsistent waren. Das werde ich euch aber erst Morgen berichten können.
Drückt mir die Daumen

NCIceWolf, 2024-07-05:
Das klingt doch nach einem echten Gewinn 😁 
Dann drücken wir die Daumen und sind gespannt auf den Log von heute!

Corneys, 2024-07-06:
Einen schönen Guten Morgen zusammen,

ich weiß gar nicht, was ich sagen soll, aber heute Nacht hatte ich nicht einen error in meinem Log.
Es scheint tatsächlich so, als ob das korrigieren der Einstellungen vom RTC-Kundenservice hier die Lösung war.
Bitte fragt mich aber nicht, wieso eine falsch eingetragene Anlagenpeakleistung in den Geräteeinstellungen nachts zu Verbindungsabbrüchen führt.

Ich werde heute die Instanzen auf meinem Echtsystem wieder starten und heute Abend dort beobachten.

NCIceWolf, 2024-07-06:
Na dann war ich ja mit den "internen Timeouts" im WR ziemlich nah dran!!

Haben wir ja immerhin viel gelernt und die Verbindung sollte nun auch für nicht 100% stabile Verbindungen wie WiFi und Co. besser funktionieren.

Hatten also wenigstens alle was von 😊

aruttkamp, 2024-07-06:
Perfekt gelaufen würde ich sagen ! 
Adapter verbessert - Eigene Hardware besser kennengelernt - was will man mehr !

aruttkamp, 2024-07-06:
Mache dann hier zu.

closed 2024-07-06 by aruttkamp