aruttkamp/ioBroker.rct #148: regelmäßige Verbindungsfehler

plm2310, 2024-07-09:
Seit den Versionsupdate auf 1.2.8 ist mein iobroker log für den rct-Adapter voll mit Fehlermeldung wegen Verbindungsabbrüchen in unregelmäßigen Abständen. Im Release 1.2.8 und 1.2.9 sind es nach einiger Zeit Verbindungsabbrüche im Sekundentakt, die nach einige Stunden, den gesamten ioBroker überlasten. Mit Release 1.2.12 bauen sich die Verbindungsabbrüche nicht mehr auf, treten aber immer noch regelmäßig auf.
[image]

Das Refresh Intervall ist 10 Sekunden und zwischen den Verbindungsabbrüche werden auch Werte gelesen:
[image]

Versionsstände IoBroker
RCT Adapter: 1.2.12
Node.js: v18.20.3
Admin: 6.17.14

Versionsstände Wechselrichter
RCT Control Version: 2.3.5674
BMS Version: 5599
(Wechselrichter ist per Lan-Kabel ans Netzwerk angeschlossen)

NCIceWolf, 2024-07-09:
Hallo!

Ja, in den vorherigen Versionen war das einfach nicht sichtbar, d.h. Verbindungsabbrüche waren zwar da, man konnte das aber nur an den fehlenden Werten erkennen.
Du kannst gerne mal meinen Fork bei dir installieren. Hier ist ein zusätzliches Debugging aktiv, welches etwas besser erkennen lässt was da abläuft.

Aus dem bisherigen Log lässt sich nur schließen dass die Datenverbindung nicht sauber abgebaut werden kann, die Neuverbindung dann allerdings wieder funktioniert.
Da wir gerade erst ein Thema mit Slave-WRs hatten: Wie ist denn dein Setup? Und was hängt zwischen ioBroker und WR?
Zusatzfrage: Wenn das regelmäßig auftritt, kann da mal ein Mirror-Port für ein Netzwerk-Tracing eingerichtet werden?

plm2310, 2024-07-09:
Hallo,
den Fork habe ich gerade installiert und den log-level auf debug gestellt. 
Die Verbindungen sehen eigentlich alle 10 Sekunden, ganz ok aus. sporadisch gibt es dann aber Verbindungsfehler. Komischerweise kann die Verbindung Millisekunden später erfolgreich aufgebaut werden.
[image]
[image]
Mein Setup ist eigentlich recht simpel. Ich habe nur einen WR, der per Lan über einen passiven Switch verbunden ist. 
Mit Mirror-Port und Netzwerktracing kenne ich mich nicht aus. Hast du evtl. einen Link zu dem Thema, dann kann ich versuchen das einzurichten.

NCIceWolf, 2024-07-09:
War nur eine Bonus-Frage.
Wenn das Equipment/Wissen dafür nicht rein zufällig da ist, scheint mir der Aufwand dafür etwas hoch.

Ich habe noch einmal zwei/drei Zeilen Code mehr bei mir eingefügt, probier noch einmal die aktuellste Version aus meinem Fork, vielleicht sehen wir noch etwas besser wo er hängen bleibt.
Für mich sieht es nach wie vor so aus als würde jeweils ein Connect nicht funktionieren.
Alternativ kannst du mal den Intervall auf 30 Sekunden oder höher stellen, der Verbindungs-Timeout kommt nach ca. 25 Sekunden.

plm2310, 2024-07-11:
deinen Fork habe ich installiert. Hier ein paar Beispiele mit Verbindungsabbrüchen:
[image]
[image]

NCIceWolf, 2024-07-11:
Hm, mit Screenshots ist das immer so eine Sache, da wäre eine Log-Datei besser.
Ich gehe mal davon aus dass die IP-Adresse nicht öffentlich erreichbar ist? Dann müsste die auch nicht unkenntlich gemacht werden.

Also ich sehe hier genau ein Verbindungsproblem (11.07. 01:57:30 Uhr), sonst keines.
Und hier kann wie vermutet keine TCP-Verbindung zum WR hergestellt werden.
Das merkt der Adapter dann beim nächsten Intervall (10 Sekunden später) und verzögert die nächste Verbindung auf eine Minute. Diese läuft dann wieder sauber ab.

Mit den vorliegenden Daten kann ich hier also nicht mehr tun.
Vielleicht lässt sich am Logfile eines kompletten Tages noch etwas erkennen, aber momentan scheint es als würde die TCP-Verbindung in ein Timeout laufen, was irgendwo zwischen "Adapter -> Host -> Netzwerk (Kabel, Switch) -> WR" liegen wird.

NCIceWolf, 2024-07-16:
Im Sinne der besseren Analysefähigkeiten sollte mind. die 1.2.12 oder besser noch die kommende Version verwendet werden.
Zum einen stimmt jetzt der Ablauf der Log-Meldungen und diese werden korrekt klassifiziert.

@aruttkamp: Wie das @plm2310 hier richtig dargestellt hat wurden in der 1.2.9 noch viel zu viele Logs generiert und auch die Verbindungen teilweise mehrfach neu erstellt was am Ende tatsächlich zu einer Überlastung des ioBrokers führen konnte.
Auch wenn das nicht die Masse betreffen sollte, sollte nach entsprechender Laufzeit auch die stable-Version mit den letzten Verbesserungen versehen werden.

@plm2310: Hattest du noch einmal die Gelegenheit das weiter zu prüfen und ein Logfile zu ziehen?

closed 2024-07-17 by aruttkamp