aruttkamp/ioBroker.rct #208: 'Stromlose' Inverter Warnung abstellen?

Madaraxa, 2025-02-15:
Hi.
Wenn es dunkel wird geht mein slave Inverter offline (wird Stromlos, da nichts auf der DC Seite ankommt, was normal ist laut RCT).
Der Master Inverter mit Batteriespeicher bleibt online.
Das ganze führt dazu, dass ich jede 'Nacht' folgende Meldung alle 5-10 Minuten kommt
`rct.0
2025-02-15 18:43:18.594	warn	RCT: Connection error! Previous interval connection not successfully completed!`

Kann man das irgendwie ändern?
Wahrscheinlich nicht, aber ich wollte gefragt haben :)

aruttkamp, 2025-02-15:
Du hast 2 Instanzen laufen ? rct.0 für den Slave und rct.1 für den Master ?

Madaraxa, 2025-02-16:
Hi. Ich habe nur eine Instanz (rct.0). Vom slave brauche ich keine Daten außer dem AC Wert. Dieser steht unter io_board - > s0_external_power.
Da ich die o.g. Warnings erst bekomme seit ich nen slave habe und mein Master auch online bleibt (power switch und Batterie) gehe ich davon aus, dass die Warnungen durch den slave kommen?
Den Adapter komplett beenden möchte ich ungerne, da ich in den Sommer Monaten durch die Batterie durch die Nacht komme.

aruttkamp, 2025-02-16:
Wenn du keine Instanz für den Slave hast kennt RCT den ja gar nicht. Ich wüsste nicht wie er dann darauf kommen sollte. 
Es wird nur mit dem Master kommuniziert. Liefert der Master denn Daten an den Adapter ? Also Nachts, wenn die Fehlermeldung auftritt

NCIceWolf, 2025-02-16:
Kann mir da auch vorstellen dass der Master keine Daten vom Slave bekommt und deshalb zu spät oder gar nix an den Adapter versendet (internes timeout?).

Stell das Log mal auf debug um und wirf hier mal einen Tag rein.
Vielleicht kann man da etwas an den Timeouts (2 Sekunden) drehen, wird aber schwierig da das ja nur du nachstellen kannst.

Und wenn du ohnehin irgendwann zwei Instanzen verwenden willst...

Madaraxa, 2025-02-16:
So. Musste warten bis keine Leistung mehr vom Dach kommt.
Log ist mit eingestelltem debug:

`
rct.0
2025-02-16 17:59:35.795	info	RCT: Initial connection successful to inverter at 192.168.9.108!

rct.0
2025-02-16 17:58:35.790	warn	RCT: Connection error! Previous interval connection not successfully completed!

rct.0
2025-02-16 16:48:45.417	info	RCT: Initial connection successful to inverter at 192.168.9.108!

rct.0
2025-02-16 16:48:45.253	info	config rct_ip: 192.168.9.108

rct.0
2025-02-16 16:48:45.212	info	starting. Version 1.2.17 in /opt/iobroker/node_modules/iobroker.rct, node: v20.18.3, js-controller: 7.0.6
`
Also nichts neues. Die Batterie liefert fröhlich den aktuellen ladestand und auch der Master liefert zum Beispiel den aktuellen Hausverbrauch.
Es kommen also Daten.

NCIceWolf, 2025-02-16:
Hm, da müssten wir wohl das Detail-Debugging einstellen.
Hier sieht man nur dass es ein Problem bei der Verbindung gab und er nach einer Minute neu verbindet.
So weit, so erwartbar und wie gewünscht.
Ist auch als "Warnung" richtig, als Fehler würde das erst kommen wenn der Neuaufbau nicht funktionieren würde.

Ist nur die Frage welches Problem genau es bei dem Datenabruf gibt. Und warum das "nur" alle 5-10 Minuten auftritt.

Welches Intervall hast du denn eingestellt?
10 Sekunden oder länger?

Madaraxa, 2025-02-16:
Wie stelle ich Detail debugging ein?
Ich habe an der Standard Einstellung (10) nichts geändert. Hier nochmal der log seit dem ersten Warnung. Dann siehst du die Abstände der Meldungen.
`
rct.0
2025-02-16 18:42:56.031	info	RCT: Initial connection successful to inverter at 192.168.9.108!

rct.0
2025-02-16 18:41:56.028	warn	RCT: Connection error! Previous interval connection not successfully completed!

rct.0
2025-02-16 18:35:05.996	info	RCT: Initial connection successful to inverter at 192.168.9.108!

rct.0
2025-02-16 18:34:05.991	warn	RCT: Connection error! Previous interval connection not successfully completed!

rct.0
2025-02-16 18:28:15.957	info	RCT: Initial connection successful to inverter at 192.168.9.108!

rct.0
2025-02-16 18:27:15.953	warn	RCT: Connection error! Previous interval connection not successfully completed!

rct.0
2025-02-16 18:21:25.912	info	RCT: Initial connection successful to inverter at 192.168.9.108!

rct.0
2025-02-16 18:20:25.908	warn	RCT: Connection error! Previous interval connection not successfully completed!

rct.0
2025-02-16 18:14:35.872	info	RCT: Initial connection successful to inverter at 192.168.9.108!

rct.0
2025-02-16 18:13:35.868	warn	RCT: Connection error! Previous interval connection not successfully completed!

rct.0
2025-02-16 17:59:35.795	info	RCT: Initial connection successful to inverter at 192.168.9.108!

rct.0
2025-02-16 17:58:35.790	warn	RCT: Connection error! Previous interval connection not successfully completed!

rct.0
2025-02-16 16:48:45.417	info	RCT: Initial connection successful to inverter at 192.168.9.108!
`

aruttkamp, 2025-02-16:
und die IP 192.168.9.108 ist die des Masters  ? 
oder evtl. doch die des Slaves?

Madaraxa, 2025-02-16:
Ist der Master (power storage PS). Mein slave ist nen Power Inverter (Pi).

[image]

NCIceWolf, 2025-02-17:
> Wie stelle ich Detail debugging ein? Ich habe an der Standard Einstellung (10 Sekunden) nichts geändert. 

Müsste man im Code machen, also entweder durch Andreas, oder alternativ in meinem Fork. Das generiert aber eine gaaaaanze Menge Logs...

> Hier nochmal der log seit dem ersten Warnung. Dann siehst du die Abstände der Meldungen. Scheinen ja immer so ca 7 Minuten zu sein. ` rct.0 2025-02-16 18:42:56.031 info RCT: Initial connection successful to inverter at 192.168.9.108!
> 
> rct.0 2025-02-16 18:41:56.028 warn RCT: Connection error! Previous interval connection not successfully completed!
> 
> rct.0 2025-02-16 18:35:05.996 info RCT: Initial connection successful to inverter at 192.168.9.108!
> 
> rct.0 2025-02-16 18:34:05.991 warn RCT: Connection error! Previous interval connection not successfully completed!
> 
> rct.0 2025-02-16 18:28:15.957 info RCT: Initial connection successful to inverter at 192.168.9.108!
> 
> rct.0 2025-02-16 18:27:15.953 warn RCT: Connection error! Previous interval connection not successfully completed!
> 
> rct.0 2025-02-16 18:21:25.912 info RCT: Initial connection successful to inverter at 192.168.9.108!
> 
> rct.0 2025-02-16 18:20:25.908 warn RCT: Connection error! Previous interval connection not successfully completed!
> 
> rct.0 2025-02-16 18:14:35.872 info RCT: Initial connection successful to inverter at 192.168.9.108!
>
> Edit: wie gesagt ist das alles nicht tragisch. Ich bin nur ein Freund von einem 'sauberen' log. Ohne Warnungen. Das ganze tritt immer nur über Nacht auf, also bei 0 Produktion. Tagsüber läuft alles ohne Warnungen.

Kann ich absolut verstehen, bin ich auch 😄 
Es ist zwar relativ regelmäßig, aber leider nicht komplett.
Ich gehe da nach wie vor von einem internen Timeout aus und da beantwortet der PS dann die externe Anfrage nicht. Könnte man vielleicht verbessern wenn man den Timeout des Adapters von 2 Sekunden auf vielleicht 4-5 Sekunden erhöht, das gilt dann aber für jede einzelne Abfrage. Und man müsste testen ob das dann ausreicht.
"Gesunde" Abfragen brauchen eigentlich immer so 0,3-0,5 Sekunden.

Oder man ändert dieses Thema generell von "Warnung" zu "Information". Sollte die Verbindung auch nach 1 Minute nicht neu aufgebaut werden, wird ohnehin ein "Fehler" erzeugt. Das sollte aber nur der Fall sein wenn keine Batterie angeschlossen ist. Und auch das könnte man noch etwas schöner gestalten, da diese Fälle schon ein paar Meldungen über die Nacht bekommen.

Madaraxa, 2025-02-17:
Also wenn ich über die Katze deinen rct Adapter drüber installiere bekomme ich bei debug automatisch viele Logs? Dann würde ich das abends mal machen kurz bevor es dunkel wird. Das sollte dann ja reichen. Und sobald die 2-3 warnungen da sind kann ich ja auch wieder zurück auf den Main'Fork'

NCIceWolf, 2025-02-17:
Hab das eben wieder aktiviert. Also ja, installierst du den Adapter aus meinem Fork ist die debug-console auf "true".

Bitte da nicht erschrecken, da wird jede einzelne Verbindung zum WR mitgeloggt. Aber vielleicht kriegen wir da eine Idee ob das wirklich ein Timeout ist.

Madaraxa, 2025-02-17:
Also nachdem er immer folgendes geloggt hat:

`
<html><body>

rct.0 | 2025-02-17 18:11:29.997 | debug | RCT: Interval connection to inverter at 192.168.9.108 closed
-- | -- | -- | --
rct.0 | 2025-02-17 18:11:29.996 | debug | RCT: Terminating interval connection to inverter at 192.168.9.108
rct.0 | 2025-02-17 18:11:29.984 | debug | RCT: starting to terminate interval connection to inverter at 192.168.9.108
rct.0 | 2025-02-17 18:11:28.198 | debug | RCT: received: Grid-separated: 512
rct.0 | 2025-02-17 18:11:28.197 | debug | RCT: received: Battery voltage: 469.5 V
rct.0 | 2025-02-17 18:11:28.196 | debug | RCT: received: Battery state machine: 3
rct.0 | 2025-02-17 18:11:28.195 | debug | RCT: received: HeiPhoss mode: 1
rct.0 | 2025-02-17 18:11:28.195 | debug | RCT: received: io_board.s0_external_power: 0
rct.0 | 2025-02-17 18:11:28.194 | debug | RCT: received: Solar generator B voltage: 261.3 V
rct.0 | 2025-02-17 18:11:28.193 | debug | RCT: received: Solar generator A voltage: 262.4 V
rct.0 | 2025-02-17 18:11:28.192 | debug | RCT: received: Battery power: 5717 W
rct.0 | 2025-02-17 18:11:28.192 | debug | RCT: received: Load household - external Power: 5585.5 W
rct.0 | 2025-02-17 18:11:28.190 | debug | RCT: received: Total grid power: -0.3 W
rct.0 | 2025-02-17 18:11:28.189 | debug | RCT: received: Error bit field 4: 0
rct.0 | 2025-02-17 18:11:28.188 | debug | RCT: received: Error bit field 3: 0
rct.0 | 2025-02-17 18:11:28.187 | debug | RCT: received: Error bit field 2: 0
rct.0 | 2025-02-17 18:11:28.186 | debug | RCT: received: Error bit field 1: 0
rct.0 | 2025-02-17 18:11:28.182 | debug | RCT: received: Solar generator B power: 0 W
rct.0 | 2025-02-17 18:11:28.181 | debug | RCT: received: Solar generator A power: 0 W
rct.0 | 2025-02-17 18:11:28.180 | debug | RCT: received: SOC target low: 97 %
rct.0 | 2025-02-17 18:11:28.179 | debug | RCT: received: SOC target high: 97
rct.0 | 2025-02-17 18:11:28.176 | debug | RCT: received: Target SOC: 100 %
rct.0 | 2025-02-17 18:11:28.175 | debug | RCT: received: SOC (State of charge): 73.3 %
rct.0 | 2025-02-17 18:11:28.049 | debug | RCT: received: battery.bat_status: 0
rct.0 | 2025-02-17 18:11:27.981 | debug | RCT:  Requesting elements  "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag"  from inverter
rct.0 | 2025-02-17 18:11:27.979 | debug | RCT: Interval connection to inverter at 192.168.9.108 successfully established
rct.0 | 2025-02-17 18:11:27.975 | debug | RCT: Starting interval connection to inverter at 192.168.9.108

</body>
</html>
`
kam dann einige Male:

`
<html><body>

rct.0 | 2025-02-17 18:11:49.991 | debug | RCT: Terminating interval connection to inverter at 192.168.9.108
-- | -- | -- | --
rct.0 | 2025-02-17 18:11:49.982 | debug | RCT: starting to terminate interval connection to inverter at 192.168.9.108
rct.0 | 2025-02-17 18:11:47.979 | debug | RCT:  Requesting elements  "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag"  from inverter
rct.0 | 2025-02-17 18:11:47.979 | debug | RCT: Interval connection to inverter at 192.168.9.108 successfully established
rct.0 | 2025-02-17 18:11:47.976 | debug | RCT: Starting interval connection to inverter at 192.168.9.108

</body>
</html>
`

Dann kam einmal

`
<html><body>

rct.0 | 2025-02-17 18:15:47.995 | warn | RCT: Connection error! Previous interval connection not successfully completed!
-- | -- | -- | --
rct.0 | 2025-02-17 18:15:39.999 | debug | RCT: starting to terminate interval connection to inverter at 192.168.9.108
rct.0 | 2025-02-17 18:15:37.997 | debug | RCT:  Requesting elements  "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag"  from inverter
rct.0 | 2025-02-17 18:15:37.996 | debug | RCT: Interval connection to inverter at 192.168.9.108 successfully established
rct.0 | 2025-02-17 18:15:37.993 | debug | RCT: Starting interval connection to inverter at 192.168.9.108

</body>
</html>
`
Während dessen konnte ich auch über die APP nicht auf die Daten zugreifen.
Jetzt kommt wieder das lange Log,
Ich schicke eine neue Antwort, wenn die mehreren Fehler erscheinen.

Madaraxa, 2025-02-17:
Ja, leider nichts detaillierteres... immer nur 

rct.0 | 2025-02-17 18:49:08.151 | debug | RCT: Interval connection to inverter at 192.168.9.108 closed
rct.0 | 2025-02-17 18:49:08.150 | warn | RCT: Connection error! Previous interval connection not successfully completed!
rct.0 | 2025-02-17 18:49:00.159 | debug | RCT: starting to terminate interval connection to inverter at 192.168.9.108
rct.0 | 2025-02-17 18:48:58.155 | debug | RCT:  Requesting elements  "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag"  from inverter
rct.0 | 2025-02-17 18:48:58.152 | debug | RCT: Interval connection to inverter at 192.168.9.108 successfully established
rct.0 | 2025-02-17 18:48:58.148 | debug | RCT: Starting interval connection to inverter at 192.168.9.108

NCIceWolf, 2025-02-17:
Das ist für mich schon detailliert genug.
Im "Fehlerfall" kriegt er nicht nur (gar) keine Daten vom WR, es wird nicht einmal die Verbindung sauber abgebaut.

Dafür hatte ich das damals so eingebaut, er "killt" dann vor dem Neuaufbau komplett die alte Verbindung.

Daran ist der Adapter damals auch hängen geblieben, da immer die gleiche Verbindung genutzt wurde und nie neu aufgebaut wurde. Und dann blieb er an der alten "kaputten" Verbindung hängen und es gab über Stunden gar keine Daten mehr.

Aus Adapter Sicht ist das also "works as designed".
Es wird festgestellt dass die letzte Verbindung nicht korrekt verwendet werden konnte und deshalb gibt es eine Warnung. Hast du ja auch so in der App feststellen können.

Was genau der WR in dieser Zeit macht, kann dir wahrscheinlich nur RCT beantworten. Ich vermute nach wie vor: mit sich selbst beschäftigt sein.

Wenn du den Loglevel auf "Fehler" umstellst, würdest du nur noch sehen wenn auch die Neuverbindung 1 Minute später scheitern würde (z.b. wenn du keine Batterie hättest oder der WR komplett aussteigt).

@aruttkamp: Wir siehst du das?

aruttkamp, 2025-02-18:
Moin @NCIceWolf ,
ja, ich sehe das ähnlich ! 
Lass uns doch den Teil mit dem Detail Log in den Adapter übernehmen und über einen Schalter im Admin aktivieren. 
dann entfällt für den User das hin und her mit der Installiererei.
Magst einen PR dafür erstellen ?

NCIceWolf, 2025-02-18:
Ist eine sehr gute Idee, bin ich nie drauf gekommen 😆 
Bau ich gerne ein

closed 2025-02-21 by aruttkamp

Madaraxa, 2025-04-23:
I just wanna Thank you again. I wrote rct about this issue and they somehow found out, that my battery Master is broken and Sent a new one.
The First nights after this i still had the errors.
3 days ago i got a minor Update of my Inverters and battery System. Since then no more warnings. I am Not sure if that is fixed by the update or new Master or if it just doesnt occure as my battery brings me through the night at the Moment

NCIceWolf, 2025-04-23:
Thanks for your feedback!!
That does indeed fit what we found back then!
Interesting that errors within the BMS can be found by the adapter 😄 
And I guess it really was the BMS, the new one probably had an old software version and had to be updated to work correctly with the inverter.

Really happy to hear that you received a new BMS and RCT was indeed able to fix your problem!!

PR #209 (2025-02-21): Adding debug-mode in admin-panel
- Resolves #208
- Resolves #207
- Adds the possibility to enable "debug mode" in admin panel
- Fixed "Incomplete string escaping or encoding" (referring to: CodeQL; Rule ID: js/incomplete-sanitization)