aruttkamp/ioBroker.rct #218: RCT: Connection error! Previous interval connection not successfully completed!

BenS89, 2025-03-08:
Seit 2 Tagen ist das Log voll mit den Meldungen RCT: Connection error! Previous interval connection not successfully completed!

Keine Änderungen am Wechselrichter oder Adapter vorgenommen 

Wechselrichter Firmware: 2.3.5678
Adapter Version: 1.2.18

Über die RCT App ist der WR problemlos erreichbar. Bislang kannte ich nur die Problematik, dass der WR gar nicht mehr erreichbar ist, weil der TCP/IP Stack im WR abgeraucht ist und der WR neu gestartet werden musste 

Das scheint hier aber nicht der Fall zu sein 

Habt ihr eine Idee, wie ich das beheben kann?

aruttkamp, 2025-03-08:
Moin !
Und bitte gerne deine Parameter Posten die du abrufst !

BenS89, 2025-03-13:
Sorry für die späte Rückmeldung. Lag tatsächlich an was anderem: das Netzwerkinterface meines Nuc hatte wohl einen schlechten Treiber, sodass sich dieses aufgehängt hatte. 

Nach zig Stunden der Analyse in Proxmox Foren hab ich gestern nen Workaround gefunden 

Sorry für die Falschmeldung, Adapter funktioniert bestens

closed 2025-03-13 by BenS89

aruttkamp, 2025-03-14:
Danke @BenS89 für die Rückinfo ! Ja, manchmal muss man erst drauf kommen das es ganz woanders hakt. 
Kein Problem !

Madaraxa, 2026-01-02:
Hi. Ich bin auf der 2.3.5687/5615. Ich bin nicht 100% sicher glaube aber, dass ich mit 2.3.5687/5611 noch keine Probleme hatte. Das WR Update kam früher als das Update des BMS.
Ich hab mal den debug Modus im Adapter aktiviert und melde mich morgen mit logs.
Netzwerkprobleme kann ich 'ausschließen' (auch wenn das ja nie zu 100% geht). Da hab ich schon echt lange nichts dran geändert und die Meldungen kommen ja auch nur nachts.
Evtl ist ja jemand auf der 5687/5611 ohne Probleme der dies einfach kurz bestätigen kann und jemand anderes auf der 5687/5615 der das selbe Phänomen hat :) auch wenn das ganze dann nicht gelöst ist, würde es zumindest für nen rct 'Fehler' sprechen

Madaraxa, 2026-01-03:
Ich hatte bis gestern für mehrere Tage (mindestens 10) diese warnings (mehrere) jede Nacht. Gestern den debug Modus aktiviert und heute Nacht keine einzige Meldung... Ich lasse ihn mal weiter laufen bis morgen und schalte ggf übermorgen wieder zurück. Ich werde berichten...

Madaraxa, 2026-01-03:
Hi.
Ja, das Portal habe ich seit Ewigkeiten aktiviert. Quasi seit dem Anfang als es das Portal gab (mehrere Jahre schon). Service ist nicht aktiviert und war es auch ewig nicht mehr.

Madaraxa, 2026-01-04:
Was soll ich sagen. Seit vorgestern keine warnings mehr. Hab heute tagsüber auch das logging wieder zurück gestellt.

Madaraxa, 2026-01-05:
ok. ich war dumm. Ich hab durch das viele logging im debug modus einfach nicht gesehen, dass warnings im log waren (da diese nicht mehr im web interface angezeigt wurden). Hab das log runter geladen und mehrere warnings sind da. 
reicht dieser Ausschnitt ?
EDIT: wie bekomme ich das hier gepostet in ner Form, die ansehnlich ist :)?
EDIT2: Ich habe hier erstmal weniger Zeilen in der hoffnung, dass das schon reicht. Nach dem Absatz kommt dann (unübersichtlich) ein größerer Auszug.

2026-01-04 21:16:07.233  - [34mdebug[39m: rct.0 (90238) RCT: starting to terminate interval connection to inverter at 192.168.9.108
2026-01-04 21:16:07.242  - [34mdebug[39m: rct.0 (90238) RCT: Terminating interval connection to inverter at 192.168.9.108
2026-01-04 21:16:07.244  - [34mdebug[39m: rct.0 (90238) RCT: Interval connection to inverter at 192.168.9.108 closed
2026-01-04 21:16:15.226  - [34mdebug[39m: rct.0 (90238) RCT: Starting interval connection to inverter at 192.168.9.108
2026-01-04 21:16:15.229  - [34mdebug[39m: rct.0 (90238) RCT: Interval connection to inverter at 192.168.9.108 successfully established
2026-01-04 21:16:15.229  - [34mdebug[39m: rct.0 (90238) RCT: Requesting elements "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag" from inverter
2026-01-04 21:16:17.232  - [34mdebug[39m: rct.0 (90238) RCT: starting to terminate interval connection to inverter at 192.168.9.108
2026-01-04 21:16:25.227  - [33mwarn[39m: rct.0 (90238) RCT: Connection error! Previous interval connection not successfully completed!
2026-01-04 21:16:25.228  - [34mdebug[39m: rct.0 (90238) RCT: Interval connection to inverter at 192.168.9.108 closed
2026-01-04 21:17:25.229  - [34mdebug[39m: rct.0 (90238) RCT: Starting interval connection to inverter at 192.168.9.108
2026-01-04 21:17:25.232  - [32minfo[39m: rct.0 (90238) RCT: Initial connection successful to inverter at 192.168.9.108!
2026-01-04 21:17:25.232  - [34mdebug[39m: rct.0 (90238) RCT: Interval connection to inverter at 192.168.9.108 successfully established
2026-01-04 21:17:25.233  - [34mdebug[39m: rct.0 (90238) RCT: Requesting elements "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag" from inverter

---------------------------------Ab hier nicht so ansehnlich---------------------------------------------------

2026-01-04 23:38:07.988  - [34mdebug[39m: rct.0 (90238) RCT: starting to terminate interval connection to inverter at 192.168.9.108
2026-01-04 23:38:07.997  - [34mdebug[39m: rct.0 (90238) RCT: Terminating interval connection to inverter at 192.168.9.108
2026-01-04 23:38:07.999  - [34mdebug[39m: rct.0 (90238) RCT: Interval connection to inverter at 192.168.9.108 closed
2026-01-04 23:38:15.981  - [34mdebug[39m: rct.0 (90238) RCT: Starting interval connection to inverter at 192.168.9.108
2026-01-04 23:38:15.983  - [34mdebug[39m: rct.0 (90238) RCT: Interval connection to inverter at 192.168.9.108 successfully established
2026-01-04 23:38:15.984  - [34mdebug[39m: rct.0 (90238) RCT: Requesting elements "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag" from inverter
2026-01-04 23:38:17.986  - [34mdebug[39m: rct.0 (90238) RCT: starting to terminate interval connection to inverter at 192.168.9.108
2026-01-04 23:38:17.996  - [34mdebug[39m: rct.0 (90238) RCT: Terminating interval connection to inverter at 192.168.9.108
2026-01-04 23:38:17.997  - [34mdebug[39m: rct.0 (90238) RCT: Interval connection to inverter at 192.168.9.108 closed
2026-01-04 23:38:25.982  - [34mdebug[39m: rct.0 (90238) RCT: Starting interval connection to inverter at 192.168.9.108
2026-01-04 23:38:25.984  - [34mdebug[39m: rct.0 (90238) RCT: Interval connection to inverter at 192.168.9.108 successfully established
2026-01-04 23:38:25.985  - [34mdebug[39m: rct.0 (90238) RCT: Requesting elements "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag" from inverter
[... cut ...]
2026-01-04 23:39:45.989  - [34mdebug[39m: rct.0 (90238) RCT: Requesting elements "battery.bat_status,battery.soc,battery.soc_target,battery.soc_target_high,battery.soc_target_low,dc_conv.dc_conv_struct[0].p_dc_lp,dc_conv.dc_conv_struct[1].p_dc_lp,fault[0].flt,fault[1].flt,fault[2].flt,fault[3].flt,g_sync.p_ac_grid_sum_lp,g_sync.p_ac_load_sum_lp,g_sync.p_ac_sum_lp,g_sync.p_acc_lp,g_sync.u_sg_avg[0],g_sync.u_sg_avg[1],io_board.s0_external_power,power_mng.is_heiphoss,power_mng.state,power_mng.u_acc_mix_lp,prim_sm.island_flag" from inverter

Madaraxa, 2026-01-06:
Hab das Portal mal deaktiviert. Am Netzwerk kann es meiner Meinung nach genauso wenig liegen wie am Host, da es nur wenn DC 0 ist zu Warnungen kommt...
Die rct app habe nur ich auf dem Handy und habe sie den ganzen Tag nicht geöffnet. :(
Scheint also irgendwie Pech zu sein. Es war vor einigen Wochen alles gut ohne warnings. Daher meine Vermutung mit dem aktuellen Update. Aber da du das ja auch hast.
Ich hab hier nen PS mit Batterie, und nen PI als slave. Evtl findet sich ja jemand mit genau dieser Konstellation der keine Fehler hat. Dann würde ich mir überlegen rct zu kontaktieren. Das letzte mal, als ich die Probleme hatten war mein BMS Master kaputt und wurde durch rct getauscht. Da kam ich aber auch regelmäßig nachts über die app nicht an die daten. Das ganze hab ich jetzt noch nicht testen können, da die Ausfälle immer nur selten und eher kurz sind.

NCIceWolf, 2026-01-11:
Ja, die Beschriftung "Leistungstest" ist vor einiger Zeit rein gekommen. Heißt nix anderes als dass keine Energie ins Hausnetz gespeist wird. Ist aus meiner Sicht auch keine sehr elegante Beschriftung.

Wie du bemerkt hast wird aber dennoch der WR selbst weiter durch die Batterie gespeist. Das geht bis auf 5% runter, dann wird die Batterie wieder aus dem Netz auf mind. 7% geladen.

Ist also alles normal.